arnemauer/Ducobox-ESPEasy-Plugin #1: Hardware

markiey, 2019-02-16:
Met heel veel interesse lees ik hier over deze oplossing. Ik heb zelf de duco silent box en wil deze graag uitlezen en dmv domiticz aansturen.

Alleen weet ik niet goed welke hardware ik moet bestellen, heb jij een lijst met links naar de verschillende producten? zo weet ik zeker dat ik de juiste heb :)

Alvast bedankt

arnemauer, 2019-04-11:
Beste Markiey,

Had je de wiki pagina's al bekeken? 
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/Hardware---RF-gateway

Mocht er iets niet duidelijk zijn dan hoor ik het graag.

arcangel321, 2019-04-28:
Helaas snap ik niet welke hardware er nodig is.
Ik zie wel de hardware op de hoofdpagina maar hoe moet ik deze verbinden aan elkaar.
Ik zie dat je nog een extra print hebt gebruikt?
En hoe monteer ik de serial verbinding?

Alvast bedankt

broekieman, 2019-09-20:
> Met heel veel interesse lees ik hier over deze oplossing. Ik heb zelf de duco silent box en wil deze graag uitlezen en dmv domiticz aansturen.
>
> Alleen weet ik niet goed welke hardware ik moet [...]

Is het jou inmiddels gelukt om iets werkend te krijgen?

mkampstra, 2019-11-30:
Beste Arne, ik heb eigenlijk een beetje dezelfde vraag.
Ik wil graag de module nabouwen die op HARDWARE.JPG te zien is. 

Welke hardware zit daarin? Is dit de serial en RF gateway samengevoegd? 

Heb je onderdeelnummers of websites waar de onderdelen gekocht zijn?
Ik kan de juiste spullen niet makkelijk vinden. En om nu te bestellen en 6 weken te wachten en dan het verkeerde ontvangen is frustrerend. 

Groeten Merte

arnemauer, 2019-11-30:
Beste Merte,

Je kan hier de onderdelen en een aansluitschema vinden:
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/x3.-RF-Gateway---Hardware

De serial en RF gateway kun je los van elkaar maar ook samen op één Wemos D1 aansluiten. Alle onderdelen kun je wel op Aliexpress vinden. Let op dat er veel Wemos D1 pro namaak versies zijn waarbij de weerstanden op de seriële lijn veel te laag zijn. Daardoor kan het zijn dat de seriële gateway niet werkt. Je kan de Wemos het beste bij de officiele fabrikant kopen: <link volgt>.

De PCB kun je hier vinden:
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/master/DucoGWv3%20schematic.pdf
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/master/DucoGWv32%20PCB.pdf

mkampstra, 2019-12-03:
Thanks Arne,

Ik heb over veel onderdeeltjes heen gekeken blijkbaar. 
De PCB print service is toch wat duurder dan verwacht. Ik las dat je er nog hebt. Aanvraag is gedaan. 

Groeten Merte

HA-TB303, 2020-02-26:
Van mij ook een aanvraag. Ik hoop dat je er nog een paar hebt liggen :)

IgorYbema, 2020-03-24:
@arnemauer vandaag een ducobox silent geinstalleerd in mijn woning en stuitte zojuist toevallig op je repo :-)
Ik heb gelijk een vraag, ook hardware gerelateerd en daarom steel ik deze topic even.
Waarom gebruikt je niet de 3.3v van de ducobox als voeding voor de wemos? Ik was dit van plan te doen totdat ik in je wiki zag "don't use this".

IgorYbema, 2020-03-24:
Negeer mijn bericht. Het blijkt dat mijn ducobox geen 3.3v meer op die pin heeft. Er vlak boven zit ook trace met een solderjumper die open is en vlak daar voor zit 5v. Die trace zie ik niet op jouw foto terug. Lijkt wel of ze op mijn pcb (nieuwer?) nu 5v gebruiken en dus hem van die pin hebben afgehaald.

arnemauer, 2020-04-14:
> Negeer mijn bericht. Het blijkt dat mijn ducobox geen 3.3v meer op die pin heeft. Er vlak boven zit ook trace met een solderjumper die open is en vlak daar voor zit 5v. Die trace zie ik niet op jouw [...]

Het zou kunnen dat jou pcb nieuwer is. Los daarvan zal de Ducobox niet voldoende stroom kunnen leveren om de ESP te kunnen voeden:

[image]

IgorYbema, 2020-04-14:
Ik heb geprobeerd om een scherpe foto te maken maar niet helemaal duidelijk
[image]

arnemauer, 2020-04-14:
Ik vermoed dat ze dezelfde voltage regulators hebben gebruikt. De 3.3v regulator zal maximaal 150mA leveren. De ESP8266 gebruikt ongeveer 170mA met pieken tot 350mA. Aangezien de voltage regulator slechts 150mA levert gaat dat niet werken.

[image]

closed 2020-04-14 by arnemauer

IgorYbema, 2020-04-14:
Nou nee, er zit nu dus 5v op de uitgang van de regulator ipv 3.3v. Kan net niet goed zien wat voor type het is.

IgorYbema, 2020-04-14:
Net nog even beter gekeken. Is exact zelfde 3.3v regulator. De input van de regulator is 5v (geen idee waar die gemaakt wordt overigens). En diezelfde input hebben ze dus via die solderjumper beschikbaar gemaakt op de laatste pin (die pin en de onderkant van de solderjumper is verbonden, zojuist doorgemeten).

arnemauer, 2020-04-14:
> Net nog even beter gekeken. Is exact zelfde 3.3v regulator. De input van de regulator is 5v (geen idee waar die gemaakt wordt overigens). En diezelfde input hebben ze dus via die solderjumper [...]

Waarschijnlijk wordt die 5v gemaakt door de LM1117. Die kan maximaal 800mA leveren. Mocht je een ESP rechtstreeks van je Ducobox willen voeden dan zul je al stroom (bijna) rechtstreeks vanaf te transformator moeten pakken. Na de gelijkrichter (paars) zou je tweetal draden aan de pcb kunnen solderen (blauw). Ik zou zelf kiezen voor een aparte usb adapter omdat je naast je Ducobox toch al stroom hebt.
[image]