arnemauer/Ducobox-ESPEasy-Plugin #2: Domoticz IDX > 255

Lxother, 2019-03-16:
Beste Arne,

Dank je voor jouw mooie Ducobox Serial Gateway voor mijn Domoticz systeem en de heldere omschrijvingen!

Vandaag heb ik alles in elkaar gezet, gesoldeerd (JST kabel en Wemos D1 mini) en jouw .bin file geflashed. Na de standaard installatie van ESPeasy liep ik echter helemaal vast met het laatste puntje : het invullen van de Domoticz IDX values; die zijn bij mij namelijk hoger dan 255 :-(

Ik ben niet zo'n held in programmeren maar heb even in jouw code gekeken en zie daar bij de betreffende invulvelden 255 staan. Ik ben alleen bang dat die waarde er niet voor niks staat.

Zou je aan kunnen en willen geven of mijn uitdaging simpelweg verholpen kan worden door betreffende 255 waarden aan te passen naar bijvoorbeeld 512 of dat er dan meer gedaan moet worden in de code?

Indien mogelijk zou een aangepaste .bin file zeer op prijs gesteld worden.

Hartelijk dank bij voorbaat voor het bekijken van mijn issue.

Met vriendelijke groet,

Lx

arnemauer, 2019-04-11:
Beste Lx,

Ik heb de code aangepast zodat je een IDX op kan geven tot maximaal 5000. Is dat voldoende? :)
De bin firmware is ook geupdate.

https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/00736206029fb7001451bc514095708d36a678b3

closed 2019-04-11 by arnemauer

Lxother, 2019-04-11:
Beste Arne,

Dat is zeker voldoende :) Grote dank voor het aanpassen en de update van de firmware!

Met vriendelijke groet,

Alexander