arnemauer/Ducobox-ESPEasy-Plugin #4: No CO2 PPM value found in package

nielsvn92, 2019-04-15:
Hi,

I got your plugin working on my ESP8266, but it fails in parsing the values from the `Network` command.

This is the table that I get:
```
node|type|ptcl|cerr|prnt|asso|stts|stat|cntdwn|%dbt|trgt|cval|snsr|ovrl|capin |capout|tree|temp|info
  1   |VIRT|1   |1   |BOX     |15  |0   |0   |0   |0x00|AUTO|0     |20  |12  |12  |0   |255 |220   |3520  |0x01|0   |AUTO:0(0)
  2   |RF  |2   |1   |UCBAT   |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-   |0   |-   |-     |-     |-   |0   |-
  3   |RF  |3   |1   |UCBAT   |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-   |0   |-   |-     |-     |-   |0   |-
  25  |VIRT|1   |1   |UCCO2   |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-   |0   |-   |-     |-     |-   |217 |-
  26  |VIRT|1   |1   |UCRH    |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-   |10  |-   |-     |-     |-   |217 |-
  --- end list -
```

The `addr` column seems to be missing vs what you expect in https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/master/_P151_DucoSerialGateway.ino#L388.

I'm getting both an incorrect status (something like 125), and no sensor values.
Is this something you could change? Not sure if this is due to a different firmware version or something.

Kind regards,
Niels

arnemauer, 2019-04-19:
Could you post which ducobox you have and what the software/hardware version is?

arnemauer, 2019-04-19:
Hi Niels, the table you posted. Did you get this from the espeasy log or did you connect your pc to the ducobox?

I'm missing some column names in your table:
`                 node|type	  |ptcl|cerr|prnt|asso|stts|stat|cntdwn|%dbt|trgt|cval  |snsr  |ovrl|capin |capout|tree|temp|info
  1   |VIRT|1   |1   |BOX     |15  |0   |0   |0   |0x00|AUTO|0     |20  |12  |12    |0     |255 |220   |3520  |0x01|0   |AUTO:0(0)
  2   |RF  |2   |1   |UCBAT   |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-     |0     |-   |-     |-     |-   |0   |-
  3   |RF  |3   |1   |UCBAT   |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-     |0     |-   |-     |-     |-   |0   |-
  25  |VIRT|1   |1   |UCCO2   |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-     |0     |-   |-     |-     |-   |217 |-
  26  |VIRT|1   |1   |UCRH    |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-     |10    |-   |-     |-     |-   |217 |-
  --- end list -`

nielsvn92, 2019-04-20:
Hi Arne,

I just checked, I have the Ducobox Silent with software version 4.8.0.
The table I posted is from the ESPEasy logs, but it could very well be that there is some missing data, because it caps the logs :-).

I could try in the upcoming few days to connect my PC again and see if I can get the raw output of the Network command via the Duco Network Tool.

arnemauer, 2019-04-20:
Duplicate of #5

arnemauer, 2019-04-21:
I have used this serial port monitor (https://www.eltima.com/products/serial-port-monitor/) before to catcht he communication between the networktool and the ducobox. Can you send me the logs of the serial port? In the networktool you need to open the page of the co2 device to fetch the variables :)

nielsvn92, 2019-04-21:
Here is the output of the `network` command from Duco Network Tool:
```
> network
  Network:
  --- start list ---
  node|netw|addr|sub |type    |ptcl|cerr|prnt|asso|stts|stat|cntdwn|%dbt|trgt|cval|snsr|ovrl|capin |capout|tree|temp|info
  1   |VIRT|1   |1   |BOX     |15  |0   |0   |0   |0x00|AUTO|0     |20  |13  |13  |0   |255 |220   |3520  |0x01|0   |AUTO:0(0)
  2   |RF  |2   |1   |UCBAT   |13  |0   |1   |1   |0x01|-   |0     |-   |-   |-   |0   |-   |-     |-     |-   |0   |-
  3   |RF  |3   |1   |UCBAT   |13  |0   |1   |1   |0x06|-   |0     |-   |-   |-   |0   |-   |-     |-     |-   |0   |-
  25  |VIRT|1   |1   |UCCO2   |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-   |0   |-   |-     |-     |-   |233 |-
  26  |VIRT|1   |1   |UCRH    |255 |0   |1   |1   |0x00|-   |0     |-   |-   |-   |10  |-   |-     |-     |-   |233 |-
  --- end list ---
```

Here is a link to 3 captures that I've done: https://drive.google.com/open?id=1Hc0U5gOJNOHqXsujX_1vQHdrsG_ya401
- One with a global refresh
- One with a refresh for the CO2 sensor
- One with a refresh for the Humidity sensor

Perhaps a good opportunity to add in the Humidity sensor as well? :-)

Pinkel, 2019-04-24:
@nielsvn92 : with new builds (>=23/4/2019) both the CO2 and RH should work. I don't think @arnemauer  has uploaded a new build yet, but can you try it once he does?

Your CO2 seems to be a box sensor, not a room sensor. I've added support for this in pull request #11 and expect it to show up with the command 'sensorinfo'.
To test it, once there is a new build, configure a new device of the type 'DUCO Serial Box Sensors'.
For me it shows up like: 

[image]

nielsvn92, 2019-04-24:
Ahh ok, I thought that meant the same :-). Thank you for clarifying!
I'll wait on the new build, and reply back once I know more.

arnemauer, 2019-04-24:
There is a new build available! 
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/abd56c39824d97324f8feee859d292d8cab0cbdb

nielsvn92, 2019-04-25:
Awesome, I'm now able to read out my box sensors!

Thanks guys!

closed 2019-04-25 by nielsvn92