arnemauer/Ducobox-ESPEasy-Plugin #5: Duco serial gateway only CO2 PPM

NLNoID, 2019-04-18:
Hi,

I'm receiving only the CO2 PPM value, the other values are 0. Any idea what this could be. It's a ducobox Silent.

BasVanBoven, 2019-04-19:
I have the same issue, although the status is often not zero but a value between 120-130 for me. Ducobox Silent as well. Fortunately I only need the PPM value for my application, although it would be nice to retrieve the status as well. :)

arnemauer, 2019-04-19:
Could you tell mee what kind of ducobox you have and which software/hardware version?

Is it possible for you to connect a pc directly to your ducobox and post the device table here?

NLNoID, 2019-04-20:
Hi, can confirm the status value also. I'm getting 124 consistently now. I'll connect a serial cable and let you know the results.

arnemauer, 2019-04-20:
The serial gateway now uses fixed columnnumbers based on my ducobox output. Thee device table is different per ducobox type/ hardware version. So i added a function 'getColumnNumberByColumnName' to get the columnnumber based on the columnname. Could you please check if this firmware works with your ducobox?

#4d0ede5689a99da63174382100f84583f0016e40

BasVanBoven, 2019-04-20:
Amazing! Surely I will be able to test it, will be home again sunday evening so test results will be in monday at the latest :)

nielsvn92, 2019-04-21:
Awesome, I can confirm I at least have the correct status and flow percentage!
However, I don't have any CO2 values. Will check the Network output.

NLNoID, 2019-04-21:
We're out for the weekend also but I'll let you know as soon as possible. Sounds good!

arnemauer, 2019-04-21:
Could you send a piece of the espeasy log?

Specific for the CO2 PPM reading:
83221476: [P151] DUCO SER GW: Start readCO2PPM
83221859: [P151] DUCO SER GW: Send command successfull!
83221880: [P151] DUCO SER GW: Package received with stopword.
83221881: [P151] DUCO SER GW: Exptected command received.
83221882: start_CO2_PPM_byte: 47
83221903: co2 ppm valuebytes: raw = 34 33 36
83221924: [P151] DUCO SER GW: CO2 PPM value = 0  >>> **_is always zero_**, I changed it in next commit.

NLNoID, 2019-04-21:
Hi, Many thanks!! Just remotely updated the ESP and I'm now receiving all values. Let you know more tomorrow, thanks again!

nielsvn92, 2019-04-21:
It actually says it couldnt find the CO2 PPM:
```51035590: [P151] DUCO SER GW: read networkList
51035590: [P151] DUCO SER GW: Start readCO2PPM
51035970: [P151] DUCO SER GW: Send command successfull!
51036957: [P151] DUCO SER GW: Package received with stopword.
51036957: [P151] DUCO SER GW: Exptected command received.
51036957: start_CO2_PPM_byte: 0
51036977: co2 ppm valuebytes: raw = 6E 6F 64
51036998: [P151] DUCO SER GW: No CO2 PPM value found in package.
51036998: [P151] DUCO SER GW: CO2 PPM value = 0```

Pinkel, 2019-04-22:
#6 should fix the CO2 PPM value showing up

NLNoID, 2019-04-24:
Can confirm that it's working correctly now, receiving all values. Many thanks!!

closed 2019-04-24 by NLNoID