arnemauer/Ducobox-ESPEasy-Plugin #16: Wrong parameter name when sending CO2 PPM to domoticz

arnemauer, 2019-05-07:
For sending data to domoticz for an "air quality" device we need to use this url:
/json.htm?type=command&param=udevice&idx=IDX&**nvalue**=PPM

The problem is espeasy sends all values with the parameter **svalue**=1234&nvalue=0. 

**Workaround**
Enable the Rules in ESPEasy and add this rule:
Where:
<DUCO_CO2_SENSOR> = Devicename of the MHZ19 unit in ESPEASY
<DomoticsIP> = IP address of your domoticz server
<port> = port used by your domoticz server
<IDX> = The IDX of the device in Domoticz

`on <DUCO_CO2_SENSOR>#CO2_PPMdo
  SendToHTTP,<DomoticsIP>,<port>,/json.htm?type=command&param=udevice&idx=<IDX>&nvalue=[DUCO_CO2_SENSOR#CO2_PPM]`

arnemauer, 2022-04-06:
Toegevoegd aan wiki handleiding: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/Domoticz#workaround-voor-p152box-sensor---co2--p154external-co2-sensor
Vooralsnog geen fix in espeasy voor dit type sensor.

closed 2022-04-06 by arnemauer

Jimboer, 2022-04-12:
Werkt nu na wat aanpassingen van de rule:

On <DUCO_CO2_SENSOR>#CO2_PPM do 
 SendToHTTP,< DomoticsIP>,< port>,/json.htm?type=command&param=udevice&idx=< IDX>&nvalue=[<DUCO_CO2_SENSOR>#CO2_PPM]
endon