arnemauer/Ducobox-ESPEasy-Plugin #17: Pcb layout/design

phoenix-blue, 2019-08-10:
Hi,

I see on the photo's that there is used an nice pcb design. 
Are those pcb design avaliable?
Let me know it. 
Regards,

broekieman, 2019-09-20:
Kun jij Arne nog bereiken? Heb jij het voor elkaar gekregen om met de DucoBox te werken via de RF Gateway?

phoenix-blue, 2019-09-20:
Hallo,
Nee nog niet, inmiddels wel de seriële interface normaal werkend. 
Helaas als ik de cc1101 er bij prik crashed de Esp8266, hier moet ik me nog in verdiepen. 

Hou me nog wel aanbevolen in het Pcb ontwerp.

arnemauer, 2019-09-20:
@broekieman het is hier geen helpdesk waar je binnen 48 uur antwoord krijgt ;) 

@phoenix-blue ik zal het weekend de pcb op github zetten.

arnemauer, 2019-09-25:
@phoenix-blue 
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/master/DucoGWv32%20PCB.pdf
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/master/DucoGWv3%20schematic.pdf

closed 2019-09-25 by arnemauer