arnemauer/Ducobox-ESPEasy-Plugin #19: Duco laten werken met domoticz

broekieman, 2019-09-18:
Hoi,

Heb je een oplossing om de duco box draadloos te laten werken met domoticz. Ik wil via een blocky event de ventilatie op hoogste stand laten draaien zodra de luchtvochtigheid in de badkamer omhoog gaat. Ik heb een bedrade de Co2 schakelaar in de woonkamer en een draadloze op de badkamer. Domoticz gebruik ik al en sensors ook.

arnemauer, 2019-09-18:
Beste Broekieman,

De Ducobox RF/serial module werkt out-of-the box met domoticz! :) Je verbindt de module via wifi met je thuisnetwerk. Vervolgens kun je domoticz zo instellen dat je de ventilatiewaarden ziet en ook de ventilatiestanden van de Ducobox kan aansturen. Op de wiki vind je hier informatie over hoe de gateway werkt met Domoticz: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/Configuration-of-Domoticz

Je moet nog wel de gateway bouwen: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/Hardware---RF-gateway 

Of je kan hem kant en klaar bestellen: https://forms.gle/PX1tb2zWaGPE9N7P9
![alt text](https://lh3.googleusercontent.com/BwMq_kRBSWiwzLijpkmwPA1R2sJZLi7zy4GyY_how3kKHnGUe9z8nwNkQxhgmt_3JARkmPTbv7FDmz99WHIDQ6mGINUzBPpCQxCJR7BuxoFqu5GyYdGNuIPVW96-RQD9ebi2uOOi0qQMArJFxurA-l80G9yedwmcSlkiwf3SRvTiT6x6ZZG6vA01nDOni1H-FuWfYvo7SWxxftWyZ-NyQvb3njz-nzTb0cMnRHbCEaqT4UBUpOQsn-rpI0YrKb8rvKOaej1RvaKkod9okyXCCn8X40NpM0o04w3SbqwSpESJ7hqvjz--geNArxl7PCEA0UW3I3IfthfQQJU_csnE970Vhe4jU0bwnGglWz2aplGbeDiMzHTiZzZPQh7z5NSvGXKWVCCnN63BEo6vuuTTtMcAY0CYZ1ES1yyu0iymhjRiZbabDb6GXdfPrybYpEv4iB5yM7qDF8k6ruhUFaU8mYdV-I1Z_eVRadBLM7QW-XyTBrdH7gM0OG1KhW3qdEt1Ysr6_0GKrQ7MHowRg-kUCMh-8XDiBYesPoEjRMJKZ0ps6zgtTaIdvBWv-5qbaxoA8tkoS_zBqwKdmUqTlLbR1Gdcrcgdejus7f3S6mF4PyTGx_hg2v_EsrepZOmgZljLt41QRuViZft-SJQR0X0TU57VdC7-c2VgbV325QjTjPjNF2VC0NkvolwzNDdHPYAtpYrBypOkmGv5lghzeqkYk-uGWJHcq-78vN3ARnh8IvCegAtiDA=w300)

broekieman, 2019-09-18:
Zijn er 2 soorten hardware of zitten zowel seriële en rf in het kastje? Kan ik alles uitlezen via een webinterface? Is het dan hetzelfde wat je ermee kunt als de techneuten/monteurs van Duco. Geef alvast je gegevens voor het bestellen want ik vind het leuk om ermee aan de slag te gaan. Met vriendelijke groet Maarten Met vriendelijke groet,Maarten Broekman-------- Oorspronkelijk bericht --------Onderwerp: Re: [arnemauer/Ducobox-ESPEasy-Plugin] Duco laten werken met domoticz  (#19)Van: arnemauer Aan: arnemauer/Ducobox-ESPEasy-Plugin Cc: broekieman ,Author Beste Broekieman,
De Ducobox RF/serial module werkt out-of-the box met domoticz! :) Je verbindt de module via wifi met je thuisnetwerk. Vervolgens kun je domoticz zo instellen dat je de ventilatiewaarden ziet en ook de ventilatiestanden van de Ducobox kan aansturen. Op de wiki vind je hier informatie over hoe de gateway werkt met Domoticz: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/Configuration-of-Domoticz
Je moet nog wel de gateway bouwen: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/Hardware---RF-gateway
Of je kan hem kant en klaar bestellen: https://forms.gle/PX1tb2zWaGPE9N7P9

broekieman, 2019-09-19:
> Beste Broekieman,
>
> De Ducobox RF/serial module werkt out-of-the box met domoticz! :) Je verbindt de module via wifi met je thuisnetwerk. Vervolgens kun je domoticz zo instellen dat je de [...]

Hoi Arne,

Ik weet niet hoe het precies hier werkt met berichten. Kun je nog even naar mijn vragen kijken?

arnemauer, 2019-09-20:
**- Zijn er 2 soorten hardware of zitten zowel seriële en rf in het kastje?** 
Beide zitten in de gateway. De seriële gateway werkt via een kabel die je aansluit op de ducobox. Je kan daarmee de ventilatiestand, ventilatiepercentage, co2 PPM en co2 temperatuur. Via de seriele kabel kan je echter niet ducobox aansturen (ventilatiestand veranderen). Daarom is er een RF-gateway die zich gedraagt als draadloze afstandbediening. 

**- Kan ik alles uitlezen via een webinterface?** 
De hierboven genoemde waardes kun je uitlezen.

**Is het dan hetzelfde wat je ermee kunt als de techneuten/monteurs van Duco.** 
Nee, daarvoor kun je beter een losse usb naar serieel kabel gebruiken. Als die kabel een Silicon Labs CP210x chipset heeft werkt hij out of the box met de duco network tool. Dat is de software die installateurs gebruiken om de ducobox verder in te stellen. Het heeft weinig nut om die instellingen via de gateway beschikbaar te stellen omdat je ze daarna toch nooit meer aanpast.

broekieman, 2019-09-20:
Kan ik dan de waardes via een ipadres uitlezen? Heb je nog een gateway liggen voor me? Graag je gegevens voor betalen. Met vriendelijke groet,Maarten Broekman-------- Oorspronkelijk bericht --------Onderwerp: Re: [arnemauer/Ducobox-ESPEasy-Plugin] Duco laten werken met domoticz  (#19)Van: arnemauer Aan: arnemauer/Ducobox-ESPEasy-Plugin Cc: broekieman ,Author - Zijn er 2 soorten hardware of zitten zowel seriële en rf in het kastje?
Beide zitten in de gateway. De seriële gateway werkt via een kabel die je aansluit op de ducobox. Je kan daarmee de ventilatiestand, ventilatiepercentage, co2 PPM en co2 temperatuur. Via de seriele kabel kan je echter niet ducobox aansturen (ventilatiestand veranderen). Daarom is er een RF-gateway die zich gedraagt als draadloze afstandbediening.
- Kan ik alles uitlezen via een webinterface?
De hierboven genoemde waardes kun je uitlezen.
Is het dan hetzelfde wat je ermee kunt als de techneuten/monteurs van Duco. 
Nee, daarvoor kun je beter een losse usb naar serieel kabel gebruiken. Als die kabel een Silicon Labs CP210x chipset heeft werkt hij out of the box met de duco network tool. Dat is de software die installateurs gebruiken om de ducobox verder in te stellen. Het heeft weinig nut om die instellingen via de gateway beschikbaar te stellen omdat je ze daarna toch nooit meer aanpast.

broekieman, 2019-09-24:
Ik probeer het nog maar een keer. Je schreef me dat ik een gateway bij je kan kopen. Ik wil dit graag maar ondanks meerdere pogingen om dit te regelen, krijg ik geen enkele reactie. Ja, ik weet dat je geen helpdesk bent maar ik mag op zijn minst toch wel een reactie verwachten als ik je aanbod geaccepteerd heb.

arnemauer, 2019-09-25:
> Kan ik dan de waardes via een ipadres uitlezen? Heb je nog een gateway liggen voor me? Graag je gegevens voor betalen. Met vriendelijke groet,Maarten Broekman

Ja, zie https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/Data-uit-Ducobox

> Ik probeer het nog maar een keer. Je schreef me dat ik een gateway bij je kan kopen. Ik wil dit graag maar ondanks meerdere pogingen om dit te regelen, krijg ik geen enkele reactie. Ja, ik weet dat [...]

Ik vind het jammer dat ik niet aan je verwachting voldaan heb. We hebben het hier echter wel over een paar dagen. In Nederland zijn we helaas verwend met de bol.com 'voor 23.59 uur besteld morgen in huis' verzending. In andere landen zie je dat nauwelijks terug. Dit project is een hobbyproject en mijn gezin gaat voor op alles. Daarom duurt het soms een paar dagen voordat je een reactie van mij krijgt.

Maargoed, mocht snelle installatie en stabiliteit noodzakelijk zijn kan ik je de communication print van Ducobox zelf aanraden: https://www.duco.eu/nl-nl-duco-ventilation-app. Zij kunnen deze voor je installeren en kunnen je ook de verwachte support bieden.

Mocht je alsnog geïnteresseerd zijn in een gateway raad ik je aan om alle pagina's van de de wiki eens door te lezen: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki

closed 2019-09-25 by arnemauer