arnemauer/Ducobox-ESPEasy-Plugin #20: Stability of your plugin vs the Itho plugin from jodur?

svollebregt, 2020-01-26:
I was wondering if you could comment on the stability of your plugin with respect to the plugin originally developed by Jodur and updated by Klockie86 (https://github.com/klockie86/ESPEasy/) and me to fix some issues with remotes not being detected and compatibility with more recent mega builds.

What I notice is that the Itho plugin causes frequent reboots and I am trying to figure out what the origin for these reboots is. With the latest mega-20191208 the plugin always crashes when a command is sent. On mega-20190928 it can run either for a day or just a few hours. The source of the reboots changes over time.

I am trying to figure out if it is related to the interaction between the ESPEasy core and the plugin, the plugin itself, or the CC1101 libraries. ESPEasy has had its own instability issues in the past, but with the plugin disabled my local mega-20190928 built appears to run rock-stable. It could still be related to the ESP8266 Arduino core, though.

As you made some considerable changes to the CC1101 Itho/Duco packet systems and the plugins it would be very interesting to hear your experience.

arnemauer, 2020-01-26:
Hi there,

That sounds familiar to me. The plugin uses Ticker to start receiving data after an interrupt from the CC1101. Interrupts are special kinds of functions that have some unique limitations most other functions do not have. Interrupt handlers should be as short as possible so it isn't advised to do IO operations in an interrupt. The same applies to Ticker interrupts. 

The arduino manual states that:
"It is currently not recommended to do blocking IO operations (network, serial, file) from Ticker callback functions. Instead, set a flag inside the ticker callback and check for that flag inside the loop function."

I'm currently rewriting my plugin to set a flag in the interrupt en check in 'PLUGIN_FIFTY_PER_SECOND' if de flag is set. This solution has a side effect.... It may take some extra seconds before the esp reads the RX of the CC1101 and that can lead to RX buffer overflows. I am now making some changes to the CC1101 library to better handle this.

svollebregt, 2020-01-27:
Ok, thanks! Interrupts I am familiar with, Ticker is new to me. What is the purpose of calling a Ticker with an interrupt? Wouldn't it be possible to directly act on the interrupt or is it used to read-out the RX buffer with a certain interval?

I notice that in the Itho plugin besides setting some states there is also some log activity. As this can cause IO action I might experiment with removing this, or adding a button to specifically enable logging when looking for remote IDs. There is quite some chatter being picked up by my CC1101, which could cause a lot of log writing actions.

svollebregt, 2020-02-05:
I noticed you got rid of the Ticker and are using a flag set by the ISR in your latest code, followed by actual message decoding in the 50 p.s. routine. I like this idea! Ticker appears to a very likely source for instability as if receiving is disabled the Itho plugin appears to be quite stable. How is this working for your Duco plugin so far? No message drops?

closed 2020-02-23 by svollebregt