arnemauer/Ducobox-ESPEasy-Plugin #22: Porting your  P150 plugin to Itho?

svollebregt, 2020-02-23:
Would it be doable to port your very nice work on the DucoCC1101 library back to the Itho library? I've been looking at improving the plugin made by Jodur taking inspiration form your code in order to improve stability (as functionally the code works). The culprit is as you mentioned the Ticker in the RX side, as disabling RX makes the code run stable. I implemented a simple way of using the 50 p.s. plugin call to use this instead of a Ticker. While it now 'works' on my side, there are some odd things happening which for now are quite puzzling too me: https://gathering.tweakers.net/forum/list_message/61511044

My C++ knowledge is quite rusty as I haven't used it for more than a decade, but I am trying to figure out how the improvements you made to the library work. I noticed you use a better way to filter out correct packets and assert the interrupt from the CC1101 module. It doesn't help that at the moment it is not clear to me yet what the differences between the Duco and Itho protocols are.

How realistic is implementing your changes into the original Itho code from supersjimmie and the plugin made by Jodur? Do you perhaps have some additional documentation on the code and protocol workings that could aid in achieving this?

Dave7NL, 2020-10-26:
Would love to know this as well - then i can go ahead and buy the hardware. I have 2 ITHO WTW boxes at home that i would love to  be able to control.

arnemauer, 2022-04-06:
Ik heb er zelf helaas geen tijd voor. Wellicht dat iemand anders dit op wil pakken :)