arnemauer/Ducobox-ESPEasy-Plugin #23: Unreliable communication when gateway is close to ventilationbox

arnemauer, 2020-04-18:
After moving the gateway closer to the ventilationbox the communication between the two gets disturbed. However, another gateway, a few meters further, did receive the messages from the ventilationbox and the gateway. 

If two transceivers are too close to each other and TX power is at maximum, the signal is just too strong to be received correctly. Reducing TX power or increasing the distance (about half a meter should be fine) will solve this issue. Moving the gateway further away from the ventilation box isn't possible because of the serial cable. 

The solution is to lower the TX power of the gateway. The current PAtable value is 0xC5  8,9 power @ 868MHz [dBm]. 

Idea: add a checkbox to change the value from 0xC5 to 0x50 (-0,3).

!(patable)[https://raw.githubusercontent.com/arnemauer/Ducobox-ESPEasy-Plugin/master/Screenshots/patable%20CC1101%20868mhz.png]

IgorYbema, 2020-05-23:
Same here. With the serial cable connected i'm too close to the ducobox. Need to unscrew the antenna (which is not good!) to make a proper connection.

arnemauer, 2020-05-23:
Aangezien Duco voornamelijk in NL en BE levert kunnen we prima overgaan op het Nederlands :) 

Ik zal vanavond een nieuwe firmware op github zetten waar de zendsterkte instelbaar is.

IgorYbema, 2020-05-29:
Lukt het met deze aanpassing?

arnemauer, 2020-05-29:
Ik heb de aanpassing klaar. Ik heb in de plugin een dropdown menu toegevoegd waar je uit vier zendsterktes kan kiezen. Morgen nog even testen en dan zet ik hem op github.
```

		case P150_RADIO_POWER_DEFAULT: return F("Default - radio power 0,6 dBm (0x8D)");
		case P150_RADIO_POWER_1:  return F("Radio power 10,3 dBm (0xC1)");
		case P150_RADIO_POWER_2:  return F("Radio power 5,0 dBm (0x81)");
		case P150_RADIO_POWER_3:  return F("Radio power -5,0 dBm (0x67)");
		case P150_RADIO_POWER_4:  return F("Radio power -20 dBm (0x0F) - lowest");
```

IgorYbema, 2020-05-30:
Top! En als ik ergens mee kan helpen zoals bijv. testen roep maar :)

arnemauer, 2020-06-01:
Ik heb de mogelijkheid toegevoegd om zelf de zendsterkte van de CC1101 te bepalen (2c20f83). De standaard waarde staat wat lager zodat gebruikers minder snel problemen ervaren. Helaas lost dit maar een deel van het probleem op. De CC1101 heeft wel de mogelijkheid om de zendsterkte aan te passen maar niet de sensitiviteit van de ontvangen. De gateway moet daarom minimaal een meter uit de buurt van de Ducobox hangen. 

Een langere kabel zal het probleem oplossen maar creëert ook een ander probleem. Het 3.3 volt seriële signaal is namelijk niet gemaakt om over een lange kabel te werken. Hoe langer de kabel hoe slechter het signaal wordt en hoe meer kans op storingen door externe invloeden. Ik heb een aantal verlengkabels besteld om te kijken of het signaal sterk genoeg is om 3 meter te overbruggen.

Mocht dat niet werken dan moeten we het signaal om gaan zetten zodat hij wel de afstand kan overbruggen. Een van de opties is het gebruik van RS-232. Daarvoor moet er in zowel de gateway als in de Ducobox een signal converter (bijvoorbeeld MAX232) die het signaal omzet naar RS-232. Geen ideale situatie en kostenverhogend.

[image]

samvdb, 2020-06-02:
Ik loop zit met hetzelfde probleem denk ik. De gateway stuurt join command, de duco network tool ziet de RF device maar blijft op communication failure staan. Zonder antenne en/of duco minder RF sterkte geven en de nieuwe firmware met laagste RF strength helpen niet.

Hopelijk komt er een oplossing.

IgorYbema, 2020-06-02:
@samvdb zou kunnen dat je een ander probleem hebt, zonder antenna werktte het bij mij direct

Voor mij is met nieuwe firmware het probleem opgelost met de nu default (lagere) waarde.

arnemauer, 2020-06-02:
@samvdb  Ik zie in de logs die je per e-mail hebt gestuurd dat je gateway wel goed is gekoppeld. Waarschijnlijk is het adres niet goed opgeslagen (dat was een bugje en is in de nieuwe firmware opgelost). Zie xxxx.

Voor anderen met hetzelfde probleem: zou je de seriële kabel uit de gateway willen halen (altijd de Ducobox eerst uitschakelen!) en de gateway op een paar meter afstand in een usbadapter willen aanzetten? Daarna kun je het join commando geven. Als het join commando dan succesvol is uitgevoerd dan ligt het aan een te korte kabel.

arnemauer, 2020-07-02:
Opgelost in 2c20f83c2dc4d9f404276a73b20ac85003bcbb98

closed 2020-07-02 by arnemauer