arnemauer/Ducobox-ESPEasy-Plugin #24: Home-Assistant integration

arcangel321, 2020-04-26:
Reading CO2:
Add below to Configuration.yaml
sensor:
  - platform: mqtt
    state_topic: "VENTILATION_GATEWAY/serialgw_CO2/CO2_PPM"
    unit_of_measurement: 'PPM'
    name: "CO2_PPM"
Set FanSpeed:
Add below to Configuration.yaml
switch:
  - platform: mqtt
    name: "Ventilatie Hoog"
    state_topic: "VENTILATION_GATEWAY/Ventilation_gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,HIGH,0"
    payload_off: "VENTMODE,AUTO,0"
    state_on: "3"
    state_off: "0"
    optimistic: false
    qos: 0
    retain: true

Jandokus, 2021-04-11:
Indien je de RF Gateway met de erbij geleverde RJ11 kabel met de Serial Gateway hebt verbonden aan je DucoBox en je wilt via Home Assistant (HA) de DucoBox Ventilator bedienen en/of de waardes in je HA tonen dan dien je het volgende te doen (ik ben er vanuit gegaan dat je MQTT al hebt draaien):

- Maak een Controller ‘Home Assistant (openHAB) MQTT’ aan met de ‘Add’ button op het Controllers tabblad en geef vervolgens het IP adres op van je MQTT server (en evt. de user en password indien je je MQTT beveiligd hebt) en zet een vinkje achter ‘Enabled:’;
- Zet bij elk van de devices op het tabblad ‘Devices’ een vinkje achter de ‘Send to Controller’ om de waarde naar HA te verzenden;
- Zet in ieder geval bij het device ‘DUCO Serial Gateway’ dat de Ventilation mode naar HA verzend, het aantal decimals op ‘0’.

Voor het vervolg ga ik uit van de volgende situatie:

[image]

Vervolgens de stap naar HA. Hiervoor dien je alleen wat sensoren en switches aan te maken in het bestand ‘Configuration.yaml’ (óf voor degene die een apart bestand hebben voor het aanmaken van de sensoren dienen het bestand ‘sensor.yaml’ aan te passen). Met dank aan @arcangel321

switch:
  - platform: mqtt
    name: "Duco Afzuiging Hoog (15 min)"
    unique_id: duco_afzuiging_hoog
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,HIGH,0"
    payload_off: "VENTMODE,AUTO,0"
    state_on: "3"
    state_off: "0"
    optimistic: false
    qos: 0
    retain: true

  - platform: mqtt
    name: "Duco Afzuiging Middel (15 min)"
    unique_id: duco_afzuiging_middel
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,MIDDLE,0"
    payload_off: "VENTMODE,AUTO,0"
    state_on: "2"
    state_off: "0"
    optimistic: false
    qos: 0
    retain: true

  - platform: mqtt
    name: "Duco Afzuiging Laag (15 min)"
    unique_id: duco_afzuiging_laag
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,LOW,0"
    payload_off: "VENTMODE,AUTO,0"
    state_on: "1"
    state_off: "0"
    optimistic: false
    qos: 0
    retain: true

sensor:
  - platform: mqtt
    name: Duco Ventilatie Status
    unique_id: duco_ventilatie_status
    state_topic: "VENTILATION_GATEWAY/SERGW_Ventilation_Mode/VentilationMode"
    value_template: >-
      {% set status = value | int %}
      {% if status == 13 %}
        {{'Perm. Hoog'}}
      {% elif status == 12 %}
        {{'Perm. Middel'}}
      {% elif status == 11 %}
        {{'Perm. Laag'}}
      {% elif status == 3 %}
        {{'Hoog'}}
      {% elif status == 2 %}
        {{'Middel'}}
      {% elif status == 1 %}
        {{'Laag'}}
      {% elif status == 0 %}
        {{'Auto'}}
      {% else %}
        {{'Undefined'}}
      {% endif %}

  - platform: mqtt
    name: Duco Ventilatie Percentage
    unique_id: duco_ventilatie_percentage
    state_topic: "VENTILATION_GATEWAY/SERGW_Ventilation_Percentage/VentilationPercentage"
    unit_of_measurement: "%"

  - platform: mqtt
    name: Duco Ventilatorsnelheid
    unique_id: duco_ventilatorsnelheid
    state_topic: "VENTILATION_GATEWAY/SERGW_Current_Fan_Speed/FanSpeed"
    unit_of_measurement: "rpm"

  - platform: mqtt
    name: Duco Afteltijd
    unique_id: duco_afteltijd
    state_topic: "VENTILATION_GATEWAY/SERGW_Countdown/Countdown"
    unit_of_measurement: "s"

Hopelijk helpt bovenstaande bij de integratie van de Ventilation Gateway in HA! Je kan de switches natuurlijk nog uitbreiden om de ventilator permanent op ‘Hoog’, ‘Middel’ of ‘Laag’ te zetten.
Zelf heb ik geen externe sensoren aan mijn DucoBox hangen. Om die in HA te kunnen uitlezen, hoop ik dat je met bovenstaande voldoende kennis hebt om dat zelf te doen.

In onderstaande afbeelding is het resultaat van mijn HA te zien:

[image]

Trigun, 2021-05-06:
@Jandokus, ik heb in HA inderdaad dezelfde informatie echter merk ik dat bij een switch(bijv. Hoog) deze aan gaat maar ook direct weer uit(zoals een pushbutton) het mooiste zou zijn als dit op aan bleef staan. wanneer je dan een andere stand kiest(bijv. middel) dat dan stand Hoog uitgaat en stand Middel aan.

zou dit een mogelijkheid zijn?

Jandokus, 2021-05-06:
@Trigun, dat klopt. Ook bij mij gaat de switch direct weer terug naar zijn uitgangspositie. Feitelijk zou je willen dat de switch pas weer terug gaat, na de ingestelde periode (van default 15 minuten) óf, zoals jij terecht zegt, dat in de tussentijd een andere switch wordt bediend. Het wordt nog ingewikkelder als je ook de switches met de permanente toestanden laat aanmaken in HA, zoals hieronder:

  - platform: mqtt
    name: "Duco Afzuiging Auto (Permanent)"
    unique_id: duco_afzuiging_auto_perm
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,AUTO,0"
    payload_off: "VENTMODE,AUTO,0"
    state_on: "0"
    state_off: "0"
    optimistic: false
    qos: 0
    retain: true

  - platform: mqtt
    name: "Duco Afzuiging Hoog (Permanent)"
    unique_id: duco_afzuiging_hoog_perm
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,PERMANENTHIGH,0"
    payload_off: "VENTMODE,AUTO,0"
    state_on: "13"
    state_off: "0"
    optimistic: false
    qos: 0
    retain: true

  - platform: mqtt
    name: "Duco Afzuiging Midden (Permanent)"
    unique_id: duco_afzuiging_middle_perm
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,PERMANENTMIDDLE,0"
    payload_off: "VENTMODE,AUTO,0"
    state_on: "12"
    state_off: "0"
    optimistic: false
    qos: 0
    retain: true

  - platform: mqtt
    name: "Duco Afzuiging Laag (Permanent)"
    unique_id: duco_afzuiging_laag_perm
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,PERMANENTLOW,0"
    payload_off: "VENTMODE,AUTO,0"
    state_on: "11"
    state_off: "0"
    optimistic: false
    qos: 0
    retain: true

Mogelijk bepaald de 'optimistic' instelling dit mechanisme, maar dat weet ik niet zeker omdat ik dat deel van @arcangel321 heb overgenomen.

Gibletron, 2021-05-06:
Ik maak hier gebruik van een value_template:
value_template: "{{ '100' if value == '0' else '99' }}"
Value_template pakt de waarde uit het state_topic.
Controlleert of het gelijk is aan een specifieke waarde (in dit geval 0) en zet zijn output vervolgens.
Als de waarde van state_topic gelijk is aan 0, dan zet hij zijn output op 100
Als de waarde van state_topic niet gelijk is aan 0 dan zet hij zijn output op 99
Vervolgens gebruik ik state_on: "100" en state_off: "99" om de switch aan/uit weer te geven.

Dit werkt erg goed en zorgt dat alleen de switch aan staat van welke de mode ook daadwerkelijk aan staat.
Er zit wel een kleine vertraging in, dit heeft er mee te maken dat wanneer je een switch omzet hij eerst bericht terug moet krijgen van de ducobox dat die mode ook daadwerkerlijk actief is geworden, dus dat kan een seconde of twee duren.
Moor mij werkt dit echt uitstekend en absoluut niet storend.

-  platform: mqtt
    name: "Duco Afzuiging Auto (Permanent)"
    unique_id: duco_afzuiging_auto_perm
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,AUTO,0"
    value_template: "{{ '100' if value == '0' else '99' }}"
    state_on: "100"
    state_off: "99"
    optimistic: false
    qos: 1
    retain: true

 - platform: mqtt
    name: "Duco Afzuiging Hoog (Permanent)"
    unique_id: duco_afzuiging_hoog_perm
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,PERMANENTHIGH,0"
    payload_off: "VENTMODE,AUTO,0"
    value_template: "{{ '100' if value == '13' else '99' }}"
    state_on: "100"
    state_off: "99"
    optimistic: false
    qos: 1
    retain: true

 - platform: mqtt
    name: "Duco Afzuiging Midden (Permanent)"
    unique_id: duco_afzuiging_middle_perm
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,PERMANENTMIDDLE,0"
    payload_off: "VENTMODE,AUTO,0"
    value_template: "{{ '100' if value == '12' else '99' }}"
    state_on: "100"
    state_off: "99"
    optimistic: false
    qos: 1
    retain: true

 - platform: mqtt
    name: "Duco Afzuiging Laag (Permanent)"
    unique_id: duco_afzuiging_laag_perm
    icon: hass:fan
    state_topic: "VENTILATION_GATEWAY/Ventilation_Gateway_RF/Ventilationmode"
    command_topic: "VENTILATION_GATEWAY/cmd"
    payload_on: "VENTMODE,PERMANENTLOW,0"
    payload_off: "VENTMODE,AUTO,0"
    value_template: "{{ '100' if value == '11' else '99' }}"
    state_on: "100"
    state_off: "99"
    optimistic: false
    qos: 1
    retain: true

Theoreet, 2021-05-27:
> Indien je de RF Gateway met de erbij geleverde RJ11 kabel met de Serial Gateway hebt verbonden aan je DucoBox en je wilt via Home Assistant (HA) de DucoBox Ventilator bedienen en/of de waardes in je [...]

in mijn ESPEasy zie ik die PERM High , PERM Mid enz niet terug ? is dit een aangepaste config / plugin ?

[image]

Jandokus, 2021-05-27:
@Theoreet, ik heb sinds begin maart 2021 de RF-Gateway en mogelijk daardoor ook recente ESP Easy software, maar ik weet niet of het daarmee te maken heeft dat ik wel Perm. HIGH, Perm. MID, enz. zie.
Dit zijn mijn firmware gegevens (zie Main tabblad):

Misschien dat je je vraag aan @arnemauer kan stellen.

Grtn,
Jan

Theoreet, 2021-05-27:
> @Theoreet, ik heb sinds begin maart 2021 de RF-Gateway en mogelijk daardoor ook recente ESP Easy software, maar ik weet niet of het daarmee te maken heeft dat ik wel Perm. HIGH, Perm. MID, enz. zie. [...]

ik had deze firmware inmiddels ook gevonden, Is nog niet de nieuwste versie, maar nadat ik deze versie geflash heb zie ik de high enz wel . THNX

arnemauer, 2021-05-27:
Ik heb hier een begin gemaakt met een stappenplan: 
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/Home-Assistant

Ik heb zelf geen HA dus heb wat geknipt en geplakt :) Graag hoor ik jullie wijzigingen/aanvullingen zodat we hem compleet kunnen maken.

Jandokus, 2021-05-27:
Stap 1 heb ik van commentaar voorzien (zie pdf). Op de 2 t/m 4 kom ik nog terug.

Jandokus, 2021-05-27:
Stap 1 heb ik van commentaar voorzien (zie onderstaande pdf). Op de 2 t/m 4 kom ik nog terug.

[Home Assistant 2.pdf](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/6556735/Home.Assistant.2.pdf)

Jandokus, 2021-05-27:
Zonet ook stap 2 en 3 van commentaar voorzien. Zie hieronder:

[Home Assistant 3.pdf](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/6557041/Home.Assistant.3.pdf)

Theoreet, 2021-05-28:
goed bezig mannen ! Nu de port naar ESPHome nog :D Dat zal helemaal TOP
zijn 😂

Theoreet, 2021-06-02:
> * platform: mqtt
>   name: Duco Ventilatorsnelheid
>   unique_id: duco_ventilatorsnelheid
>   state_topic: "VENTILATION_GATEWAY/SERGW_Current_Fan_Speed/FanSpeed"
>   unit_of_measurement: "rpm"
> * platform: [...]

Ben er bijna, maar krijg het aftellen nog niet voor elkaar, heb alles ingesteld en dus voor 99% werkende, maar hoe heeft u het aftellen geregeld (is hiervoor een losse sensor bijv ) ?

Gibletron, 2021-06-02:
> Ben er bijna, maar krijg het aftellen nog niet voor elkaar, heb alles ingesteld en dus voor 99% werkende, maar hoe heeft u het aftellen geregeld (is hiervoor een losse sensor bijv ) ?

[image]

- platform: mqtt
  unique_id: duco_vent_countd
  name: Duco Ventilation Countdown
  state_topic: "VENTILATION_GATEWAY/SERGW_COUNTDOWN/Ventilationcountdown"
  unit_of_measurement: "s" 
  icon: mdi:timer-outline

Theoreet, 2021-06-02:
Duidelijk, maar dan de vervolg vraag waarvoor zetten jullie dit in ? Als ik
je tegoed begrijp is in uw screenshot de aftel tijd dus 60seconden

Gibletron, 2021-06-02:
> 
> Duidelijk, maar dan de vervolg vraag waarvoor zetten jullie dit in ? Als ik je tegoed begrijp is in uw screenshot de aftel tijd dus 60seconden Op wo 2 jun. 2021 16:58 schreef Gibletron ***@***.***>:
> […](#)

Die 60 seconde is hoe vaak de sensor informatie wordt doorgestuurd naar de controller (oftewel naar MQTT)
Die countdown staat ingesteld in de ducobox network tool software en is standaard 15 Minuten

Gibletron, 2021-06-02:
> 
> […](#)

De aftelling wordt getriggered door alle niet permanente standen
Als het goed is kent je Ducobox 7 standen:
- Low
- Med
- High
- Auto
- Perm Low
- Perm Med
- Perm High
Alleen de eerste drie (tijdelijke) standen starten een countdown, de overige blijven zoals de naam al zegt permanent aanstaan

Theoreet, 2021-06-04:
> het aftellen geregeld (is hiervoor een losse sensor bijv ) ? [image:

Inmiddels alles werkend gekregen. Nu alleen het probleem dat ik na een dag
niet meer terecht kan op de ESPEASY website, maar krijg wel reply op IP
adres (ping) . Nu kreeg ik net de melding : low memory cannot display
webpage ?
Hebben er hier meer last van, of is er een tip met wat ik in moet stellen /
aan moet passen ?

Gibletron, 2021-06-04:
> Inmiddels alles werkend gekregen. Nu alleen het probleem dat ik na een dag niet meer terecht kan op de ESPEASY website, maar krijg wel reply op IP adres (ping) . Nu kreeg ik net de melding : low [...]

Had hier zelf ook last van, doe nu elke nacht een automatische reboot van de ESPEASY
Eerst de Rule engine aanzetten:
In ESPEasy naar Tools -> Advanced ->Vinkje zetten bij Rules (en ik heb hem ook staan bij Old Engine dus misschien die ook?)
Vervolgens naar het rules tabblad (als die niet zichtbaar is even de ESPEasy rebooten)
Daar een ruleset toevoegen met de volgende:

```On Clock#Time=All,01:10 do
  delay 65000 // Needed else it will reboot many times at 01:10
  reboot // reboot only this module
endon 
```

Basically wacht tot 1:10 dan nog even 65s (anders gaat hij om 1:10 herstarten, is hij binnen enkele seconde terug, is het nogsteeds 1:10 en gaat hij nog een keer herstarten)
En dan reboot...

arnemauer, 2021-06-04:
Wat is de interval in seconden die je bij de plugins hebt ingesteld?

Gibletron, 2021-06-04:
VENTILATION_GATEWAY  60s
SERGW_VENTILATION_MODE 1s   -   Dit is zodat de sensoren in HA vrijwel instant updaten
SERGW_VETILATION_PERCENTAGE 60s
SERGW_CUR_FAN_VALUE 60s
SERGW_COUNTDOWN 60s
SERGW_FANSPEED 60s
SERGW_CO2_PPM 60s
SERGW_CO2_TEMP 60s
SERGW_TEMP 60s

Dus basically.. alles op 60s behalve de check voor de huidige ventilatiemode die staat op elke 1s zodat de knop direct feedback geeft in de HA dashboard

arnemauer, 2021-06-04:
> VENTILATION_GATEWAY 60s
> SERGW_VENTILATION_MODE 1s - Dit is zodat de sensoren in HA vrijwel instant updaten
> SERGW_VETILATION_PERCENTAGE 60s
> SERGW_CUR_FAN_VALUE 60s
> SERGW_COUNTDOWN 60s
> SERGW_FANSPEED [...]

Als je hebt op 1 seconde zet zal hij iedere seconde data opvragen van de Ducobox. Daardoor blijft er weinig tijd over voor overige taken zoals de webinterface. Het zou zelfs zo kunnen zijn dat de andere plugins geen data opvragen omdat de seriële plugin de "lijn bezet houd). 

Zou je hem eens op 60 sec. willen zetten en dan kijken of je dezelfde foutmelding krijgt? Je kan overigens het beste de rf gateway gebruiken voor de actuele ventilatiestand. Die pushed de data richting HA op het moment dat de ventilatiestand  veranderd.

Gibletron, 2021-06-04:
> Als je hebt op 1 seconde zet zal hij iedere seconde data opvragen van de Ducobox. Daardoor blijft er weinig tijd over voor overige taken zoals de webinterface. Het zou zelfs zo kunnen zijn dat de [...]

Dat is goed te weten, dan moet ik even gaan uitzoeken welk MQTT topic /payload ik daarvoor moet hebben

Theoreet, 2021-06-05:
Top, thnx voor de heldere uitleg

Gibletron, 2021-06-09:
En nog even de terugkoppeling: na enkele dagen zonder reboot geen probleem bemerkt, dus voor mij blijft de rule uit staan
Mocht ik er toch weer tegenaan lopen dan zet ik de rule weer aan, maar lijkt nu niet nodig

Theoreet, 2021-06-28:
ik heb nu de reboot rule erin staan, maar de gateway is toch niet meer
benaderbaar . Dus ondanks de dagelijkse reboot alsnog ineens niet te
benaderen .
Misschien toch teveel

Theoreet, 2021-07-12:
> ik heb nu de reboot rule erin staan, maar de gateway is toch niet meer benaderbaar . Dus ondanks de dagelijkse reboot alsnog ineens niet te benaderen . Misschien toch teveel Op wo 9 jun. 2021 om 22:04 schreef Gibletron ***@***.***>:
> […](#)

Helaas blijkt dat alleen de reboot rule niet voldoende is . Moet met regelmaat de Wemos "met de hand" stroomvrij maken en zo resetten. De Wemos is geeft op dat moment wel gewoon reply op een ping request maar MQTT is verbroken en ESPHome site niet meer beschikaar. Zijn er toevallig al gebruikers die een nieuwere ESPHome build "ge-compiled" hebben ?

bartlauwers53, 2021-08-08:
Beste allemaal, Ik ben heel blij met dit draadje. Hiermee de aansturing van de box in HA aan de praat gekregen, tevens het uitlezen van diverse sensoren. 
Ik loop alleen tegen één ding aan bij het toevoegen van de config van @Jandokus in deze post: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/24#issuecomment-817308542
Uiteraard heb ik die aangepast naar mijn config zodat het klopt voor mijn setup, maar één ding blijft een fout geven:
de eerste regel onder value_template ( {% set status = value | int %} ) geeft de melding "missed comma between flow collection entries at line x, column y:
Ik zie alleen niet waar ik de mist in ga. Ook al een uurtje Google tegenaan gegooid, maar geen succes tot nu toe. Ik ben nog steeds echt een HomeAssistant Noob, dus hopelijk kunnen jullie mij helpen? :) Alvast dank!

Gibletron, 2021-08-08:
> 
> 
> Beste allemaal, Ik ben heel blij met dit draadje. Hiermee de aansturing van de box in HA aan de praat gekregen, tevens het uitlezen van diverse sensoren.
> Ik loop alleen tegen één ding aan bij het toevoegen van de config van @Jandokus in deze post: [#24 (comment)](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/24#issuecomment-817308542)
> Uiteraard heb ik die aangepast naar mijn config zodat het klopt voor mijn setup, maar één ding blijft een fout geven:
> de eerste regel onder value_template ( {% set status = value | int %} ) geeft de melding "missed comma between flow collection entries at line x, column y:
> Ik zie alleen niet waar ik de mist in ga. Ook al een uurtje Google tegenaan gegooid, maar geen succes tot nu toe. Ik ben nog steeds echt een HomeAssistant Noob, dus hopelijk kunnen jullie mij helpen? :) Alvast dank!

Ik gebruik dezelfde config als Jandokus en krijg daarbij geen foutmelding
Wel even een belangrijke sidenote: de foutmelding hoeft niet perse te slaan op de regel die je noemt, maar kan in principe over het hele "value_template" gaan, dus ergens in je value template heb je een foutje staan of iets vergeten
Controlleer regel voor regel de syntax, alle hoge komma's die gebruikt worden zijn enkele hoge komma's > ' <
Er worden nergens ticks gebruikt > ` < 
Nog even mijn exacte value_template:
```
  value_template: >-
    {% set status = value | int %}
    {% if status == 13 %}
    {{'Perm. High'}}
    {% elif status == 12 %}
    {{'Perm. Mid'}}
    {% elif status == 11 %}
    {{'Perm. Low'}}
    {% elif status == 3 %}
    {{'High'}}
    {% elif status == 2 %}
    {{'Mid'}}
    {% elif status == 1 %}
    {{'Low'}}
    {% elif status == 0 %}
    {{'Auto'}}
    {% else %}
    {{'Undefined'}}
    {% endif %}
```

bartlauwers53, 2021-08-08:
Thanks @Gibletron voor de hele snelle reactie, ik denk een indentation issue. Blijkbaar is dat heel belangrijk in YAML. Thanks a lot!

Losstaand van bovenstaand issue:
Verder nog een tip omtrent de status: bij de Duco Ventilation Remote kun je onderaan een "value" opgeven die je moet uitlezen voor de status. Deze zie je niet bij je devices overzicht terug en kan afwijken van wat je daar ziet. (ik had "Ventmode" voor de ventilationmode gebruikt... Maar nu werkt het helemaal :D

BrendanxP, 2021-09-01:
Wij gaan binnenkort een nieuwe ducobox aanschaffen. Hebben jullie nog tips om op te letten bij de aanschaf om hem zo makkelijk mogelijk te integreren in HA? Ik run op dit moment HassOS op een RPi4 met Zigbee USB stick op ZHA.

rhvs, 2021-10-04:
Mijn sensor voor ventilatie percentage doet het prima, maar de modus absoluut niet.... zie ik nu iets over het hoofd?

`sensor:

  - platform: mqtt
    name: Duco Ventilatie Status
    unique_id: duco_ventilatie_status
    state_topic: "VENTILATION_GATEWAY_1/SERGW_Ventilation_Mode/Ventmode"
    value_template: >-
      {% set status = value | int %}
      {% if status == 13 %}
      {{'Perm. High'}}
      {% elif status == 12 %}
      {{'Perm. Mid'}}
      {% elif status == 11 %}
      {{'Perm. Low'}}
      {% elif status == 3 %}
      {{'High'}}
      {% elif status == 2 %}
      {{'Mid'}}
      {% elif status == 1 %}
      {{'Low'}}
      {% elif status == 0 %}
      {{'Auto'}}
      {% else %}
      {{'Meh'}}
      {% endif %}
`

bartlauwers53, 2021-10-04:
Ik vemoed aan de hand van de omschrijving in je code dat je hem uitleest bij "device" DUCO Serial Gateway(er staat SERGW_Ventilation_Mode)? Je kunt hem beter uitlezen van device "DUCO ventilation remote" want die pusht de waarde als de stand van de box verandert. Zie ook hierboven: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/24#issuecomment-854982882. 
Check ook bij de Ducobox dat de waarde die je onderaan de device pagina bij "value" opgeeft overeenkomt met de laatste waarde in je state_topic (hier "Ventmode"). Als dat anders is doet ie het ook niet, kan je uit eigen ervaring vertellen ;-).

rhvs, 2021-10-04:
Super! Percentage blijft hopeloos achter, maar de modus vind ik al leuk genoeg. Vooral blij met MQTT op de Ducobox en de link naar HomeKit.

broekieman, 2022-01-25:
Ik ben er ook aan begonnen nadat ik het goed werkend had in Domoticz.
Kan iemand mij een complete configuration.yaml sturen?
Ik krijg met copy paste allerlei foutmeldingen zoals deze: 
missed comma between flow collection entries at line 68, column 2:
    {% set status = value | int %}

Ik wil graag permanent erin hebben.

bartlauwers53, 2022-01-25:
Ik heb dit staan(maar ik weet voordat ik post eigenlijk niet of dit als file(bijlage) of in-line weergegeven wordt...): 
[conifguration.yaml-DUCO.txt](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/7933788/conifguration.yaml-DUCO.txt)

broekieman, 2022-01-25:
> Ik heb dit staan: [conifguration.yaml-DUCO.txt](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/7933788/conifguration.yaml-DUCO.txt) Ik heb alleen de relevante delen laten staan. Moet je zelf dus even onder het goede deel(sensor of switch) invoegen in je eigen configuration.yaml denk ik.

Dankjewel! Ik heb het er allemaal in staan, maar helaas krijg ik bij alle switches en sensoren status: "niet beschikbaar"
Mqtt is correct aangemeldt. 
Instellingen Gateway: https://user-images.githubusercontent.com/47141060/150993964-2eb84b83-46d6-4ded-ba2a-42873e635b47.png.
Instellingen configuration.yaml:
[conifguration.yaml-DUCO.txt](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/7934302/conifguration.yaml-DUCO.txt)

Wat zie ik over het hoofd?

Gibletron, 2022-01-25:
> 

Controleer of je de juiste state en command topics gebruikt
Simpel gezegd, de topics die je instelt in de ventilation gateway moeten overeenkomen met Home Assisstant
Je kunt een mqtt client zoals mqtt explorer gebruiken om te controleren of je de juiste state en command topics gebruikt

[image]

In mijn mqtt explorer is bijvoorbeeld te zien dat ik als Topics gebruik maak van "VENTILATION_GATEWAY/SERGW_VENTILATION_MODE/Ventilationmode" voor de ventilatiemode

In de config van bartlauwers53 zou ik dus onder sensor -> Duco Ventilatie Status het state_topic aanpassen naar:
VENTILATION_GATEWAY/SERGW_VENTILATION_MODE/Ventilationmode

De waardes hiervan kun je dus via mqtt explorer uitvinden, of je kunt ze herleiden uit de ventilation gateway interface
Loop alle command EN state topics na en pas ze aan waar nodig

bartlauwers53, 2022-01-25:
Wellicht ten overvloede voordat je het naar HomeAssistant probeert te krijgen, maar als ik je screenshot bekijk @broekieman , dan zie ik ook de eerste regel "ventmode" op "0" staan, terwijl dit "1" zou moeten zijn wanneer de lage tijdelijke modus aan staat. Dit zie ik terug bij de DUCO_VG_RF regel in je screenshot. 
Verder zou dan ook de countdown waarde ergens tussen 900 seconden (= 15 minuten) en 0 moeten aangeven, aangezien die ergens tussen de 0 en 15 minuten zou moeten aftellen. 
Klopt dat deel van je config wel? Of had je net ventmode 1 ingeschakeld en screenshot gemaakt, terwijl die waardes nog niet geüpdate waren? 
Verder geeft je CO2_PPM waarde normaal geen nul aan, ergens tussen 400 en 800 zou realistisch zijn. 
Als laatste opmerking (eerder hierboven ook al genoemd geloof ik [https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/24#issuecomment-854982882](url) in het laatste stukje): je kunt de waarde VentilationMode het snelste uitlezen van de "DUCO ventilation remote". Via "DUCO Serial Gateway" loopt ie achter en wordt bij aanpassing niet gepusht via MQTT.

broekieman, 2022-01-25:
> Controleer of je de juiste state en command topics gebruikt Simpel gezegd, de topics die je instelt in de ventilation gateway moeten overeenkomen met Home Assisstant Je kunt een mqtt client zoals [...]

Volgens mij heb ik alles overgenomen van bartlauwers53 en zou het moeten werken. Ik heb mqtt explorer gedownload, maar weet niet wat ik waar moet invullen helaas.

broekieman, 2022-01-25:
> Wellicht ten overvloede voordat je het naar HomeAssistant probeert te krijgen, maar als ik je screenshot bekijk @broekieman , dan zie ik ook de eerste regel "ventmode" op "0" staan, terwijl dit "1" [...]

Ik heb alles ingevuld volgens jouw voorbeel, zie de bijlages. De waardes waren niet geüpdate en de Co2 sensor heeft kuren. Ik moet de Duco Box nog uitlezen met de netwerktool.

Gibletron, 2022-01-25:
> > Volgens mij heb ik alles overgenomen van bartlauwers53 en zou het moeten werken. Ik heb mqtt explorer gedownload, maar weet niet wat ik waar moet invullen helaas.
> 
> Het is vooral niet zeker dat het moet werken als je het van mij overneemt ;-) . Je moet het zeer waarschijnlijk op jouw situatie aanpassen. In MQTT explorer vul je je MQTT broker in (ik neem dat je die in HA draait? vaak ip-adres van je HA) en tevens de bijbehorende poort, en user-password. Name is iets wat je zelf kiest.

Je kunt exact dezelfde MQTT instellingen gebruiken die je in de ventilation gateway gebruikt om met HA te verbinden

broekieman, 2022-01-25:
Vreemd dat er zowel een VENTILATION_GATEWAY als een VENTILATION_GATEWAY_1 aanwezig is. Ik heb in de configuration.yaml alles aangepast naar VENTILATION_GATEWAY.
Kan het iets te maken hebben met Hoofletters en kleine letters?

Gibletron, 2022-01-25:
> Vreemd dat er zowel een VENTILATION_GATEWAY als een VENTILATION_GATEWAY_1 en ook nog een Duco_Ventilation aanwezig is. Ik heb in de configuration.yaml alles aangepast naar VENTILATION_GATEWAY. Kan het iets te maken hebben met Hoofletters en kleine letters [image] ?

Dat is inderdaad wel vreemd vooral omdat ze beiden dezelfde "status" terugsturen
Ik zou de Ventilation Gateway een keer een reboot geven en kijken wat er daarna gebeurt

broekieman, 2022-01-25:
Zou je me een afbeelding van je geconfigureerde Mqtt Controller willen sturen?

Gibletron, 2022-01-25:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

sensors:
```
- platform: mqtt
  unique_id: duco_gateway_status
  name: Duco Gateway Status
  state_topic: "VENTILATION_GATEWAY/status/LWT"
- platform: mqtt
  unique_id: duco_vent_mode
  name: Duco Ventilation Mode
  state_topic: "VENTILATION_GATEWAY/VENTILATION_GATEWAY/Ventilationmode"
  icon: mdi:hvac
  value_template: >-
    {% set status = value | int %}
    {% if status == 13 %}
    {{'Perm. High'}}
    {% elif status == 12 %}
    {{'Perm. Mid'}}
    {% elif status == 11 %}
    {{'Perm. Low'}}
    {% elif status == 3 %}
    {{'High'}}
    {% elif status == 2 %}
    {{'Mid'}}
    {% elif status == 1 %}
    {{'Low'}}
    {% elif status == 0 %}
    {{'Auto'}}
    {% else %}
    {{'Undefined'}}
    {% endif %}
- platform: mqtt
  unique_id: duco_vent_perc
  name: Duco Ventilation Percentage
  state_topic: "VENTILATION_GATEWAY/SERGW_VETILATION_PERCENTAGE/Ventilationpercentage"
  unit_of_measurement: "%"
  icon: mdi:hvac
- platform: mqtt
  unique_id: duco_fan_perc
  name: Duco Current Fan Percentage
  state_topic: "VENTILATION_GATEWAY/SERGW_CUR_FAN/ventilationfan"
  unit_of_measurement: "%"
  icon: mdi:fan
- platform: mqtt
  unique_id: duco_fan_rpm
  name: Duco Fan Speed
  state_topic: "VENTILATION_GATEWAY/SERGW_FANSPEED/Ventilationfanspeed"
  unit_of_measurement: "rpm" 
  icon: mdi:fan
[... 169 lines cut ...]
          - type: custom:button-card
            entity: switch.duco_ventilation_high
            name: High
            styles:
              card:
                - height: 50px
          - type: custom:button-card
            entity: switch.duco_ventilation_low_perm
            name: P. Low
            styles:
              card:
                - height: 50px
          - type: custom:button-card
            entity: switch.duco_ventilation_mid_perm
            name: P. Mid
            styles:
              card:
                - height: 50px
          - type: custom:button-card
            entity: switch.duco_ventilation_high_perm
            name: P. High
            styles:
              card:
                - height: 50px
```

broekieman, 2022-01-25:
Helaas! Ik heb alles aangepast naar jouw configuratie, maar alle entiteiten blijven "niet beschikbaar".
Hoe kan ik testen of het überhaupt wel bij Home Assistant aankomt?

rhvs, 2022-01-25:
Heb je alle mqtt berichten even verwijderd? Als je hebt zitten prutsen met de retain flag, dan kan er onbedoeld wat blijven hangen.

Gibletron, 2022-01-25:
> Helaas! Ik heb alles aangepast naar jouw configuratie, maar alle entiteiten blijven "niet beschikbaar". Hoe kan ik testen of het überhaupt wel bij Home Assistant aankomt?

Dit gaat allemaal VEEL verder dan hulp bij Ventilation Gateway, en ik zou je aanraden om zeker eens uit te zoeken hoe HA nou echt werkt

We weten dankzij MQTT explorer dat de VentGW informatie naar MQTT stuurt, en deze wordt ontvangen
Uit jouw screenshot maak ik tevens op dat er een aantal shelly en sonoff devices op mqtt zit, ik ga er van uit dat die vanuit HA wel werken, als dat het geval is werkt de mqtt integratie tussen HA en MQTT ook

Ga nu naar home assisstant -> integrations en klik op de "configure" knop in de mosquitto integration
[image]

Vul dan een Topic in bij het "start listening" veld:
Bijvoorbeeld: VENTILATION_GATEWAY/VENTILATION_GATEWAY/Ventilationmode
En verander de ventilatie mode via bijvoorbeeld het muurkastje
[image]

Je zou nu meldingen moeten zijn dat er iets veranderd als de berichten door Home Assistant worden gezien

broekieman, 2022-01-25:
> Heb je alle mqtt berichten even verwijderd? Als je hebt zitten prutsen met de retain flag, dan kan er onbedoeld wat blijven hangen.

Hoe verwijder ik berichten?

Gibletron, 2022-01-25:
> > Heb je alle mqtt berichten even verwijderd? Als je hebt zitten prutsen met de retain flag, dan kan er onbedoeld wat blijven hangen.
> 
> Hoe verwijder ik berichten?

https://lmgtfy.app/?q=home+assistant+mqtt+remove+retained+messages

broekieman, 2022-01-25:
> Dit gaat allemaal VEEL verder dan hulp bij Ventilation Gateway, en ik zou je aanraden om zeker eens uit te zoeken hoe HA nou echt werkt
>
> We weten dankzij MQTT explorer dat de VentGW informatie naar [...]

Verdraaid man, er gebeurt wat. Moest ik die integratie aanmaken, of.....? De addon zegt nl. dat er een verbinding is. 
Ik heb nu switches en als ik daarop klik verandert de waarde van de gateway, mooi dus. Deze schakelaar gaat gelijk weer op uit, maar de gateway blijft gewoon op de geschakelde stand vanuit Home Assistant? De sensors zijn van niet beschikbaar naar onbekend gegaan, maar geven dus géén waarde.

Gibletron, 2022-01-25:
> Verdraaid man, er gebeurt wat. Moest ik die integratie aanmaken, of.....? De addon zegt nl. dat er een verbinding is. Ik heb nu switches en als ik daarop klik verandert de waarde van de gateway, [...]

Lees dit topic, dat de schakelaar direct weer naar uit  gaat, inclusief oplossing, wordt besproken
Als je mijn config gebruikt zou je hier in principe geen last van moeten hebben

broekieman, 2022-01-25:
Shit, moet ik weer terug naar jouw configuratie. Doe ik morgen. Allemaal bedankt en jullie horen vast nog van me want ik wil de sensoren nog zien werken. Moet ik de integratie Mqtt dus configureren om het werkend te houden?

Gibletron, 2022-01-25:
Home Assistant moet verbinding hebben met MQTT, de Moaquitto broker is niet meer dan een addon, een extra docker container op je Home Assistant host, maar je moet wel zorgen dat Home Assistant tegen die container aan gaat praten..
Dus ja.. je moet de integration hebben zodat Home Assistant ook MQTT gaat praten

broekieman, 2022-01-26:
> LETOP onderstaande config gebruikt custom:button-card en werkt dus NIET out of the box! Gebruik dezelfde sensoren/switches maar niet EXACT deze config, tenzij je weet hoe de custom button-card werkt!
> 
> Lovelace Config:
> 
> ```
> cards:
>   - type: entities
>     title: Ventilatie
>     entities:
>       - entity: sensor.duco_gateway_status
>       - entity: sensor.duco_current_fan_percentage
>       - entity: sensor.duco_fan_speed
>       - entity: sensor.duco_ventilation_mode
>       - entity: sensor.duco_ventilation_percentage
>       - entity: sensor.duco_ventilation_countdown
>       - entity: sensor.duco_box_1_temp
>       - entity: sensor.duco_box_1_co2_ppm
>       - entity: sensor.duco_box_2_temp
>     show_header_toggle: false
>   - type: grid
>     columns: 1
>     square: false
>     cards:
>       - type: custom:button-card
>         entity: switch.duco_ventilation_auto
>         name: Auto Mode
>         styles:
>           card:
>             - height: 75px
>       - type: grid
>         columns: 3
>         square: false
>         cards:
>           - type: custom:button-card
>             entity: switch.duco_ventilation_low
>             name: Low
>             styles:
>               card:
>                 - height: 50px
>           - type: custom:button-card
>             entity: switch.duco_ventilation_middle
>             name: Mid
>             styles:
>               card:
>                 - height: 50px
>           - type: custom:button-card
>             entity: switch.duco_ventilation_high
>             name: High
>             styles:
>               card:
>                 - height: 50px
>           - type: custom:button-card
>             entity: switch.duco_ventilation_low_perm
>             name: P. Low
>             styles:
>               card:
>                 - height: 50px
>           - type: custom:button-card
>             entity: switch.duco_ventilation_mid_perm
>             name: P. Mid
>             styles:
>               card:
>                 - height: 50px
>           - type: custom:button-card
>             entity: switch.duco_ventilation_high_perm
>             name: P. High
>             styles:
>               card:
>                 - height: 50px
> ```

Als ik de button-card van Haccs gebruik en jouw code plak gebeurt er niets. Ik heb het in de Layout-card plak krijg ik iets leuks.
[image]

Gibletron, 2022-01-26:
> 

Ziet er redelijk uit, lijkt er op dat de sensoren werken! Neem aan dat de knoppen het ook doen.
Wel zie ik dat de knoppen te laag zijn en daardoor de icons verborgen zijn, je kunt ze wat hoger maken als je die graag wil zien
Ik heb een redelijk uitgebreide HA setup, maar is niet echt de scope van wat er op deze github besproken moet worden, daarvoor zijn de HA forums ;)

broekieman, 2022-01-26:
Als ik de knoppen hoger maak krijg ik nog steeds geen icon in de onderste rij.

broekieman, 2022-01-26:
Ik probeer een automation te maken voor de badkamer.
Switch permanent on als humidity boven 83 komt.
Switch off als humidity onder 60 komt.

Dit werkt nl. niet.:
#badkamer ventilatie
- id: badkamer_ventilatie
  alias: 'Badkamer ventilatie'
  trigger:
    platform: numeric_state
    entity_id: sensor.prologue_90d2_hum
    above: 83
    for:
      minutes: 2
  action:
  - service: switch.turn_on
    entity_id: switch.duco_ventilation_high_perm

JJ-82, 2022-04-07:
Afgelopen week de gateway ontvangen en vanochtend de meest verse firmware erop gezet. Spullenboel netjes aangesloten/ingesteld en met behulp van wat info hierboven de schakelaars in Home Assistant aan kunnen maken. Ik loop alleen vast op de stand van de schakelaars.

Hoog/Midden/Laag gaat goed, maar ik '15min' of 'permanent' wordt niet opgepakt. Dat wil zeggen, als ik voor een permanent hoog kies, dan gaat de box keurig naar permanent hoog. De wandbediening geeft ook blauwe golfjes, maar in Home Assistant zie ik na enkele seconden de schakelaar van 'permanent hoog' uitgaan en de schakelaar van '15 min hoog' aangaan.

Een snelle blik in MQTT Explorer leert dan de Ventilationmode (VENTILATION_GATEWAY/Ventilation_gateway_RF/Ventilationmode), een single digit teruggeeft. Dit is altijd een 1/2/3. De voorlopende 1 die de status 'permanent' zou moeten duiden, komt niet voor. :-(

Wat zie ik over het hoofd?

[44 more replies]