arnemauer/Ducobox-ESPEasy-Plugin #25: calls to ventilation-gateway.arnemauer.nl?

jorisvansoest, 2020-05-04:
After setting up the serial gateway (without RF for now) I noticed a lot of calls to the domain `ventilation-gateway.arnemauer.nl` via my Pi-hole setup. Every ~2,5 minutes a call is made to that domain.

What is the purpose for this?

arnemauer, 2020-05-23:
Aangezien Duco voornamelijk in NL en BE levert kunnen we prima overgaan op het Nederlands :)

Het klopt dat de gateway om de 2,5 minuten (in de nieuwe firmware om de 10 minuten) een bericht stuurt. In de oudere firmwares waren een aantal bugs waardoor het systeem onstabiel was. Om de stabiliteit van de verschillende firmwareversies te monitoren stuurt de gateway om de 10 minuten een bericht met de actuele uptime en versienummer. Ik zal in de volgende firmwareversie een checkbox toevoegen voor het in- en uitschakelen van de functie.

IgorYbema, 2020-05-24:
@arnemauer Waarom is dat niet zichtbaar in de code op github dan maar wel in de binaries die op github staan? Is de code op github niet actueel?

samvdb, 2020-05-27:
@IgorYbema de code op github bevat enkel de plugins die nodig zijn om met duco te praten. 
Als je een binary maakt voor espeasy ( en andere esp chips ), include je in het build process meerdere libraries en je eigen code + wat extra config waaronder de auto update url's.
In deze repo zit dus niet alle code maar enkel de plugins.
Het is een hele poos geleden dat ik nog met esp gewerkt heb maar zo zat het toen in elkaar :)

jorisvansoest, 2020-05-30:
> Aangezien Duco voornamelijk in NL en BE levert kunnen we prima overgaan op het Nederlands :)
>
> Het klopt dat de gateway om de 2,5 minuten (in de nieuwe firmware om de 10 minuten) een bericht stuurt. [...]

Dat laatste zou ik wel prettig vinden. Ik had verwacht dat de binary gelijk zou zijn aan de beschikbare code (plug-ins) hier. Is er nog meer toegevoegd aan jouw binary?

sanderlv, 2021-07-07:
> Aangezien Duco voornamelijk in NL en BE levert kunnen we prima overgaan op het Nederlands :)
>
> Het klopt dat de gateway om de 2,5 minuten (in de nieuwe firmware om de 10 minuten) een bericht stuurt. [...]

Hoi Arne, is dit al "opgelost"?
[image]

arnemauer, 2021-07-07:
@sanderlv  Heb jij de meest recente firmware? Volgens mij is die functie in de recente firmware eruit gehaald.
Over twee weken heb ik vakantie en zal ik er verder naar kijken.

sanderlv, 2021-07-08:
Ja, meest recente firmware. Alhoewel, zo recent is die ook weer niet :-)

De tijd gaat harder dan je denkt...

[image]

arnemauer, 2022-03-30:
Gefixt 

> After setting up the serial gateway (without RF for now) I noticed a lot of calls to the domain `ventilation-gateway.arnemauer.nl` via my Pi-hole setup. Every ~2,5 minutes a call is made to that domain.
> 
> What is the purpose for this?
> 
> Web Log (dev):
> 
> ```
> 329672266: REP : Reporting status to ventilation-gateway.arnemauer.nl
> 329672271: REP : connection failed
> ```
> 
> Without my Pi-hole blocking:
> 
> ```
> 329822266: REP : Reporting status to ventilation-gateway.arnemauer.nl
> 329822341: REP : report uploaded
> ```

Deze functie is verwijderd uit de binaries ([17a29af](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/17a29af05e39d4e8057534e4bf7e0409b5c83714)) en komt niet meer terug :)

closed 2022-03-30 by arnemauer