arnemauer/Ducobox-ESPEasy-Plugin #26: Lost RF 'Join' to Ducobox

shuffleznl, 2020-05-13:
Upon initial configuration RF join was OK and RF control worked.
After power off (replacing USB PSU) of gateway, the Network ID is still recorded, but Device Address appears lost (at least set to 0). Tried rejoining once, but to no avail. Hesitant to try rejoin again because of RF-reasons stated in Wiki.

(Off topic; any option to remove/reset all RF connections on Duco?)

IgorYbema, 2020-05-23:
Same here. Should it auto rejoin or not?

arnemauer, 2020-05-23:
Aangezien Duco voornamelijk in NL en BE levert kunnen we prima overgaan op het Nederlands :)

@shuffleznl: Dat is vervelend. Weet je het adres nog? Dan kun je deze handmatig invullen. 
Zou je een adres, bijvoorbeeld 10, willen invullen en dan op "save" klikken. Daarna de gateway stroomloos maken en vervolgens kijken of het adres nog wel is ingevuld?

Er is een optie om alle RF-verbindingen te resetten. Je moet daarna alleen wel alles weer opnieuw instellen. Uit pagina 9 van de Ducobox connect handleiding:

> • Alle componenten uit het netwerk verwijderen (bv. bij problemen): Activeer ‘Installer mode’ en druk lang op ‘INST’ tot de LED rood begint te knipperen. De DucoBox zal herstarten (ongeveer 15 seconden) en de LED zal stoppen met knipperen.
> • Fabrieksinstellingen herstellen van de DucoBox en alle aangemelde componenten (VANAF DucoBox versie 18xxxx): Druk lang op ‘INST’ en ‘ENTER’ wanneer niet in ‘Installer mode’. Het netwerk blijft behouden.

@IgorYbema @shuffleznl Wanneer je gateway versie 6 of hoger hebt kun je ook de Duco Network Tool gebruiken om het adres van de gateway te achterhalen.

IgorYbema, 2020-05-24:
Bij mij werkt het nu gewoon ook na power resets. En toch had ik er heel veel last van bij het initieel configureren. Nu kan ik dat niet meer reproduceren.

arnemauer, 2020-05-24:
Wellicht dat in de nieuwe espeasy versie er iets is veranderd qua opslaan. Ik ga het uitzoeken.

shuffleznl, 2020-06-01:
Na verwijderen alle componenten en opnieuw joinen werkt het weer soepel!

shuffleznl, 2020-06-08:
Na enige tijd is het 'paar' cq de join weer vervallen.
Ik denk vergelijkbaar issue met #34

arnemauer, 2020-06-08:
> Na enige tijd is het 'paar' cq de join weer vervallen.

Wat bedoel je met vervallen? Het netwerkadres staat niet meer in de pluginconfiguratie?

shuffleznl, 2020-06-15:
Netwerkadres staat er nog, maar hij verliest DeviceID (die gaat naar 0).
Daarmee zijn de beschikbare 'Actions' ook weg (alleen nog 'Join').

arnemauer, 2020-06-15:
> Netwerkadres staat er nog, maar hij verliest DeviceID (die gaat naar 0).
> Daarmee zijn de beschikbare 'Actions' ook weg (alleen nog 'Join').

Heb je de nieuwste firmware gebruikt?
->  firmware aa380ce 08-06-2020 (ESPEasy mega-20200515)

shuffleznl, 2020-06-15:
Heb één met nieuwe, één met oudere geprobeerd, maar blijft vervallen.
Misschien dat mijn Duco de geest gegeven heeft, valt me nu pas op dat Mode & Speed ook op 0 staan (Serial).

arnemauer, 2020-06-15:
Dat het adres niet blijft opgeslagen in de gateway heeft niets met de ducobox te maken. 

Begrijp ik het goed dat je handmatig het adres hebt ingevuld en opgeslagen. En dat het adres na een dagen op 0 staat?

Wat staat er in de weblog nadat je het adres erin hebt gezet?

shuffleznl, 2020-06-15:
`3639924:  Webserver args: 0: 'index' length: 1 1: 'page' length: 1
3644356: FILE : Saved config.dat
3644356: SaveToFile: free stack after: 3328
3644357: SaveToFile: free stack: 2992
3644365: FILE : Saved security.dat
3644365: SaveToFile: free stack after: 2992
3644375: [P150] DUCO RF GW: Extra Settings PLUGIN_150 loaded
3644386: [P150] DUCO RF GW: Values set from config. DeviceID: 0, networkId: 082cd0, radio power: Default - radio power 0,6 dBm (0x8D)
3644389: [P150] DUCO RF GW: CC1101 868Mhz transmitter initialized
3644462: [P150] DUCO RF GW: TIMER => 3644461 3644386 75
3644462: [P150] DUCO RF GW: Start of RF signal received
3644462: [P150] DUCO RF GW: Ignoring RF noise
3644468: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
3644469: [P150] DUCO RF GW:  State: 0.00
3644563: EVENT: Ventilation_gateway_RF#Ventilationmode=0.00
3648518:  Webserver args: 0: 'setpage' length: 1
3650180:  Webserver args: 0: 'index' length: 1 1: 'page' length: 1
3660388: [P151] DUCO SER GW: Start readNetworkList
3660428: [P156] DUCO FanSpeed: start read, eventid:2
3660429: [P156] DUCO FanSpeed: Serial port in use, set flag to read data later.
3660463: EVENT: GroundFloor_VentilationMode#Ventilationmode=0.00
3660547: [P151] DUCO SER GW: Received invalid response on network
3660548: [P151] DUCO SER GW: Stop reading from serial
3660563: EVENT: GroundFloor_FanSpeed#FanSpeed=0.00
3661405: [P156] DUCO FanSpeed: start read, eventid:2
3661405: [P156] DUCO FanSpeed: Start reading fanspeed
3661584: [P156] DUCO FanSpeed: Received invalid response
3661585: [P156] DUCO FanSpeed: Stop reading from serial
3661961: WD   : Uptime 61 ConnectFailures 0 FreeMem 11792 WiFiStatus 3
3666454: NTP  : NTP replied: delay 20 mSec Accuracy increased by 0.241 seconds
3666454: Time set to 1592249696.241 Time adjusted by 24.49 msec. Wander: 0.01 msec/second
3666457: Local time: 2020-06-15 19:34:56
3666463: EVENT: Time#Set
3666490: [P150] DUCO RF GW: TIMER => 3666490 3666474 16
3666490: [P150] DUCO RF GW: Start of RF signal received
3666491: [P150] DUCO RF GW: Ignoring RF noise
3669142: FILE : Saved config.dat
3669142: SaveToFile: free stack after: 3328
3669143: SaveToFile: free stack: 2992
3669151: FILE : Saved security.dat
3669151: SaveToFile: free stack after: 2992
3669162: [P150] DUCO RF GW: Extra Settings PLUGIN_150 loaded
3669172: [P150] DUCO RF GW: Values set from config. DeviceID: 4, networkId: 082cd0, radio power: Default - radio power 0,6 dBm (0x8D)
3669175: [P150] DUCO RF GW: CC1101 868Mhz transmitter initialized
3669249: [P150] DUCO RF GW: TIMER => 3669248 3669173 75
3669249: [P150] DUCO RF GW: Start of RF signal received
3669249: [P150] DUCO RF GW: Ignoring RF noise
3669254: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
3669255: [P150] DUCO RF GW:  State: 0.00
3669350: EVENT: Ventilation_gateway_RF#Ventilationmode=0.00
3670352:  Webserver args: 0: 'setpage' length: 1
3670426: EVENT: Clock#Time=Mon,19:35
`

arnemauer, 2020-06-15:
Je hebt de bruikbare informatie met betrekking tot het opslaan  uit de log weg ge-edit.

Zou je de vragen nog willen beantwoorden?

Zou je

closed 2020-07-02 by arnemauer