arnemauer/Ducobox-ESPEasy-Plugin #28: Send CNT1,CNT2,CNT3 using RF gateway possible?

IgorYbema, 2020-05-24:
I notice that currently it isn't possible to send permanent manual modes using the RF gateway to the ducobox. However from the ducobox manual it seems that a co2 controller should be able to request those modes (cnt1,cnt2,cnt3) by pressing a mode selector a few seconds.

Did you decode those messages already and is it possible to add them? I can contribute code also, if you only give me a head start on which code I need (from ESP easy and your repo).

IgorYbema, 2020-05-24:
Owhja,.. nederlands. Zo gewend om Engels te praten op github :)

arnemauer, 2020-05-24:
Ja dat is mogelijk. Het zit er voor een deel ook al in de code verwerkt: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/4df3138b4200b09548501dd4c389ee53b075db64/_P150_DucoRFGateway.ino#L269

In de volgende versie zal ik deze optie toevoegen (+- 1 week).

arnemauer, 2020-06-01:
Ik ben een tijdje bezig geweest met het bekijken van het radioverkeer. Kom er net pas achter dat mijn afstandsbediening helemaal geen permanent stand heeft. 

`Afhankelijk van de software versie van uw bedieningsschakelaar, kan permanente stand niet aanwezig zijn. In dat geval wordt bij bovenstaande acties afwezigheidsstand geactiveerd voor stand 1, en tijdelijke stand voor stand 2 en 3.`

Zou iemand een log kunnen sturen waarin het radioverkeer te zien is terwijl de stand aangepast wordt naar continu high 1, 2 en 3? (usb aansluiten op laptop, alle seriële plugins uitschakelen, logging serial op Debug en een Putty terminal openen).

IgorYbema, 2020-06-01:
Ik heb helaas geen draadloze schakelaar.  Enige wat ik wel zie is dat als ik m'n 3-standenschakelaar (wat de CNT1,CNT2,CNT3 ook schakelt) omzet ik de volgende berichten krijg vanuit de ducobox:
FF,20,00,12,0C = CNT1
FF,20,00,12,0D = CNT2
FF,20,00,12,0E = CNT3

Misschien heb je hier wat aan. Maar is de verkeerde kant op natuurlijk :)

arnemauer, 2020-06-01:
> Ik heb helaas geen draadloze schakelaar. Enige wat ik wel zie is dat als ik m'n 3-standenschakelaar (wat de CNT1,CNT2,CNT3 ook schakelt) omzet ik de volgende berichten krijg vanuit de ducobox: [...]

Bedankt! Blijkbaar geeft de Ducobox de "normale standen" door aan de schakelaars (0x0C =low, 0x0D=middle, 0x0F=high). Dus helaas geen speciale continu-stand. Op zich ook wel logisch aangezien dan ook oudere schakelaars de stand goed kunnen weergeven. 

Wellicht iemand anders die een draadloze schakelaar heeft met de mogelijkheid om een continu stand te activeren (5 seconden knop ingedrukt houden). Het gaan dan om ventilatiestand middel en hoog.

Pinkel, 2020-06-01:
Ventilatiestand laag kan volgens mij ook permanent aan?
Ik zal morgen kijken of ik er een log van kan trekken. Volgens mij deed die
van mij het wel

IgorYbema, 2020-06-01:
@Pinkel ja als je dus de nieuwere draadloze controller hebt moet je een standknop dus 5 seconden vast houden om hem in die stand te zetten continu
Daar zou @arnemauer graag een log van willen hebben.

arnemauer, 2020-06-01:
> Ventilatiestand laag kan volgens mij ook permanent aan? Ik zal morgen kijken of ik er een log van kan trekken. Volgens mij deed die van mij het wel

Klopt! Die werkt bij mij dus wel maar dat is de emptyhome stand.

Pinkel, 2020-06-01:
Ik heb even snel een log gemaakt via de weblog. Hopelijk heb je hier al
genoeg aan. Hierbij zet ik de ventilator via de draadloze schakelaar in
´permanent stand 1´ -> 'permanent stand 2' -> 'permanent stand 3' -> 'auto'.
Ik zie op de bedrade schakelaar dat deze inderdaad de stand overneemt; de
stand wordt hier paars.
Op het RF schakelaartje zie je alleen dat een lampje groen oplicht en deze
weer uit gaat na ~5 seconde ingedrukt houden.

Hopelijk helpt dit.

*-- 1e tot deze in parmanente stand zit --*
457441491: [P150] DUCO RF GW: TIMER => 457441490 457441483 7
457441491: [P150] DUCO RF GW: Start of RF signal received
457441493: [P150] DUCO RF GW: 457441492 - Received message: SRC:1; DEST:2;
ORG.SRC:1; ORG.DEST:2; Ntwrk:0005c507;
457441493: [P150] DUCO RF GW: 457441492 - Type:6; Bytes:9; Counter:1;
RSSI:-56 (0x25);
457441493: [P150] DUCO RF GW: 457441492 - FF,
457442162: WD : Uptime 7624 ConnectFailures 0 FreeMem 9448 WiFiStatus 3
457442291: [P150] DUCO RF GW: TIMER => 457442290 457442279 11
457442291: [P150] DUCO RF GW: Start of RF signal received
457442292: [P150] DUCO RF GW: 457442292 - Received message: SRC:1; DEST:2;
ORG.SRC:1; ORG.DEST:2; Ntwrk:0005c507;
457442293: [P150] DUCO RF GW: 457442292 - Type:7; Bytes:13; Counter:15;
RSSI:-56 (0x25);
457442293: [P150] DUCO RF GW: 457442292 - FF,20,00,12,0C,
457442691: [P150] DUCO RF GW: TIMER => 457442690 457442672 18
457442691: [P150] DUCO RF GW: Start of RF signal received
457442692: [P150] DUCO RF GW: 457442692 - Received message: SRC:1; DEST:3;
ORG.SRC:1; ORG.DEST:3; Ntwrk:0005c507;
457442693: [P150] DUCO RF GW: 457442692 - Type:7; Bytes:13; Counter:1;
RSSI:-56 (0x25);
457442693: [P150] DUCO RF GW: 457442692 - FF,20,00,12,0C,
457442990: [P150] DUCO RF GW: TIMER => 457442990 457442983 7
457442991: [P150] DUCO RF GW: Start of RF signal received
457442992: [P150] DUCO RF GW: 457442992 - Received message: SRC:1; DEST:3;
ORG.SRC:1; ORG.DEST:3; Ntwrk:0005c507;
457442993: [P150] DUCO RF GW: 457442992 - Type:7; Bytes:13; Counter:1;
RSSI:-56 (0x24);
457442993: [P150] DUCO RF GW: 457442992 - FF,20,00,12,0C,
457443391: [P150] DUCO RF GW: TIMER => 457443390 457443380 10
457443391: [P150] DUCO RF GW: Start of RF signal received
457443392: [P150] DUCO RF GW: 457443392 - Received message: SRC:1; DEST:3;
ORG.SRC:1; ORG.DEST:3; Ntwrk:0005c507;
457443393: [P150] DUCO RF GW: 457443392 - Type:6; Bytes:9; Counter:3;
RSSI:-56 (0x25);
457443393: [P150] DUCO RF GW: 457443392 - FF,
457444671: [P150] DUCO RF GW: TIMER => 457444670 457444668 2
457444671: [P150] DUCO RF GW: Start of RF signal received
457444672: [P150] DUCO RF GW: 457444672 - Received message: SRC:1; DEST:3;
ORG.SRC:1; ORG.DEST:3; Ntwrk:0005c507;
457444673: [P150] DUCO RF GW: 457444672 - Type:7; Bytes:13; Counter:2;
RSSI:-56 (0x24);
457444673: [P150] DUCO RF GW: 457444672 - FF,20,00,12,0C,
457444991: [P150] DUCO RF GW: TIMER => 457444990 457444975 15
457444991: [P150] DUCO RF GW: Start of RF signal received
457444993: [P150] DUCO RF GW: 457444992 - Received message: SRC:1; DEST:3;
ORG.SRC:1; ORG.DEST:3; Ntwrk:0005c507;
457444993: [P150] DUCO RF GW: 457444992 - Type:7; Bytes:13; Counter:2;
RSSI:-56 (0x25);
457444993: [P150] DUCO RF GW: 457444992 - FF,20,00,12,0C,
457446891: [P150] DUCO RF GW: TIMER => 457446890 457446878 12
457446891: [P150] DUCO RF GW: Start of RF signal received
457446893: [P150] DUCO RF GW: 457446892 - Received message: SRC:1; DEST:3;
ORG.SRC:1; ORG.DEST:3; Ntwrk:0005c507;
457446893: [P150] DUCO RF GW: 457446892 - Type:7; Bytes:13; Counter:3;
RSSI:-56 (0x25);
457446893: [P150] DUCO RF GW: 457446892 - FF,20,00,12,0C,
457447191: [P150] DUCO RF GW: TIMER => 457447190 457447185 5
457447191: [P150] DUCO RF GW: Start of RF signal received
[... 181 lines cut ...]
457621965: [P150] DUCO RF GW: 457621964 - Type:6; Bytes:9; Counter:9;
RSSI:-56 (0x25);
457621965: [P150] DUCO RF GW: 457621964 - FF,
457622162: WD : Uptime 7627 ConnectFailures 0 FreeMem 9448 WiFiStatus 3
457622289: [P150] DUCO RF GW: TIMER => 457622289 457622238 51
457622290: [P150] DUCO RF GW: Start of RF signal received
457622291: [P150] DUCO RF GW: 457622291 - Received message: SRC:1; DEST:3;
ORG.SRC:1; ORG.DEST:3; Ntwrk:0005c507;
457622292: [P150] DUCO RF GW: 457622291 - Type:7; Bytes:13; Counter:10;
RSSI:-56 (0x25);
457622292: [P150] DUCO RF GW: 457622291 - FF,20,00,12,00,
457624149: [P150] DUCO RF GW: TIMER => 457624148 457624144 4
457624149: [P150] DUCO RF GW: Start of RF signal received
457624151: [P150] DUCO RF GW: 457624150 - Received message: SRC:1; DEST:3;
ORG.SRC:1; ORG.DEST:3; Ntwrk:0005c507;
457624151: [P150] DUCO RF GW: 457624150 - Type:7; Bytes:13; Counter:11;
RSSI:-56 (0x25);
457624151: [P150] DUCO RF GW: 457624150 - FF,20,00,12,00,
457624469: [P150] DUCO RF GW: TIMER => 457624468 457624455 13
457624469: [P150] DUCO RF GW: Start of RF signal received
457624470: [P150] DUCO RF GW: 457624470 - Received message: SRC:1; DEST:3;
ORG.SRC:1; ORG.DEST:3; Ntwrk:0005c507;
457624471: [P150] DUCO RF GW: 457624470 - Type:7; Bytes:13; Counter:11;
RSSI:-56 (0x25);
457624471: [P150] DUCO RF GW: 457624470 - FF,20,00,12,00,

arnemauer, 2020-06-03:
@Pinkel Ik heb de logs bekeken maar helaas zit het bericht van de bedieningsschakelaar richting de Ducobox er niet tussen. Wellicht dat die wegvalt in de weblog door de vele berichten achter elkaar. Als je alle berichten wilt kunnen zien dan kun je het beste de gateway op een pc aansluiten en serial debug aanzetten.

Het commando begint met: 0x40 0x00 0x22
Dus in de logs zie je : FF,40,00,22,

Pinkel, 2020-06-07:
Hey Arne,

Dit heb ik zojuist gedaan, zie de logs hieronder. Ik heb alleen het gevoel
dat ze toch nog niet getoond worden. Zou het kunnen zijn dat de plugin iets
wegfiltert / niet logt? Ik gebruik een versie van ~2 weken geleden.

*Perma stand 1 -- start: 678377, end: 715336*
678377 : Debug : [P150] DUCO RF GW: Ignoring RF noise
678973 : Info  : [P150] DUCO RF GW: 678967 - CheckforAck: still waiting for
ACK. Sending message again...
678974 : Info  : [P150] DUCO RF GW: 678967 -
0F,07,00,05,C5,07,20,48,16,FF,40,00,22,00,00,6D,
678976 : Debug : [P150] DUCO RF GW: TIMER => 678976 678973 3
678977 : Debug : [P150] DUCO RF GW: Start of RF signal received
678977 : Debug : [P150] DUCO RF GW: Ignoring RF noise
679573 : Info  : [P150] DUCO RF GW: 679566 - CheckforAck: still waiting for
ACK. Sending message again...
679573 : Info  : [P150] DUCO RF GW: 679567 -
0F,07,00,05,C5,07,20,48,16,FF,40,00,22,00,00,6D,
679575 : Debug : [P150] DUCO RF GW: TIMER => 679575 679573 2
679576 : Debug : [P150] DUCO RF GW: Start of RF signal received
679578 : Info  : [P150] DUCO RF GW: 679577 - Received message: SRC:24;
DEST:20; ORG.SRC:3; ORG.DEST:16; Ntwrk:00000505;
679579 : Info  : [P150] DUCO RF GW: 679577 - Type:6; Bytes:9; Counter:9;
RSSI:-40 (0x45);
679579 : Info  : [P150] DUCO RF GW: 679578 - 02,
679580 : Info  : [P150] DUCO RF GW: 679578 - Ignore message, not our
networkID.
679595 : Debug : [P150] DUCO RF GW: TIMER => 679595 679578 17
679596 : Debug : [P150] DUCO RF GW: Start of RF signal received
679596 : Debug : [P150] DUCO RF GW: Ignoring RF noise
680075 : Debug : [P150] DUCO RF GW: TIMER => 680074 680071 3
680075 : Debug : [P150] DUCO RF GW: Start of RF signal received
680095 : Debug : [P150] DUCO RF GW: TIMER => 680095 680094 1
680095 : Debug : [P150] DUCO RF GW: Start of RF signal received
680097 : Info  : [P150] DUCO RF GW: 680096 - Received message: SRC:1;
DEST:4; ORG.SRC:1; ORG.DEST:4; Ntwrk:0005c507;
680097 : Info  : [P150] DUCO RF GW: 680096 - Type:6; Bytes:9; Counter:7;
RSSI:-85 (0xEA);
680098 : Info  : [P150] DUCO RF GW: 680097 - FF,
680099 : Info  : [P150] DUCO RF GW: 680097 - Received messagetype: ACK
680099 : Info  : [P150] DUCO RF GW: 680097 - Ack received!
680415 : Debug : [P150] DUCO RF GW: TIMER => 680415 680396 19
680415 : Debug : [P150] DUCO RF GW: Start of RF signal received
680435 : Debug : [P150] DUCO RF GW: TIMER => 680434 680434 0
680435 : Debug : [P150] DUCO RF GW: Start of RF signal received
680437 : Info  : [P150] DUCO RF GW: 680436 - Received message: SRC:1;
DEST:4; ORG.SRC:1; ORG.DEST:4; Ntwrk:0005c507;
680437 : Info  : [P150] DUCO RF GW: 680436 - Type:6; Bytes:9; Counter:8;
RSSI:-85 (0xEA);
680437 : Info  : [P150] DUCO RF GW: 680436 - FF,
680438 : Info  : [P150] DUCO RF GW: 680436 - Received messagetype: ACK
680439 : Info  : [P150] DUCO RF GW: 680436 - Ack received!
682355 : Debug : [P150] DUCO RF GW: TIMER => 682355 682351 4
682355 : Debug : [P150] DUCO RF GW: Start of RF signal received
682356 : Debug : [P150] DUCO RF GW: Ignoring RF noise
682695 : Debug : [P150] DUCO RF GW: TIMER => 682694 682679 15
682695 : Debug : [P150] DUCO RF GW: Start of RF signal received
682696 : Debug : [P150] DUCO RF GW: Ignoring RF noise
683014 : Debug : [P150] DUCO RF GW: TIMER => 683014 683007 7
683015 : Debug : [P150] DUCO RF GW: Start of RF signal received
683016 : Debug : [P150] DUCO RF GW: Ignoring RF noise
685215 : Debug : [P150] DUCO RF GW: TIMER => 685214 685196 18
685215 : Debug : [P150] DUCO RF GW: Start of RF signal received
685235 : Debug : [P150] DUCO RF GW: TIMER => 685235 685228 7
685235 : Debug : [P150] DUCO RF GW: Start of RF signal received
685236 : Debug : [P150] DUCO RF GW: Ignoring RF noise
685773 : Info  : [P150] DUCO RF GW: 685766 - CheckforAck: still waiting for
ACK. Sending message again...
685773 : Info  : [P150] DUCO RF GW: 685766 -
[... 328 lines cut ...]
DEST:4; ORG.SRC:1; ORG.DEST:4; Ntwrk:0005c507;
1020297 : Info  : [P150] DUCO RF GW: 1020291 - Type:7; Bytes:13;
Counter:10; RSSI:-81 (0xF1);
1020298 : Info  : [P150] DUCO RF GW: 1020291 - FF,20,00,12,0E,
1020299 : Info  : [P150] DUCO RF GW: 1020292 - Received messagetype: Normal
message
1020299 : Info  : [P150] DUCO RF GW: 1020292 -
00,06,00,05,C5,07,20,48,1A,FF,
1020300 : Info  : [P150] DUCO RF GW: 1020297 - SEND ACK done!
1020301 : Info  : [P150] DUCO RF GW: 1020297 - Received change ventilation
command
1020310 : Debug : [P150] DUCO RF GW: TIMER => 1020309 1020296 13
1020310 : Debug : [P150] DUCO RF GW: Start of RF signal received
1020311 : Debug : [P150] DUCO RF GW: Ignoring RF noise
1021550 : Debug : [P150] DUCO RF GW: TIMER => 1021549 1021535 14
1021550 : Debug : [P150] DUCO RF GW: Start of RF signal received
1021551 : Debug : [P150] DUCO RF GW: Ignoring RF noise
1022183 : Info  : WD   : Uptime 17 ConnectFailures 0 FreeMem 11216
WiFiStatus 3
1022184 : Debug More: REP  : Reporting status to
ventilation-gateway.arnemauer.nl
1022316 : Error : REP  : connection failed
1028239 : Debug : [P150] DUCO RF GW: TIMER => 1028238 1028227 11
1028239 : Debug : [P150] DUCO RF GW: Start of RF signal received
1028240 : Debug : [P150] DUCO RF GW: Ignoring RF noise

vaneeten, 2020-06-28:
Heb sinds paar dagen deze fantastische gateway in gebruik. Firmware is 20200608 (Jun 8 2020 21:24:35).
De bedrade en RF afstandsbedieningen ondersteunen beide de long-press continu modi.
Momenteel zijn deze modi echter van korte duur, vermoedelijk omdat de gateway ze binnen de minuut reset naar een van de ondersteunde niet-continu modi.
Dit kan ik duidelijk zien op de bedrade unit, omdat daar de ventilatiestand paars oplicht zolang de continu modus geactiveerd is, en naar een aantal seconden weer wit wordt.

Onderstaande log is gestart nét voor het indrukken van 'continu-HIGH' (long-press op de RF bediening), totdat de modus terugvalt naar tijdelijk-HIGH.
Hopelijk staat hier iets nuttigs tussen.

172022196: WD   : Uptime 2867 ConnectFailures 0 FreeMem 11696 WiFiStatus 3
172026944: [P150] DUCO RF GW: TIMER => 172026944 172026938 6
172026945: [P150] DUCO RF GW: Start of RF signal received
172026945: [P150] DUCO RF GW: Ignoring RF noise
172031360: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
172031361: [P150] DUCO RF GW:  State: 3.00
172031405: EVENT: rf#Ventilationmode=3.00
172031448: ACT  : SendToHTTP 192.168.15.10,8084,/json.htm?param=udevice&type=command&idx=680&nvalue=0&svalue=40
172031449: Command: SendToHTTP
172031450: SendToHTTP: Host: 192.168.15.10 port: 8084
172032641: EVENT: Clock#Time=Sun,15:06
172034904: [P150] DUCO RF GW: TIMER => 172034904 172034903 1
172034905: [P150] DUCO RF GW: Start of RF signal received
172035024: [P150] DUCO RF GW: TIMER => 172035024 172035010 14
172035025: [P150] DUCO RF GW: Start of RF signal received
172035224: [P150] DUCO RF GW: TIMER => 172035224 172035214 10
172035225: [P150] DUCO RF GW: Start of RF signal received
172035544: [P150] DUCO RF GW: TIMER => 172035544 172035524 20
172035545: [P150] DUCO RF GW: Start of RF signal received
172036164: [P150] DUCO RF GW: TIMER => 172036164 172036151 13
172036165: [P150] DUCO RF GW: Start of RF signal received
172041689: [P150] DUCO RF GW: 172041677 - Received messagetype: Normal message
172041689: [P150] DUCO RF GW: 172041682 - SEND ACK done!
172041689: [P150] DUCO RF GW: 172041682 - Received change ventilation command
172041689: [P150] DUCO RF GW: 172041688 - SEND sendConfirmationNewVentilationMode done!
172041690: [P150] DUCO RF GW: 172041688 - Start waiting for ack...
172041693: [P150] DUCO RF GW: TIMER => 172041693 172041688 5
172041694: [P150] DUCO RF GW: Start of RF signal received
172041694: [P150] DUCO RF GW: 172041694 - Received messagetype: ACK
172041695: [P150] DUCO RF GW: 172041694 - Ack received but counter doesnt match!
172041704: [P150] DUCO RF GW: TIMER => 172041704 172041694 10
172041705: [P150] DUCO RF GW: Start of RF signal received
172041706: [P150] DUCO RF GW: 172041705 - Received messagetype: ACK
172041706: [P150] DUCO RF GW: 172041705 - Ack received!
172041707: EVENT: rf#Ventilationmode=14.00
172042004: [P150] DUCO RF GW: TIMER => 172042004 172041999 5
172042005: [P150] DUCO RF GW: Start of RF signal received
172042017: [P150] DUCO RF GW: 172042005 - Received messagetype: Normal message
172042018: [P150] DUCO RF GW: 172042010 - SEND ACK done!
172042018: [P150] DUCO RF GW: 172042010 - Received change ventilation command
172042018: [P150] DUCO RF GW: 172042017 - SEND sendConfirmationNewVentilationMode done!
172042018: [P150] DUCO RF GW: 172042017 - Start waiting for ack...
172042036: [P150] DUCO RF GW: TIMER => 172042036 172042022 14
172042036: [P150] DUCO RF GW: Start of RF signal received
172042037: [P150] DUCO RF GW: 172042037 - Received messagetype: ACK
172042037: [P150] DUCO RF GW: 172042037 - Ack received!
172042358: [P150] DUCO RF GW: 172042357 - SEND sendConfirmationNewVentilationMode done!
172042358: [P150] DUCO RF GW: 172042357 - Start waiting for ack...
172042364: [P150] DUCO RF GW: TIMER => 172042364 172042362 2
172042365: [P150] DUCO RF GW: Start of RF signal received
172042365: [P150] DUCO RF GW: 172042365 - Received messagetype: ACK
172042366: [P150] DUCO RF GW: 172042365 - Ack received!
172042641: [P150] DUCO RF GW: ventilation mode changed -> UPDATE by PLUGIN_ONCE_A_SECOND
172042641: [P150] DUCO RF GW:  State: 3.00
172042705: EVENT: rf#Ventilationmode=3.00
172042746: ACT  : SendToHTTP 192.168.15.10,8084,/json.htm?param=udevice&type=command&idx=680&nvalue=0&svalue=40
172042748: Command: SendToHTTP
172042749: SendToHTTP: Host: 192.168.15.10 port: 8084
172043265: [P150] DUCO RF GW: TIMER => 172043265 172043255 10
172043266: [P150] DUCO RF GW: Start of RF signal received
172043366: [P150] DUCO RF GW: TIMER => 172043365 172043357 8
172043366: [P150] DUCO RF GW: Start of RF signal received
172043705: [P150] DUCO RF GW: TIMER => 172043705 172043691 14
172043706: [P150] DUCO RF GW: Start of RF signal received
172044165: [P150] DUCO RF GW: TIMER => 172044165 172044164 1
172044166: [P150] DUCO RF GW: Start of RF signal received
172044178: [P150] DUCO RF GW: 172044166 - Received messagetype: Normal message
172044178: [P150] DUCO RF GW: 172044171 - SEND ACK done!
172044178: [P150] DUCO RF GW: 172044171 - Received change ventilation command
172044178: [P150] DUCO RF GW: 172044177 - SEND sendConfirmationNewVentilationMode done!
172044179: [P150] DUCO RF GW: 172044177 - Start waiting for ack...
172044185: [P150] DUCO RF GW: TIMER => 172044185 172044183 2
172044185: [P150] DUCO RF GW: Start of RF signal received
172044712: [P150] DUCO RF GW: 172044705 - CheckforAck: still waiting for ACK. Sending message again...
172044725: [P150] DUCO RF GW: TIMER => 172044725 172044718 7
[... cut ...]
172044727: [P150] DUCO RF GW: 172044726 - Ack received but counter doesnt match!
172045312: [P150] DUCO RF GW: 172045305 - CheckforAck: still waiting for ACK. Sending message again...
172045325: [P150] DUCO RF GW: TIMER => 172045325 172045318 7
172045326: [P150] DUCO RF GW: Start of RF signal received
172045326: [P150] DUCO RF GW: 172045326 - Received messagetype: ACK
172045327: [P150] DUCO RF GW: 172045326 - Ack received!
172052196: WD   : Uptime 2868 ConnectFailures 0 FreeMem 11696 WiFiStatus 3

arnemauer, 2020-06-28:
Klopt, ik ben de bug al tegengekomen en een fix staat al klaar :) 
Ik zal hem vanavond (uiterlijk morgen) op github zetten.

arnemauer, 2020-06-29:
Er staat een nieuwe firmware klaar :) Let op, het is een beta versie dus er kunnen nog bugs inzitten.

https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/master/Binaries/20200629%20Firmware%20281cde1%2029-06-2020%20(ESPEasy%20mega-20200515).bin

vaneeten, 2020-07-01:
Heb deze firmware via de web interface ingeschoten, en vooralsnog ziet het er perfect uit!

closed 2020-07-02 by arnemauer