arnemauer/Ducobox-ESPEasy-Plugin #47: Serial gateway werkt niet

sand3rk, 2020-07-17:
**Box:**
Ducobox Silent Connect
sw 5.1.0 

**ESP**
firmware: 20200629 Firmware
[image]

Esp start alleen wanneer TX/RX pins niet aan Ducobox hangen. Ik start eerst esp > TX/RX pins > Ducobox. Daarna zie ik geen Values. Enable Serial port aan/uit geprobeerd, beide geen values.

arnemauer, 2020-07-17:
Kun je onder tools => advanced => Web Log Level op debug zetten. Onder tools => log kun je dan zien wat er gebeurt. Zou je een aantal minuten van de log hier willen plaatsen?

sand3rk, 2020-07-17:
https://pastebin.com/5tDgGnvu

kees1987, 2020-08-23:
> Een oplossing is het doorsnijden van de TX van de seriele converter. Je kan dan alleen niet meer nieuwe firmware flashen via usb.

@arnemauer Thanks! Ik heb voorzichtig de TX pin van de CH340C (pin2) van de print los gebroken. --> De waarden komen nu binnen.
Een weerstandje kan ik er altijd wel weer ergens tussen braden als ik weer zou willen flashen ;)

Ik had een Wemos clone waarbij het wel functioneerde maar die had issues met een slechte voltage regulator. Vervangen door een Lolin en die had issues met de seriele communicatie. Naar nu blijkt had de clone een weerstand :)
Dank voor de geboden hulp!

closed 2020-08-24 by arnemauer

sanderlv, 2020-10-31:
Ik dacht de serial gw aan de praat te hebben,  kreeg eenmalig waarden. Maar daarna geen verandering meer. Reboot en alles behalve temp en en vochtigheid staat op 0. Wat moet ik precies "doorhalen"?

Is deze gitnog wel in leven? Heb al een paar vragen gesteld maar nog nooit antwoord gekregen... :-(