arnemauer/Ducobox-ESPEasy-Plugin #48: Join werkt niet

benvandijk84, 2020-08-14:
Heb een Wemos D1 Pro met een CC1101 module die ik wil koppelen aan mijn Ducobox.
Maak gebruik van firmware: 20200608 Firmware aa380ce 08-06-2020 (ESPEasy mega-20200515)
Wanneer ik de JOIN knop indruk gebeurt er niets, en zie dit in het log voorbij komen.

`336713: Command: JOIN
336714: [P150] DUCO RF GW: Sent command for 'join' to DUCO unit
336720: [P150] DUCO RF GW: 336714 - SendJoinPacket()
336720: [P150] DUCO RF GW: 336715 - 00,01,00,00,7C,3E,08,02,01,FF,0C,
336720: [P150] DUCO RF GW: 336720 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
336783: [P150] DUCO RF GW: TIMER => 336783 336719 64
336784: [P150] DUCO RF GW: Start of RF signal received
336784: [P150] DUCO RF GW: Ignoring RF noise
360571: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
360572: [P150] DUCO RF GW: State: 0.00
362150: WD : Uptime 6 ConnectFailures 0 FreeMem 10752 WiFiStatus 3
392150: WD : Uptime 7 ConnectFailures 0 FreeMem 8912 WiFiStatus 3`

Idee hoe dit te fixen?

arnemauer, 2020-08-14:
Staat de Ducobox in "Installer mode"?

benvandijk84, 2020-08-14:
Jazeker! Geprobeerd via de knop in de box. En via een bestaande remote. Lampje gaat in beide gevallen wel knipperen in de box. Maar verder gebeurt er vrij weinig helaas.

arnemauer, 2020-08-14:
Zie je in de weblog ook binnenkomende rf berichten terug?

benvandijk84, 2020-08-14:
Nee helemaal niets. Dat stukje log is eigenlijk het enigste wat ik voorbij zie komen.

benvandijk84, 2020-09-10:
Spul maar eens over gezet op een Wemos D1 mini.
En ook daar zelfde probleem helaas 😞  : 

> 1061386: Command: JOIN
1061387: [P150] DUCO RF GW: Sent command for 'join' to DUCO unit
1061393: [P150] DUCO RF GW: 1061387 - SendJoinPacket()
1061394: [P150] DUCO RF GW: 1061388 - 00,01,00,00,7C,3E,08,02,01,FF,0C,
1061394: [P150] DUCO RF GW: 1061393 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
1061458: [P150] DUCO RF GW: TIMER => 1061458 1061393 65
1061458: [P150] DUCO RF GW: Start of RF signal received
1061459: [P150] DUCO RF GW: Ignoring RF noise
1080683: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
1080683: [P150] DUCO RF GW: State: 0.00
> 

Ideeën of suggesties zijn welkom 😃

arnemauer, 2020-09-15:
Dan ben ik bang dat je CC1101 module niet werkt. 
Het laatste wat je nog kan proberen is het volgende: 
Zou je met de weblog open op de configuratiepagina van de plugin op "save" klikken? En dan de log hier posten?

Nadat de configuratie is opgeslagen wordt de CC1101 gereset en opnieuw geconfigureerd. Als daar iets fout gaat dan zie je het in de weblog terug.

benvandijk84, 2020-09-22:
Ik ga een nieuwe bestellen. Eens kijken of dat verbetering geeft. Zal wel even duren, AliExpress gaat niet zo vlot.

closed 2020-09-22 by arnemauer