arnemauer/Ducobox-ESPEasy-Plugin #49: Aansturen via serieel

martenjacobs, 2020-08-19:
Ik ben er vandaag achter gekomen dat het instellen van de fan via de seriële poort mogelijk is, iets dat voor zover ik weet nog niet algemeen bekend was tot nu toe (maar ik kan het fout hebben):
```
nodesetoverrule 1 100
nodesetoverrule 1 0
```
Zet de fan respectievelijk voluit en op de laagste stand. Wellicht is dat interessant om toe te voegen aan de seriële variant van deze library.

arnemauer, 2020-08-19:
Je zou nodesetoverrule kunnen gebruiken om de ventilatiestand in % handmatig aan te sturen. Dit heeft wel een aantal nadelen:

1. De stand is niet terug te zien op de bedieningsschakelaars.

2. Je overruled hiermee al je sensoren. Je CO2/luchtvochtigheidssensor zal zelf de ventilatiestand niet kunnen wijzigen ook al is de co2/luchtvochtigheid boven de triggerwaarde.

Zie onderstaande screenshots met "nodesetoverrule 1 100":
[image]
[image]
[image]

martenjacobs, 2020-08-19:
Dat vermoedde ik al, en dat is natuurlijk een nadeel als je schakelaars of sensoren hebt, maar dat is bij mij niet het geval 😬

martenjacobs, 2020-08-19:
Op Tweakers gaf je trouwens aan dat je dacht deze functie flash wear kon veroorzaken. Enig idee hoeveel writes we van dat flash kunnen verwachten?

closed 2020-09-15 by arnemauer