arnemauer/Ducobox-ESPEasy-Plugin #50: Duco Energy Premium System

26miller, 2020-10-08:
Ik zie dat de Duco box Silent en Focus ondersteund worden zou het ook werken op deze box Duco Energy Premium System

arnemauer, 2020-10-08:
Ík denk dat dat wel moet lukken. Heb je dezelfde aansluiting op de printplaat?

[image]

26miller, 2020-10-09:
Bord layout is wat anders bij mij maar connector zit er wel volgens mij zal van het weekend eens meten of het klopt.
[image]

arnemauer, 2020-10-09:
Ik ben benieuwd! De gateway is naar je onderweg.
Als het werkt voeg ik hem toe aan de lijst.

26miller, 2020-10-11:
Serial connection lijkt te werken ben alleen nog wat aan het stoeien met de data. Lijkt of niet alle data klopt met wat ik zie. Zal in de loop van de week nog wat verder puzzelen. Heb overigens ook een probleem met het ip van van domoticz in te voeren als controller deze valt steeds terug naar een default adres wat ik niet in geef., 
[image]

arnemauer, 2020-10-11:
Zie de waarschuwing in het rode kader. Het flashgeheugen kan je maar een
maximaal aantal keer beschrijven voordat de chip kapot gaat. Daarom zit er
een bescherming op om te voorkomen dat je dit te vaak doet. Daarvoor dient
de waarschuwing in het rode vak. Opnieuw starten van de gateway reset de
teller.

26miller, 2020-10-12:
Had al wel gelezen dat ik power van gateway moest halen maar heb dit waarschijnlijk te kort gedaan. Data komt binnen in domoticz nu. Vooralsnog alleen Ventilationmode en VentilationPercentage overige sensoren op box nog niet aan de praat gekregen.  Word vervolgd..

26miller, 2020-11-11:
RF gateway werkt geheel zoals beschreven, Met de serial gateway kan ik de ventilationmode percentage en de gekoppelde afstandbediening uitlezen (CO2 en Temp) de sensoren van de box zelf heb ik (nog) niet aan de praat gekregen, 
Serial gateway verliest af en toe connectie met box maar door gebrek aan tijd dit nog niet verder kunnen onderzoeken.

closed 2020-11-11 by 26miller