arnemauer/Ducobox-ESPEasy-Plugin #54: Tijdsduur handmatige stand

fwestenberg, 2020-10-23:
Allereerst hartelijk dank voor deze geweldige plugin! Ik gebruik hem nu al een ruimte tijd zonder problemen. 
Nu lees ik in de handleiding van de afstandsbediening, dat door 1,2 of 3x op een knop te drukken, je de tijdsduur kunt instellen voor de tijdelijke stand:

[image]

Is het mogelijk dit in te bouwen in deze plugin?

arnemauer, 2020-11-10:
Volgens mij houdt de Ducobox zelf de tijdsduur bij. De afstandsbediening geeft niet iets extra's door als je vaker op een standenknop drukt. Je zou de gateway op dezelfde wijze kunnen aansturen, dus door 3 keer een commando te sturen voor een bepaalde stand.

Wil je de tijdsduur van een stand permanent aanpassen, dan kun je met behulp van de Duco Network Tool de tijdsduur wijzigen. Je kan daarbij per afstandsbediening de tijdsduur instellen. Je kan bijvoorbeeld de afstandsbediening in de badkamer standaard op 30 minuten instellen i.p.v. 15.

fwestenberg, 2021-03-13:
Bedankt nog voor je commentaar. Ik heb een poosje verschillende dingen geprobeerd, maar tot nu toe leidt het 3x doorgeven van een commando niet tot een langere hoog-stand. Dit is eigenlijk wel mijn wens, heb je nog ideeën hoe ik daar achter kan komen?

arnemauer, 2021-03-17:
Ik heb opnieuw gekeken naar wat de afstandsbediening verstuurt als je vaker op een knop drukt. Wanneer je vaker op een knop drukt dan stuurt de afstandsbediening een bericht naar de Ducobox met daarin het aantal keer dat er op een knop gedrukt is. Binnen 1-2 weken heb ik de code aangepast zodat dit ondersteunt wordt door de RF plugin.

fwestenberg, 2021-03-17:
Dat is gaaf om te horen! Ik draai deze code momenteel op een Wemos D1 mini, met ESPHome. Het grootste deel van de library gebruik ik hiervoor. Ik zal de code binnenkort op Github zetten. Ik zag (maar dat kan dus aan mijn ESPHome port liggen) dat bij het instellen van een percentage, de box meteen weer terug gaat naar de ingestelde stand. Alleen als ik continu bijvoorbeeld 100% stuur, hoor ik de box versnellen. Enig idee? En kan ik ergens het percentage uitlezen? Bedankt voor de support!

Jozziej, 2021-08-22:
> Dat is gaaf om te horen! Ik draai deze code momenteel op een Wemos D1 mini, met ESPHome. Het grootste deel van de library gebruik ik hiervoor. Ik zal de code binnenkort op Github zetten. Ik zag [...]

Hi fwestenberg,

Ik ben benieuwd naar jouw code voor EPSHome, ik zag die nog niet op jouw Github staan.
Zou je die willen posten?

TimmiT93, 2022-01-30:
Dit klinkt inderdaad heel tof! @fwestenberg zou je bereid zijn om jouw implementatie op GitHub te zetten? 😊

fwestenberg, 2022-01-31:
Ik heb problemen met de esp/cc1101, waardoor ik alleen nog maar informatie ontvang. Ik heb nu mijn draadloze afstandsbediening van Duco rechtstreeks op de ESP aangesloten (batterij vervangen door voeding vanaf de ESP) en de knoppen stuur ik met optocouplers via GPIO. Simpeler wat mij betreft en ik ben van de problemen af met de cc1101.

[Hier](https://github.com/fwestenberg/duco) kan je het project vinden, dit is waar ik ben geëindigd. Succes er mee!

closed 2022-04-06 by arnemauer

Theoreet, 2022-04-07:
> Functie is geïmplanteerd in: ec42870

Mooi nieuws weer, kan hier ook een compilede (is dit NL's? :) )bin van
worden geupload ?(in de bin folder nog niet terug te vinden)

dicksmits, 2024-11-06:
> Ik heb problemen met de esp/cc1101, waardoor ik alleen nog maar informatie ontvang. Ik heb nu mijn draadloze afstandsbediening van Duco rechtstreeks op de ESP aangesloten (batterij vervangen door [...]

Hoi, ik wil proberen de ESPHOME port op te pakken, maar deze reposotory is verdwenen. help

fwestenberg, 2024-11-06:
> Hoi, ik wil proberen de ESPHOME port op te pakken, maar deze reposotory is verdwenen. help

Sorry, ik heb die ook niet meer..