arnemauer/Ducobox-ESPEasy-Plugin #55: Serial Gateway waardes uitlezen via home assistant

sanderlv, 2020-10-28:
Hallo, onderstaand 2 screenshots van mijn setup:
[image]

Via MQTT kan ik (via Node-red diverse waarden sturen en dit werkt top! (RF gateway):
[image]

Maar nu wil ik graag de "output" (Serial gateway) waardes in home assistant krijgen... hoe kan ik deze vanuit de ESPeasy laten versturen? Via MQTT?

[image]

De ESP easy is nog niet aangesloten op de dudo (kabel/stekkertje besteld) en dus staan de waarden op 0.0 (zie screenshot).. worden deze waarden als ze gevuld worden naar MQTT gestuurd?

[image]

sanderlv, 2020-10-31:
Ok ,heb mijn output in de ESP, maar hoe krijg ik deze in home assistant? Ik zie ze niet via MQTT gepublished worden....

[image]

sanderlv, 2020-11-01:
Ok, ontdekt (denk ik):
[image]

MQTT explorer:
[image]

Rest nog 1 belangrijk punt... de waarden worden niet geupdate...

arnemauer, 2020-11-01:
Beste Sander, vanwege een verhuizing heb ik geen tijd gehad om te reageren.
Ik probeer er deze week naar te kijken :)

>

sanderlv, 2020-11-01:
Superbedankt alvast! Dit was de laatste hindernis in mijn homeassistant avontuur

sanderlv, 2020-11-09:
> 
> Beste Sander, vanwege een verhuizing heb ik geen tijd gehad om te reageren. Ik probeer er deze week naar te kijken :)
> […](#)

Friendly reminder :-)

arnemauer, 2020-11-10:
Wat bedoel je met de waarden worden niet geupdate? Bedoel je daarmee de waarden van de plugin (espeasy devices pagina) of bedoel je dat de waarden niet naar mqtt worden gestuurd?

Zou je een screenshot kunnen maken van de plugin configuratiepagina?

sanderlv, 2020-11-10:
Ik bedoel de waarden van de duco via serial naar de esp. Mqtt werkt goed. Welke screenshot bedoel je? Die van hierboven?

26miller, 2020-11-11:
Ik zie dit fenomeen ook zo af en toe wat bij mij helpt is het device even op disable te zetten en vervolgens weer op enable.
Ben er nog niet achter wat dit veroorzaakt overigens.

sanderlv, 2020-11-11:
Ik heb het maar 1 keer "gevuld" gehad. Ik heb nu net alles disbaled, minuut gewacht, enabled en dat doet niets...

[image]

arnemauer, 2020-11-11:
Je kan het beste stap voor stap plugins activeren.
- disable alle serial plugins
- enable alleen de ventilation_mode_main_plugin
- om meer informatie terug te zien in de logs kun je "Web Log Level" op DEBUG zetten (Tools -> Advanced Settings)
- kun je +- 2 minuten van de weblog hier posten? (Tools -> Log).

sanderlv, 2020-11-11:
OMG 👎 Ik dacht even de DUCO uit e naa te zetten of dit werkt, en de fysieke situatie te beoordelen... en ik zie dat het draadje los ligt. Volledig doorgesneden door de deksel, die heeft de draad dus bij montage (vastklikken) volledig doorgesneden... Gemaakt en nu kome de waarden binnen... (shame).

Iedereen super bedankt voor de tijd en reacties! Nu begint het spelen...

sanderlv, 2020-11-11:
Echt heel erg bedankt Arne! Dit is heel fijn om te weten/hebben/gebruiken!

nu kan ik de fan in de nacht bij een percentage vocht >65 UIT zetten ❤️ 

Het resultaat:
[image]

[image]

arnemauer, 2020-11-11:
Goed om te horen! Onderin de Ducobox zitten twee wartels waardoor je de kabel kan trekken. De voedingskabel zit door een van de wartels. De andere wartel is vrij :)

sanderlv, 2020-11-11:
Die wartels zitten bij mij aan de zijkant. ook de printplaat ziet er anders uit dan jouw foto.

Ik heb een heel klein gaatje gemaakt en daar kabel doorheen...

JohCo-hassio, 2021-02-02:
Hey Sanderlv, kunnen we kijken of wij met elkaar in contact kunnen komen, ik heb vandaag de gateway binnen gekregen.
Dit heb ik allemaal aan de gang gekregen gelukkig.
Alleen heb ik mega weinig verstand van MQTT naar home assistant enz (hier nog niet eerder iets mee gedaan) 
Home assistant ken ik wel goed :)
En ben ook heel benieuwd naar je node-red flow (beter goed gedeeld als slecht bedacht) hahaha

Gr Johan

sanderlv, 2021-02-02:
Wat is je vraag?

JohCo-hassio, 2021-02-02:
Hoe heb jij de gegevens in home assistant gekregen ik heb de MQTT via intergrations gekoppeld
En zou jij je node-red flow kunnen delen.

Misschien kunnen we even mailen rechtstreeks johan.doorstruur@gmail,com

JohCo-hassio, 2021-02-02:
johan.doorstuur@gmail.com email verkeerd

sanderlv, 2021-02-02:
Je zult een mqtt server moeren installeren. Ik weet niet hoe en welke installatie methode je hebt gekosen maar mocht je de mogelijkheid hebben via supervisor een addon te installeren moet je de mosquitto matt server installeren. Dáár laat je dan de esp van arne naartoe sturen. Wanneer je de mqtt integratie heb aangezet zouden alle topics automatisch moeten worden geïntegreerd. Er is heel veel informatie op community.home-assistant.io en op de manual pages van homeassistant te vinden.

closed 2021-08-04 by arnemauer