arnemauer/Ducobox-ESPEasy-Plugin #58: Interval

sanderlv, 2020-11-11:
Hallo Arne,

is het mogelijk om de interval "hoger dan 60" seconden te zetten? Bijvoorbeeld bij het aftellen van de automatische stand?

[image]

Bij veranderen naar 2 sec gebeurt er eigenlijk niets...

arnemauer, 2020-11-11:
Interval heeft betrekking op hoe vaak data naar de "controller" (domoticz, mqtt, etc) wordt gestuurd. Niet op de afteltijd van de Ducobox. De afteltijd stel je per afstandsbediening op de Ducobox zelf in.

sanderlv, 2020-11-11:
Dank je. Iets duidelijker. "hoe vaak" stuurt de duco dan een update over serial naar de esp? Is dit te "versnellen"? Zodat ik de afteltimer wat meer zie "tellen"?

arnemauer, 2020-11-11:
De gateway vraagt op basis van het interval de informatie op bij de ducobox. Ik zou de interval niet te hoog zetten, het opvragen van informatie kost namelijk 1-2 seconden. Als je de interval op 2 seconden zet dan blokkeert de plugin het uitlezen van de andere seriële plugins.

sanderlv, 2020-11-11:
Dus als ik hem op 10 seconden zet, dan vraagt de esp om de 10 seconden de waarden op? en stuurt deze om de 10 seconden naar mqtt?

arnemauer, 2020-12-28:
Klopt, maar ik zou persoonlijk 60 seconden aanhouden zodat de gateway stabiel blijft.

closed 2021-08-04 by arnemauer