arnemauer/Ducobox-ESPEasy-Plugin #60: [P150] DUCO RF GW: Initialisation -> calibration failed. No response from CC1101 or status not idle

lansi002, 2020-11-20:
Hoi Arne en anderen,

Ik krijg een foutmelding bij het activeren van de RF gateway in ESPEasy.

[P150] DUCO RF GW: Initialisation -> calibration failed. No response from CC1101 or status not idle

Het lukt mij dan ook niet te  pairen met de DucoBox in installatiemode. Er verschijnt geen Join knop.
Het blijft alsvolgt:
[image]

Ik heb alle bedrading dubbel gecheckt en doorgemeten. 

Hier mijn nog opstelling voor wat het waard is in dit stadium van mijn vraag:
[image]

[image]

Ik gebruik de volgende bin file op een nodemcu amico:
20200629 Flasher firmware 281cde1 29-06-2020 (ESPEasy mega-20200515)

De Serial Gateway werkt overigens al wel goed met de DucoBox.

Hopelijk kunnen je/jullie me beetje helpen om wat dingen te controleren.

Groet,
André

arnemauer, 2020-11-21:
Hoi André,

De melding "[P150] DUCO RF GW: Initialisation -> calibration failed. No response from CC1101 or status not idle" betekent dat de gateway niet kan communiceren met de CC1101 module. 

Als ik je foto bekijk valt mij op dat je de mosi (pin 3) op de CC1101 hebt aangesloten op 3V3. Heb je onderstaand schema gebruikt bij het aansluiten?

[image]

lansi002, 2020-11-22:
Hoi Arne,

Bedankt voor je reactie. Ik waardeer dat enorm.

Ik heb inderdaad dat schema gebruikt om de CC1101 aan te sluiten. Dit heb ik ook nog gecheckt met jou aansluitschema's (mappingstabel) en DucoGWv3 schematic.pdf.

Ik snap de verwarring, maar de 4 rechter draden (resp. zwart, rood, geel en oranje) gaan naar de seriële verbinding met de DucoBox.
He is de oranje op pin D7 die naar de CC1101 / MOSI gaat (en dus niet die op 3v3 zit op de Nodemcu).

Ik schrok al, dat ik er overeen had gekeken. Jammer dat wel, want had we het opgelost.

Hier mijn vertaalde kladje, maar is wel moeilijk leesbaar denk ik.
NB de paarse draad is niet in gebruik, maar voor het gemak op D2 aangesloten, anders bungelt die zo los.

[image]

lansi002, 2020-11-23:
Ik heb net een meting gedaan (mini oscilloscoop) op de signalen D1, D5, D6, D7, D8 bij het enablen van de RF plugin in ESPEasy. Zonder aangesloten CC101.
Dit omdat de plugin dan gelijk de CC1101 probeert te initialiseren.

Ik zie alleen wat bits voorbij komen op D5 en D8.
Op D6 en D7 zie ik niks gebeuren.

Oftewel de plugin lijkt wel iets te doen.

Ik vermoed dat de plugin vervolgens wacht op antwoord van de CC1101 alvorens weer wat te doen.

Vanavond de CC1101 erbij aansluiten en kijken of ik meer bits voorbij zie komen. Het is een beetje lastig meten vandaar dat ik dat niet gelijk mee heb genomen, maar ik wilde je dit alvast delen.

arnemauer, 2020-11-23:
Op D7 (MOSI) zou je ook zeker activiteit moeten zien. Dit zijn de bytes die je zou moeten terug zien op D7: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/3ddd3aa88b2ff08027076651f576b567183e834f/lib/Duco/DucoCC1101.cpp#L82

lansi002, 2020-11-23:
Is dat data van MOSI naar D7 als reactie op activiteit op D5 / D8 ?

arnemauer, 2020-11-23:
Nee, zie hieronder:

**D5 -> SCK**  = dit is het clock signaal dat er voor zorgt dat de master en de slave op dezelfde snelheid (de bitrate) met elkaar communiceren. Anders begrijpen ze elkaar niet.
**D6 -> MISO** = Master In Slave Out en is de verbdining waarmee de CC1101 communiceert met de ESP8266
**D7 -> MOSI** = Master Out Slave In en is de verbinding waarmee de ESP8266 met de CC1101 communiceert
**D8 -> CS =**   Chip Select (CS) hiermee wordt aangegeven met welke slave de master wil communiceren. Deze blijft laag zolang de ESP8266 communiceert met de CC1101.

lansi002, 2020-11-23:
Ik heb nu alle pulsen gemeten met CC1101 aangesloten.

Bij het activeren van de RF plugin in ESPEasy zie ik dan het volgende:
D1 geen pulsen	(standaard hoog)
D5 wel pulsen 	(standaard laag)
D6 wel pulsen  	(standaard laag)
D7 geen pulsen 	(standaard laag)
D8 wel pulsen 	(standaard hoog) 

Ik vind het raar dat er niks over D7 gaat. Dat is toch de initialisatiedata die naar de CC1101 zou moeten gaan ?

Waar is D1 trouwens voor ? Is dat een interupt signaal van de CC1101, zodra er RF-remotesignaal ontvangen wordt door de CC1101 ?

Gr
André

arnemauer, 2020-11-24:
Dat je signalen ziet op D6 is inderdaad een goede indicatie dat de CC1101 wel reageert. De clock op D5 zorgt ervoor dat de CC1101 een reactie geeft op D6.

lansi002, 2020-11-24:
Tadaaaa:

[image]

Ik heb Init SPI aangezet --> reboot ESPEasy --> klaar.

Dadelijk even met de DucoBox proberen.

closed 2020-12-28 by arnemauer

Sef1995, 2021-03-23:
@lansi002 Heb je misschien een linkje naar waar je die draadjes hebt gekocht die je aan de cc1101 heb gesoldeerd, aan de andere kant heb je het draadje vastgezet aan een female connector?