arnemauer/Ducobox-ESPEasy-Plugin #61: Not getting CO2 or RH values

cpolhout, 2020-12-01:
I configured P152 and P153 serial devices. I see in the serial log the response from my Focus, but the values are not set on the device. 

Please see the log:
[image]

And the devices:
[image]

cpolhout, 2020-12-01:
Enabled serial-debug-logging.  Now it says invalid response:

[image]

lansi002, 2020-12-06:
Just to be sure:
Are the specific Ducobox CO2 and RH sensors connected ?

Like this:
https://www.duco.eu/nl/producten/mechanische-ventilatie/sturingscomponenten/boxsensor
or a Ducobox remote control with CO2 or RH sensors

cpolhout, 2020-12-06:
Hi Ian,

Yes, they are added. I also connected with a usb2serial cable and used the Duco-software to configure the sensors. See screenshots beneath. I tried to use the serial-gateway option from the ESP, but that did not work also.

[image]
[image]

lansi002, 2020-12-09:
Ik had het ook kunnen zien in je ESPeasy log, want er komen tenslotte 'Pakket ontvangen' voorbij.
Helaas heb ik zelf geen sensoren dus ik kan niet vergelijken wat er bij jou fout gaat.

cpolhout, 2020-12-10:
@arnemauer Do you have an idea what might be wrong? I'm using the pre-build gateway from you.

arnemauer, 2020-12-28:
De CO2 en RH sensors die je hebt zijn geen boxsensors. Vandaar dat je geen waardes krijgt.

Dit is het antwoord van je Ducobox: 20205b53454e534f525d20494E464F = "   [SENSOR] INFO"
Oftewel, geen boxsensoren.

cpolhout, 2020-12-30:
Ah. Ok. Ik heb inderdaad kleppen met een sensor en geen losse sensoren. Ik wist niet dat daar verschil in zat. Zie bijgaand de logfile.

Vast bedankt voor het uitzoeken!
[DD_201230_Wednesday.log](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/5754731/DD_201230_Wednesday.log)

Joost73, 2021-04-19:
@arnemauer , ik heb idd een boxsensor voor vocht.
Ik weet dat de boxsensor alleen het vochtpercentage meet en niet de temperatuur. Waarom staat de temperatuur dan wel bij de config? Daarnaast krijg ik ook geen waarde te zien. in de log staat: 

[P153] DUCO BOX SENSOR: start read, eventid:153384887: 
[P153] DUCO BOX SENSOR: Serial port in use, set flag to read data later.

Enig idee hoe ik dit kan oplossen?

[image]

arnemauer, 2021-08-04:
> Ah. Ok. Ik heb inderdaad kleppen met een sensor en geen losse sensoren. Ik wist niet dat daar verschil in zat. Zie bijgaand de logfile.
> 
> Vast bedankt voor het uitzoeken!
> [DD_201230_Wednesday.log](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/5754731/DD_201230_Wednesday.log)

Ik kan in de log helaas niet het commando vinden voor het opvragen van de luchtvochtigheid van de klep. Zou je de stappen nog een keer willen proberen? Om er zeker van te zijn dat alle informatie in de log staat zou je een aantal keer op refresh all kunnen klikken.

arnemauer, 2022-10-25:
Waarschijnlijk lost de nieuwste firmware je probleem op: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/4e4d3791da5bb5204a622e2e6e2d3380ea2f31c2

closed 2022-10-25 by arnemauer