arnemauer/Ducobox-ESPEasy-Plugin #62: SERGW_VENTILATION_PERCENTAGE 100% with SERGW_VENTILATION_MODE_MAIN = 0.00

lansi002, 2020-12-09:
Als ik de ventilatiemode instel op automatisch gaat na verloop van tijd de Ducobox naar ventilatie percentage van 100% en ventilatie moden naar 0.00

Ik gebruik geen sensoren.

Dat ziet er tegenstrijdig uit.

[image]

arnemauer, 2020-12-28:
Hoe zet je de ducobox op ventilatiemodus automatisch? Via de RF-plugin of via een bedieningsschakelaar?
Kun je de weblog hier posten? Dan kunnen we zien wat er gebeurt.

lansi002, 2020-12-29:
Ik zet de DUCO op 3 manieren op AUTO:
- Via de bedieningschakelaar
- Via de RF-plugin (
- Via Domoticz met http commanda's wat weer via de RF-plugin loopt.
[image]

De DUCO gaat, sinds ik deze geactiveerd heb met de ESPEasy plugin, in alle 3 de gevallen na verloop van tijd naar Veltilation_Percentage 100%.

Ik heb nu de ESPEasy module losgekoppeld van de DucoBox en nu gaat de Duco nog steeds naar een hoge ventilatie modus. Ik kan niet zien of het 100% is, want ik gebruik de Networktool (nog) niet, maar ik vermoed van wel. Ik hoor hem flink zuigen.
Als ik de DucoBox op Permanent Low zet dan gaat de Duco niet naar een hoge ventilatiestand. 
Het lijkt er dus op dat de DucoBox dit nu zelf doet alleen in de AUTO stand. Dat deed hij voordat ik via installatiemode geactiveerd heb dus niet.

Kan ik met de Networktool misschien nog meer informatie bovenkrijgen denk je ?

arnemauer, 2020-12-29:
Als er een te hoog CO2 gehalte is bij een CO2 bedieningsschakelaar dan zal hij in stand auto harder gaan afzuigen. 
Als je aan de onderkant van je bedieningsschakelaar kijkt zie je bij de CO2 versie als het goed is een lange gleuf zitten. 

De snelste manier om erachter te komen waarom hij de ventilatiestand hoger zet is om de Duco Network Tool te gebruiken.

lansi002, 2020-12-29:
Ok ik zal een usb kabeltje maken.
Ik heb nog een FTDI232 liggen, die werkt met chipset/driver CH341, herkent de Duco dit ook denk je ?
In je wiki lees ik dat de Duco naar de CP210X chip zoekt namelijk.

arnemauer, 2020-12-30:
> Herkent de Duco deze FTDI ook denk je ? Deze heeft een andere chipset (FT232RL)

Nee de software gaat dit niet automatisch vinden. Je kan proberen om de naam van de driver aan te passen maar dat is behoorlijk lastig (als je windows het uberhaupt accepteert).

> In je wiki heb je het over:
> _In Ventilation Gateway versie 5 en hoger zit een extra chip waarmee de gateway zich kan gedragen als usb kabel die je voor de Duco Network Tool kan gebruiken. Je kan deze functie inschakelen op de configuratiepagina van de Serial Gateway-plugin."_
> 
> Kan ik de NodeMCU met ESPEasy-Serial-Gateway gebruiken als serial gateway naar de Duco Network tool ? Waarbij ik dan het volgende vinkje aanzet: "Disable gateway serial to use gateway as usb-cable for network tool" en de USB kabel in de PC en NodeMCU prik ?

Volgens mij zit er op de NodeMCU een cp2102 chipset. Je kan het daarmee proberen. Je moet dan wel in espeasy op de hardwarepagina de RX (GPIO-3) en TX (GPIO-1) op input zetten. Let op: de TX van je nodeMCU sluit je aan op de TX van je ducobox en de RX van je nodeMCU sluit je aan op de RX van je Ducobox. Zet eerst die gpio pinnen op input VOORDAT je de kabel in je Ducobox aansluit.

lansi002, 2021-01-06:
Ok ik ben eruit.

Ik kreeg het even niet makkelijk aan de praat met de Ducobox seriele uitlezing en de Ducobox tool.
Had ermee te maken dat ik een te lange USB kabel nodig had vanwege het ontbreken van een laptop.

Daarom heb ik maar navraag gedaan bij mijn installateur of ik een CO2 sensor in mijn woonkamer remote heb. 
Remotes met of zonder CO2 sensor kunnen dezelfde behuizing hebben namelijk (incl. een opening aan de onderkant).

Het blijkt dat ik wel een CO2 sensor heb en aangezien het wat kouder is geworden en mijn partner daarom wat roosters heeft dichtgedaan was daardoor het CO2 gehalte in de woonkamer gestegen, waardoor de Ducobox in de AUTO stand flink op 100% zat te draaien.  En dat nou net terwijl ik de Ducobox gateway geinstalleerd had.

Ik heb een aparte CO2 MH-Z19 met ESPeasy bij de Duco remote gehangen en inmiddels is het CO2 gehalte wat gedaald en is de Ducobox wat rustiger geworden in AUTO stand.

Dus lag niet aan de Ducobox gateway. Case closed.

Desondanks bedankt voor je hulp Arne.

closed 2021-08-04 by arnemauer