arnemauer/Ducobox-ESPEasy-Plugin #63: Joinen met duco box werkt niet

Carlozzz92, 2020-12-18:
Ik heb een Wemos D1 mini met een CC1101 module die ik wil koppelen aan mijn Ducobox.
De firmware die ik gebruik is: 20200629 Firmware 281cde1 29-06-2020 (ESPEasy mega-20200515)

Na het activeren van de installermode via de duco box heb ik op de join knop gedrukt. (Groene ledje knippert)
In de log verschijnen de volgende regels, maar de wemos met CC1101 maakt geen verbinding met de duco box.
Klopt het dat de GDO0 pin van de CC1101 niet met de wemos verbonden hoeft te worden?

[image]
[image]

Carlozzz92, 2020-12-27:
Does anyone know a solution? how to connect to the duco box?

arnemauer, 2020-12-28:
Voor zover ik kan zien wordt er wel een join bericht verstuurd. Daarna volgt er geen reactie van de Ducobox.
Er is wel vrij snel radio activiteit (ongeveer 60ms na het versturen van het join pakket) maar dat het pakket wordt genegeerd omdat het niet goed is ontvangen (Ignoring RF noise). 

Wat is de afstand tussen de gateway en de Ducobox?

Carlozzz92, 2020-12-29:
Ik heb het vanmorgen op verschillende afstanden geprobeerd (1 meter vanaf de duco box en een verdieping onder de ducobox) maar beide keren weer met hetzelfde resultaat, dat het signaal niet goed wordt ontvangen(Ignoring RF noise).

arnemauer, 2020-12-29:
> Klopt het dat de GDO0 pin van de CC1101 niet met de wemos verbonden hoeft te worden?
GO0 hoeft inderdaad niet aangesloten te worden.

Als je de RF-bedieningsschakelaar gebruikt zie je dan wel berichten langskomen?
Heb je een CC1101 module voor 868mhz of 433mhz?

Carlozzz92, 2020-12-29:
Ik heb een CC1101 module voor 868mhz. ik zie niks voorbijkomen als ik de RF-schakelaar gebruik.

arnemauer, 2020-12-30:
Zou je een foto willen posten van hoe je alles hebt aangesloten?

Carlozzz92, 2021-01-01:
[image]
[image]

Predjee, 2021-01-04:
Ik kan exact hetzelfde doen qua foto's met exact dezelfde aansluiting en hetzelfde "issue". Ik heb geen idee meer waar het nog fout gaat. Ik heb zelfs al bovenop de ducobox gezeten.

MichelFalke, 2021-01-05:
Ik heb het vandaag kunnen oplossen! Het issue was dat ik nog net twee kleine stukjes microscopisch groot tin tegen elkaar aan had zitten.

benvandijk84, 2021-01-17:
Hier hetzelfde probleem zie [closed issue.](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/48)
Heb een nieuwe besteld, aangesloten maar geeft hetzelfde resultaat helaas.

arnemauer, 2021-01-17:
Kun je een scherpe foto van je WEMOS d1 maken (van beide kanten)?

Wat staat er op dit onderdeel?
[image]

Carlozzz92, 2021-01-18:
Op het onderdeel in het rode vierkantje staat 4A2D (zie foto hieronder)

[image]

De weblog laat bij het drukken op Submit/Save op de Config pagina het onderstaande resultaat zien:

[image]

z6SVGEL9Q6a1, 2021-01-19:
Ik heb een gelijksoortig probleem;
- NodeMCU v3 
- TI-C1101 868mhz (andere chip dan op bovenstaande foto)
- AFM 281cde1: Jun 29 2020

------------ log bij joinen
1008597: [P150] DUCO RF GW: TIMER => 1008596 1008594 2
1008597: [P150] DUCO RF GW: Start of RF signal received
1008598: [P150] DUCO RF GW: Ignoring RF noise
1008877: [P150] DUCO RF GW: TIMER => 1008876 1008864 12
1008877: [P150] DUCO RF GW: Start of RF signal received
1008877: [P150] DUCO RF GW: Ignoring RF noise

------------ log bij saven
234666: Webserver args: 0: 'index' length: 1 1: 'page' length: 1
240611: SaveToFile: free stack: 3328
240674: FILE : Saved config.dat
240674: SaveToFile: free stack after: 3328
240675: SaveToFile: free stack: 2992
240687: FILE : Saved security.dat
240687: SaveToFile: free stack after: 2992
240701: [P150] DUCO RF GW: Extra Settings PLUGIN_150 loaded
240711: [P150] DUCO RF GW: Values set from config. DeviceID: 0, networkId: 0000, radio power: Default - radio power 0,6 dBm (0x8D)
240714: [P150] DUCO RF GW: CC1101 868Mhz transmitter initialized

arnemauer, 2021-01-21:
> Op het onderdeel in het rode vierkantje staat 4A2D (zie foto hieronder)
> 
> [image]

Waarschijnlijk heb je een clone via Aliexpress gekocht. Om het product zo goedkoop mogelijk te maken gaat zo'n chinees kijken waar hij de kosten kan drukken. De voltage regulator is daar een van.  De 4A2D voltage regulator levert maximaal 150mA. De ESP8266 gebruikt in piekmomenten 430mA. De CC1101 gebruikt tijdens het zenden 30mA.  Er is dus veel te weinig stroom om alles stabiel te voeden. Ik heb zelf deze modules ook gebruikt en het vervelende is dat het lijkt alsof ze prima werken maar na verloop van tijd ga je vreemde problemen ervaren (uitval, corrupte memory, reboots, etc.).

Zouden jullie de CC1101 kunnen voeden met een externe 3.3v voeding? En dan kijken of je de signalen van een afstandsbediening wel goed kunnen ontvangen?

lansi002, 2021-01-22:
Hallo Carloz,

Ik gebruik deze van aliexpress en werkt prima:
https://nl.aliexpress.com/item/4000264911320.html?spm=a2g0s.9042311.0.0.27424c4dq618Lh

Carlozzz92, 2021-02-04:
Ik heb een nieuwe Wemos D1 mini en een nieuwe CC1101 gekocht en nu werkt het inderdaad wel. De signalen van de draadloze bediening worden opgepakt en ik kan de Duco box nu ook bedienen vanuit de interface van de Wemos D1 mini.
Blijkbaar zat het probleem inderdaad in de hardware.

Sef1995, 2021-03-25:
@Carlozzz92 Zou je misschien een linkje kunnen sturen van de draden die je gebruikt hebt die je aan de CC1101 hebt gesoldeerd. Aan de andere kant van die draden heb je er ook connectors op zitten zie ik.. ik kan ze zelf nergens vinden.. ☺️

closed 2021-03-25 by arnemauer