arnemauer/Ducobox-ESPEasy-Plugin #65: Memory error

arcangel321, 2021-01-13:
I got an memory error.
[image]

setup:
Home assistant 
MQTT
Ventilation gateway 
Latest firmware

How can we solve this?

arnemauer, 2021-01-13:
Fixt een reboot het probleem? 
Wanneer treed het probleem op?

arcangel321, 2021-01-13:
Het apparaat reboot al elke dag om vast lopers te voorkomen. Het probleem treed op bij het openen van de webpagina.

Zover ik kan zien staat debug uit.

mpepping, 2022-04-09:
Same here. The memory error occurs after a couple of weeks without rebooting. The strange thing is that this is happening only since the past months, without any changes to the firmware. Maybe something changed on the MQTT broker side (Mosquitto via HA add-on) that affects the gateway. Can't really think of anything else.

Anyway, after a reboot, all memory thresholds seem quite OK:

```
Free Mem:            | 3992 (25160 - )
Free Stack:          | 3664 (3824 - )
Heap Max Free Block: | 2936
Heap Fragmentation:  |  25%
```

arnemauer, 2022-07-04:
Fixed :)

closed 2022-07-04 by arnemauer