arnemauer/Ducobox-ESPEasy-Plugin #66: Gateway bestellen

bartlauwers53, 2021-01-17:
Hi Arne, 

Het Google form waarmee je zogezegd een gateway kunt bestellen had ik ingevuld maar ik hoorde nog niks, dus nu even via deze weg. Heb je ze nog, verwacht je dat je er gaat maken, en indien ja, weet je wanneer dat ongeveer zou kunnen zijn? Mocht dat niet zo zijn: ook prima, dan ga ik zelf bekijken of ik aan de klus kan. 
Alvast dank voor je reactie! 
Groet, Bart.

arnemauer, 2021-01-17:
Beste Bart,

Klopt inderdaad dat je nog niets gehoord hebt. Het duurde lang voordat ik alle onderdelen binnen had.
Ik heb nu een aantal gateway's klaar. Morgen ga ik een e-mail sturen naar iedereen die het formulier heeft ingevuld.

Groet,

Arne

closed 2021-01-17 by arnemauer

bartlauwers53, 2021-01-17:
Beste Arne, 

Sorry voor mijn ongeduld, had nog net langer moeten wachten. Hartelijk dank voor je antwoord en goede bericht, en ik wacht je mail af. 

Met vriendelijke groet, Bart.