arnemauer/Ducobox-ESPEasy-Plugin #67: mDNS ondersteuning

lansi002, 2021-01-22:
Hallo

Ik gebruik de volgende firmware:
20200629 Flasher firmware 281cde1 29-06-2020 (ESPEasy mega-20200515)

Ondersteunt deze firmware ook mDNS ? Ik kan het niet vinden in de code.
Ik wil graag de NodeMCU bereiken middels hostname.

Ik kreeg deze reactie nog van TD-er ESPEasy zelf (Owner and lead developer of ESPeasy):
_Have you tried a build with mDNS enabled?
test_ESP8266_4M1M_VCC_MDNS_SD is too big lately so that PIO env. is not included in the latest builds.
But in the "custom" builds, the FEATURE_MDNS flag is set so it should be included._

arnemauer, 2021-01-22:
Hej,

Ik kan vanavond kijken of ik een binary kan maken voor je. Ik weet niet of ik dit in de standaard binaries mee kan nemen. Dat  hangt er vanaf hoeveel extra ruimte het inneemt :)

lansi002, 2021-01-23:
Hoi,

Een oplossing voor mij zou ook zijn dat ik met mqtt de stand kan bepalen van de Ducobox vanuit een Domoticz schakelaar.
Dus ipv de volgende requesten per stande een mqtt request:
[image]

lansi002, 2021-03-25:
Ik heb het als volgt opgelost:

[image]

SentMQTTFromDomoticz.sh:

#!/bin/sh
# aanroep vanuit Domoticz switch: script://SentMQTTFromDomoticz.sh ESPEasy_DucoBox/Ventilationmode 3
# param 1: topic
# param 2: message/payload
#echo $(date)" Start sending MQTT message" >> /home/pi/SentMQTTFromDomoticz.log
mosquitto_pub -t $1 -m $2
#echo $(date)" Done " $1 >> /home/pi/SentMQTTFromDomoticz.log

lansi002, 2021-03-25:
Nog een tip:

Vanaf een andere ESPEasy met humidity sensor in de badkamer kun je bijvoorbeeld de Duco automatisch op de hoogste stand zetten, zonder zelf op de remote van de Duco te drukken (dus geen batterij wissel meer nodig van deze remotes).

ESPEasy device:
[image]

[image]

ESPEasy rules:
[image]

ESPEasy zelf op een Wemos D1 mini:
[image]

closed 2022-04-09 by arnemauer