arnemauer/Ducobox-ESPEasy-Plugin #68: Reset gateway?

bartlauwers53, 2021-04-02:
Hi 

Ik heb sinds enige tijd de Gateway in huis, en gekoppeld(via kabel en radio) met mijn Duco: bedienen en uitlezen van status en verschillende waarden via webinterface lukt. Ik heb daarnaast HomeAssistant op een Rasberry Pi draaien. Die heb ik ook geconfigureerd onder Controllers (Home Assistant (openHAB) MQTT en tevens bij de plugin Send to Controller aangevinkt. Ik krijg in de MQTT log van HA wel te zien dat het device contact heeft, maar ik zie verder niks automatisch toegevoegd worden binnen HA. Ik was ook al met een MQTT scanner bezig geweest en zie wel berichten voorbij komen, maar HA pakt die blijkbaar niet op. 
Ik ben nog aardig onbekend met HA dus ik doe vast iets fout: kunnen jullie mij hierbij helpen? Bij mijn shelly-devices zit er automatische publish op zodat HA weet wat ie er mee kan. Ik weet niet of dat ook voor de Gateway geldt? 
Wat ik sowieso wel een optie vind (ik heb er inmiddels al erg veel mee lopen spelen) is om de Gateway te resetten. Ik weet alleen niet waar ik dat moet doen, en of ik dan eventueel meer kwijt ben dan me lief is. Ook met ESP Easy ben ik namelijk nieuw. 

Alvast bedankt.

arnemauer, 2021-04-08:
Hoi Bart,

Ik heb zelf geen ervaring met HASS. Ik ga binnenkort wel bezig met een overstap van Domoticz naar Hass. Als je data wel ziet binnenkomen in de logs van HASS dan denk ik dat de gateway wel goed werkt maar dat je in HASS nog wat moet configureren. Volgens mij voegt HASS niet automatisch een device toe (zoals dat bij esphome wel het geval is).

Heb je iets aan deze informatie? https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/24

Chrisiesmit93, 2021-04-08:
Als ik het goed heb, moet voor auto discovery via Home Assistant, de prefix "homeassistant" vanuit MQTT meegegeven worden, zelf niet getest, maar zoiets staat mij wel bij.
Eventueel meer info hierover hier: https://www.home-assistant.io/docs/mqtt/discovery

bartlauwers53, 2021-04-15:
Dank voor jullie antwoorden. Ik ben helaas even druk met gezin/werk momenteel maar ga er hopelijk binnenkort naar kijken. 
Wat ik verder in HA zie is alleen dat er verbinding is, komt geen data binnen. Maar dat moet ik nog eens verder uitzoeken. Ik kom hier nog hopelijk binnenkort op terug.

closed 2021-07-20 by arnemauer