arnemauer/Ducobox-ESPEasy-Plugin #69: ESPEasy hangs when sending command

Chrisiesmit93, 2021-04-08:
First, thanks for the great work!
The possibility to read the values of my Duco box is a nice feature to have! :)

I have 1 problem with the ESP8266 with this firmware on it:
when I send an command (auto, low, mid or high, does not matter what), the ESP hangs / crashes.
The GUI is not working anymore and also MQTT values are not updated anymore. Pinging the ESP is still working fine (the only part that is working).

Is this a common thing ?

First thing I thought of, is the power supply, but that does 2,3A over 5V so enough power to feed the ESP and the transmitter I think.
The ESP is connected both serial and wireless to the Duco Silent box.

arnemauer, 2021-04-08:
What kind of ESP8266 module did you use? Crashes often happen when the power supply is unstable or the voltage regulator is insufficient to power the peak usage of the esp8266.

Chrisiesmit93, 2021-04-08:
I have the "D1 WIFI Dev Kit" from here: https://nl.aliexpress.com/item/4000562770626.html

The wires are attached (now correctly, thx to you) as you can see at:
https://gathering.tweakers.net/forum/list_message/66535436#66535436

Chrisiesmit93, 2021-04-08:
NL is wel zo makkelijk inderdaad :)

Hierbij een 3 tal foto's van de DucoBox Silent + de 2 modules:
[image]
[image]
[image]

Het onderdeel geeft DE=AJ0 aan als ik het goed gelezen heb.

arnemauer, 2021-04-08:
Volgens mij is de markering "DE=A1D" en betreft het deze voltage regulator: http://www.txsemi.com/Upload/TX6211B_V12-11264794020.pdf

Chrisiesmit93, 2021-04-08:
Is volgens mij zo'n HEMA kabeltje die er nu tussen zit.

Kan ik anders de USB stekker aan de ESP zijde eraf knippen en daar zowel de ESP als die zender vanaf voeden?
Bypass van die voltage regulator dus, dan zou die meer prik moet krijgen (de volle 2,3A in dit geval).
Of sloop ik dan (eventueel op de lange termijn), de ESP module ?

arnemauer, 2021-04-09:
Zou je een ander korter kabeltje willen proberen?

De voltage regulator zit er wel met een reden :) Hij maakt van de 5v wat de usb poort levert 3.3V voor de ESP8266 en de CC1101 module. Als je die gaat voeden met 5V zijn ze direct kapot.

closed 2021-07-20 by arnemauer