arnemauer/Ducobox-ESPEasy-Plugin #70: External CO2 Sensor (P154) not returning values

mod2133, 2021-04-15:
I configured P154 plugin but am not seeing values (other than 0) for RH and CO2_PPM.
The DUCO Silent and RH sensor work fine.
Using Arne's standard gateway.

I'm pretty sure P154 is correct. this is what duco tools shows:
[image]
[image]

My configuration is like this:
[image]

Log shows this error:
[image]

That hex string decoded says something like: LOG DATA LOST (19)

I am confused by the TYPE: RH in the log as it is a CO2/Temp RF sensor.

Where do I go wrong?

arnemauer, 2021-04-15:
Als je "LOG DATA LOST (19)" tegenkomt dan is er iets niet goed gegaan bij het afsluiten van de Duco Network Tool. De Duco Network Tool zet de loglevel van de Ducobox aan om te kijken of er fouten in het RF verkeer zitten. Dat doet hij met het commando: coreloglevel 0xff.

Wanneer de Duco Network Tool niet goed wordt afgesloten (om wat voor reden dan ook) dan blijft de Ducobox logberichten sturen over de seriele poort. Je kan handmatig met het commando: "coreloglevel 0x10" dit uitschakelen of even de Duco Network Tool openen zodat die dat voor je doet.

mod2133, 2021-04-15:
Ik heb de stappen doorlopen. GW in USB modus gezet. Duco Network Tool gestart, data gefreshed en netjes gesloten.
Helaas geen verschil in de log.

Commando uitvoeren is niet gelukt. DNT in service mode gezet met PW en Commander gekozen. Alles is uitgegrijst, kan geen command invoeren.

NB al voordat ik Duco Network Tool überhaupt heb gebruikt kreeg ik deze melding al.

Zijn er nog andere troubleshoot stappen die ik kan nemen? Bijvoorbeeld log levels aanpassen?

arnemauer, 2021-04-15:
Als je de gateway in USB modus zet en je opent via Putty een seriële terminal dan kun je daar het commando intypen.

mod2133, 2021-04-15:
Met Putty en de baudrate op 115200 krijg ik een connectie met leesbare text. Echter het commando  'coreloglevel 0x10' geef CMD UNKNOWN. Vast een rookie fout maar ik ben onbekend met ESP commandos.

Maar eerlijk gezegd denk ik dat hier niet de fout zit, want het systeem werkt prima, de RF CO2/Temp bediening werkt en geeft aan Duco Netwerk Tools netjes de juiste waardes door.

arnemauer, 2021-04-15:
> Maar eerlijk gezegd denk ik dat hier niet de fout zit, want het systeem werkt prima, de RF CO2/Temp bediening werkt en geeft aan Duco Netwerk Tools netjes de juiste waardes door.

Het systeem werkt ook prima. Waar het om gaat is dat de Ducobox veel log-informatie doorgeeft op de seriële verbinding. Die log informatie verstuurd hij te pas en te onpas (dus ook tussen commando's door). Als de gateway informatie opvraagt over je CO2 waarde en tussen de waarde zet hij ineens "LOG DATA LOST (19)" dan kan de gateway daar niets mee.

Als je putty opent en verbinding maakt, ontvang je dan ook tekst van je Ducobox? Het kan overigens best zijn dat door de Duco Network Tool te openen dat logging nu uit staat :) Ontvang je nu wel waardes?

mod2133, 2021-04-16:
Dank voor het meedenken. Maar probleem is niet weg.

Wel ben ik verder met de seriële connectie. Misschien handig voor anderen wat ik moest doen:

- Gateway in USB Seriele mode zetten
- Putty config aanpassen: Speed (baud) 115200, data bits: 8, stop bits: 1, party: None, Flow Control: None, Implicit LF in every CR, implicit CR in every LF.
- Verbinding opzetten. Duco geeft antwoord!
- commando uitvoeren. Deze zijn Case Sensitive blijkbaar
- - help /all
- - Network
- - CoreLogLevel 0x10
- - NodeConfigGet 2
- - nodeparaget 2 22

Ik krijg overal keurig antwoord op.

[image]

[image]

- Gateway weer terug in normale mode
- Helaas geen waardes voor P154 (=0)
- Log blijft zeggen:
1026399: [P156] DUCO FanSpeed: start read, eventid:4
1026400: [P156] DUCO FanSpeed: Start reading fanspeed
1026575: [P156] DUCO FanSpeed: Received correct response on fanspeed
1026581: [P156] DUCO FanSpeed: Fanspeed: 385 RPM
1026582: [P156] DUCO FanSpeed: End of message received.
1026582: [P156] DUCO FanSpeed: Stop reading from serial
1027399: [P153] DUCO BOX SENSOR: start read, eventid:5
1027399: [P153] DUCO BOX SENSOR: Start readBoxSensors
1027615: [P153] DUCO BOX SENSOR: Received correct response on sensorinfo
1027618: [P153] DUCO BOX SENSOR: End of message received.
1027618: [P153] DUCO BOX SENSOR: Stop reading from serial
1028399: [P154] DUCO Ext. CO2 Sensor: start read, eventid:6
1028399: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
1028400: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: RH
1028421: [P154] DUCO Ext. CO2 Sensor: ROW:1 bytes:32
1028423: [P154] DUCO Ext. CO2 Sensor: Pakket ontvangen: 313032383434646F782032656F204C4F472044415441204C4F53542028313929 END
1028424: [P154] DUCO Ext. CO2 Sensor: Received invalid response
1028424: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
1111956: WD : Uptime 19 ConnectFailures 0 FreeMem 9408 WiFiStatus 3

Is de configuratie van de plugin wel goed? Ik blijf Type: RH vreemd vinden voor P154 die geen Relative Humidity meet.

arnemauer, 2021-04-19:
Ik ben erin gedoken. Er staat een foutje in de code waardoor je in de logs RH ziet i.p.v. CO2 en andersom.
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/3ddd3aa88b2ff08027076651f576b567183e834f/_P154_DucoSerialExternalCO2Sensor.ino#L258

Hij leest dus wel de juiste waarden (co2) uit. Aangezien je in het antwoord LOG DATA LOST terugziet wil ik je vragen om de RF gateway tijdelijk uit te schakelen. Zie je dan wel de juiste waardes? Zo niet, kijk ik graag met je mee via teamviewer.

mod2133, 2021-04-19:
Ok. Thx. Ik gebruik de GW nu alleen met de serial verbinding. Maw ik heb nog helemaal niet de RF kant van de gateway geconfigureerd. Dat is wat je bedoelt toch?

arnemauer, 2021-04-19:
Klopt, het enige wat ik mij dan nog kan bedenken is dat de logging nog aanstaat. 
Wanneer je via putty een terminal opent zie je (na een tijdje) geen LOG DATA LOST terug?

mod2133, 2021-04-19:
Ik zie als ik met Putty verbinding maak iedere keer inderdaad een DATA LOST bericht verschijnen. 
Een (1) keer en daarna niet meer. Ik dan wel de CO2 waarde uitlezen met NodeParaGet 2 74

arnemauer, 2021-04-19:
Stoppen die meldingen ook na het invoeren van coreloglevel 0x10 of coreloglevel 0x00?

mod2133, 2021-04-19:
Als ik CoreLogLevel 0x00 stopt de terminal met responden en lijkt de GW ook geen waardes meer te ontvangen. Met CoreLogLevel 0x10 herstelt de oude situatie zich weer maar dus geen waardes van de externe RF sensor.

Wat me wel opvalt dat als ik in Devices de sensor op node 3 zet (hij zit op 2) dan krijg ik ongeveer dezelfde fouten in de log (dus ook de data lost).

Hoe kom ik er achter welke commando's door de plugins echt worden uitgevoerd? Als ik de log level op Serial naar Debug of lager zet stopt de GW ook met het ophalen van waardes.

mod2133, 2021-04-19:
Ik heb de oorzaak gevonden, lijkt het. Als ik de Serial Log level op None zet krijg ik CO2 levels terug. Case closed. ;-)

closed 2021-04-19 by mod2133