arnemauer/Ducobox-ESPEasy-Plugin #71: Geen waardes van boxsensor (RH)

arnemauer, 2021-04-19:
Ik weet dat de boxsensor alleen het vochtpercentage meet en niet de temperatuur. Waarom staat de temperatuur dan wel bij de config? Daarnaast krijg ik ook geen waarde te zien. in de log staat: 

[P153] DUCO BOX SENSOR: start read, eventid:153384887: 
[P153] DUCO BOX SENSOR: Serial port in use, set flag to read data later.

Enig idee hoe ik dit kan oplossen?

[image]

_Originally posted by @Joost73 in https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/61#issuecomment-822404743_

arnemauer, 2021-04-19:
@Joost73
Het is handig als je aangeeft om welke boxsensor het gaat. Ik ga er vanuit dat je een CO2 boxsensor hebt.
De CO2 boxsensor heeft inderdaad geen temperatuurmeter aan boord. De waarde humidity kun je daarom ook negeren. 
Dat deze er wel staat heeft te maken met de manier waarop ESPeasy in elkaar zit. ESPeasy kent geen losse sensortype voor humidity. Die is er alleen in combinatie met temperatuur: 

https://github.com/letscontrolit/ESPEasy/blob/7e7c6020721da9ceabb9d6d03bddb84e0751f2bd/src/src/DataStructs/DeviceStruct.h#L31-L48

Als je een CO2 boxsensor kun je de waarde voor humidity gewoon negeren.

Joost73, 2021-04-19:
Hi,

Bedankt voor je snelle reactie.
Nee, ik maak gebruik van de vocht boxsensor. Ik zou dus graag het vochtpercentage willen uitlezen,

sensor:
https://www.ventilatieland.nl/artikel/17138/duco-vocht-boxsensor-enkel-icm-ducobox-silent.html?utm_source=googleshopping&utm_campaign=googleshopping-Feed&gclid=Cj0KCQjw1PSDBhDbARIsAPeTqrebL2St8OnS_vDQvMvIclMOt4PALa94zwarzrnT9FeMKQ3Enc3ml_YaAsHxEALw_wcB

arnemauer, 2021-04-19:
Ah duidelijk. Volgens mij zit in die sensor ook een temperatuurmeter.
Welke plugins heb je allemaal geactiveerd in espeasy?

Joost73, 2021-04-19:
Hi, 
Onderstaand screenshot toont wat ik nu in ESP heb geactiveerd. Moet er nog een seriële koppeling vanuit DUCO naar ESP? Dit omdat ik in de log zie  " [P153] DUCO BOX SENSOR: Serial port in use, set flag to read data later." 

[image]

Joost73, 2021-04-19:
Hi, nee een reboot toont hetzelfde gedrag.  helaas.

arnemauer, 2022-04-15:
Wat voor ducobox heb je?
Lost de nieuwe firmware dit op? b756d3e

#71 #74

closed 2022-10-25 by arnemauer