arnemauer/Ducobox-ESPEasy-Plugin #72: Aanmelden RF Gateway wil niet lukken.

smoldersonline, 2021-04-21:
Het lukt me niet de RF Gateway aan te melden. Waarschijnlijk zie ik iets over het hoofd? Veel dank!

30386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
30387: [P150] DUCO RF GW:  State: 0.00
31950: WD   : Uptime 1 ConnectFailures 0 FreeMem 11744 WiFiStatus 3
34025:  Webserver args: 0: 'cmd' length: 4
34033: Command: JOIN
34034: [P150] DUCO RF GW: Sent command for 'join' to DUCO unit
34042: [P150] DUCO RF GW: 34034 - SendJoinPacket()
34042: [P150] DUCO RF GW: 34035 - 00,01,00,00,7C,3E,08,02,01,FF,0C,
34043: [P150] DUCO RF GW: 34042 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
34098: [P150] DUCO RF GW: TIMER => 34098 34039 59
34099: [P150] DUCO RF GW: Start of RF signal received
34099: [P150] DUCO RF GW: Ignoring RF noise
34118: [P150] DUCO RF GW: TIMER => 34118 34099 19
34118: [P150] DUCO RF GW: Start of RF signal received
34119: [P150] DUCO RF GW: Ignoring RF noise
40386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
40387: [P150] DUCO RF GW:  State: 0.00
50386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
50387: [P150] DUCO RF GW:  State: 0.00
60386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
60387: [P150] DUCO RF GW:  State: 0.00
61950: WD   : Uptime 1 ConnectFailures 0 FreeMem 11488 WiFiStatus 3
70386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
70387: [P150] DUCO RF GW:  State: 0.00
80386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
80387: [P150] DUCO RF GW:  State: 0.00
90386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
90387: [P150] DUCO RF GW:  State: 0.00
91950: WD   : Uptime 2 ConnectFailures 0 FreeMem 11488 WiFiStatus 3
100386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
100387: [P150] DUCO RF GW:  State: 0.00
110386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
110387: [P150] DUCO RF GW:  State: 0.00
120386: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
120387: [P150] DUCO RF GW:  State: 0.00

arnemauer, 2021-04-21:
Wat is de afstand tussen je Ducobox en de gateway?

smoldersonline, 2021-04-21:
Bij de laatste poging 3 meter (ofzo), in dezelfde ruimte. Bij eerdere pogingen, 15-20 meter.

arnemauer, 2021-04-21:
Kun je van begin tot het eind exact de stappen aangeven die je uitvoert?

smoldersonline, 2021-04-21:
Veel dank voor je support!

Dat ga ik doen, misschien dat ik daarmee wacht tot ik hier een tweede paar ogen beschikbaar heb om mee te kijken (begin volgende week).

Daaraan voorafgaand 'n paar vragen:

- ik actieveer de Installer mode met de bedieningsschakelaar. De snelheid van 'groen knipperen' heeft (hier) twee standen, de eerste stand na het (5 seconden) indrukken van twee diagonale knoppen. Als ik daarna na 'n knop indruk, gaat het groene knipperen aanzienlijk sneller. Welke moet ik hebben om aan te melden?
- als ik 'Join' aanklik in het devices scherm, ga ik door naar het tools scherm waar in de Command regel dan Join staat. Klopt dat en moet ik die submitten?

Weer dank!

smoldersonline, 2021-04-23:
[image]

De instelling voor Radio Power mis ik in de Wiki en ik weet niet of deze zo goed staat?

Dit is het log bij bovenstaande instellingen.

[image]

So-far-so-good, voordat ik weer 'n poging doe?

Reuze bedankt voor je support!

smoldersonline, 2021-04-26:
Update: samen met m'n zoon nog 'n poging gedaan om de gateway aan te melden aan de Duco, en warempel dat leek te lukken. Ik heb nu waarden in de velden Network ID en Device Address. Die kan ik ook terugvinden in het log. De gateway lijkt op het oog nu goed te functioneren, maar loopt bij uitvoeren commando vast. Het is één enkele keer gelukt een commando met succes uit te voeren (plaatsen in-en uit installer mode).

Misschien 'n idee waarom de gateway telkens vastloopt?

Thanks!

arnemauer, 2021-04-26:
Zou je eens willen kijken of de uptime (op de eerdere pagina) dan ook weer op 0 staat?
[image]

Als dat het geval is dan "crashed" de esp8266. 95% van de crashes komt door een slechte voeding van de gateway.
Wat voor usb-kabel en adapter gebruik je? Ik adviseer een zo korte kwalitatief hoogwaardige kabel en een adapter die minimaal 1A kan leveren.

arnemauer, 2021-04-26:
> [image]
> 
> Ik heb er al een paar kabels geprobeerd. Ik probeer er nog een.

Wat voor adapter gebruik je?
n.b. Kabels van de action e.d. zijn waardeloos.

smoldersonline, 2021-04-26:
Ik twijfel of het m'n kabels zijn. Ik voorzie hem nu van stroom met een originele Raspberry Pi voeding. En daarmee loopt hij ook vast. Bijvoorbeeld ook al een anker charger met Ikea kabel geprobeerd.

smoldersonline, 2021-04-26:
Maar als je 'n adapter kooptip hebt, geef ik er nog graag een paar euro's aan uit, om dit werkend te krijgen.

arnemauer, 2021-04-26:
> Ik twijfel of het m'n kabels zijn. Ik voorzie hem nu van stroom met een originele Raspberry Pi voeding. En daarmee loopt hij ook vast. Bijvoorbeeld ook al een anker charger met Ikea kabel geprobeerd.

Die moet zeker goed uit. Dan hebben we voedingsproblemen in ieder geval uitgesloten :) 
Welke plugins heb je geactiveerd? En na welk commando herstart de esp?

smoldersonline, 2021-04-27:
Ik heb alleen de 'DUCO Ventilation remote' geactiveerd.

Het gedrag van de gateway kan ik niet voorspellen. Na opstarten vanmorgen leek alles in orde. Het lukte ook om de ventilator mode aan te sturen, van 'Auto' naar 'Low', en terug. Daarna lijkt de gateway (willekeurig?) verbinding met de Duco te verliezen en staat er bij  'ventilationmode': Disabled. De gateway draait nog wel, 'n reboot van de ESP helpt niet, 'ventilationmode' blijft Disabled. Als ik in 't log kijk zie ik (in geel) 'n paar keer REP : connection failed.

arnemauer, 2021-04-27:
> Daarna lijkt de gateway (willekeurig?) verbinding met de Duco te verliezen 
Puur ter info :) Er is geen actieve verbinding met de Ducobox. De Ducobox is voornamelijk reactief. De gateway stuurt commando's als je de ventilatiestand wil veranderen en de Ducobox reageert daarop. Hij kan dus geen verbinding verliezen zoals bijvoorbeeld met wifi. 

> en staat er bij 'ventilationmode': Disabled. De gateway draait nog wel, 'n reboot van de ESP helpt niet, 'ventilationmode' blijft Disabled. 
Als de plugin op disabled staan dan is de gateway 10 keer gecrashed tijdens het uitvoeren van de plugin. Na 10 crashes zal hij de plugin uitschakelen om de gateway weer bereikbaar te maken. 

> Als ik in 't log kijk zie ik (in geel) 'n paar keer REP : connection failed.
Dit is een melding niet gerelateerd aan de RF plugin.

closed 2021-08-04 by arnemauer