arnemauer/Ducobox-ESPEasy-Plugin #74: RH values are not visible

Trigun, 2021-05-05:
When installing the RF gateway all sensors seem to work fine except the Temp/humidity wall-sensors.
It seems that the given temperature is shown at the humidity sensor and the actual humidity does not work.
Strangely the Duco Network Tool does show all the correct values.

Is seems like the value are send via the wrong GPIO.
The error log from the Gateway does show the following:
[image]

any idea how to solve this?

Gibletron, 2021-06-04:
@arnemauer is there anything we can troubleshoot to help resolve this quicker?

Trigun, 2021-07-06:
@arnemauer ,  any idea? much appreciated, thnx!

arnemauer, 2021-07-07:
Over twee weken heb ik vakantie en zal ik er verder naar kijken. :)
@Gibletron: om alvast wat informatie te verzamelen... 

1. zou je de duco network tool willen openen. 
2. In het menu linksboven op service klikken.
3. Wachtwoord 'stella' invoeren.
4. in het menu 'driver log' inschakelen.
5. op de knop 'refresh all' klikken.
6. via het menu -> save het DNT bestand opslaan.
7. duco network tool afsluiten
8. het DNT bestand en de driverlog (staat in mijn documenten/duco) hier posten?

Dan kan ik namelijk zien wat de duco network tool precies vraagt aan de ducobox bij het uitlezen van de Temp/humidity wall-sensors.

Trigun, 2021-08-09:
Hi Arne,

Excuses dat het even heeft geduurd, oa de vakantie zat ertussen.
Fijn dat je tijd wil vrij maken om hiermee aan de slag te gaan.
Bijgaand vind je het DNT bestand.

Mocht je verder nog vragen hebben dan hoor ik het graag.

Groeten,

Bas

arnemauer, 2021-08-10:
> Hi Arne @arnemauer , Excuses dat het even heeft geduurd, oa de vakantie zat ertussen. Fijn dat je tijd wil vrij maken om hiermee aan de slag te gaan. Bijgaand vind je het DNT bestand. Mocht je verder nog vragen hebben dan hoor ik het graag. Groeten, Bas
> […](#)

De bijlage wordt niet overgenomen uit de e-mail richting Github. Via de browser kun je wel een bijlage uploaden :)

Trigun, 2021-08-17:
Hi @arnemauer, ik was benieuwd of je wellicht al de mogelijkheid hebt gehad om naar de data te kijken? mocht ik nog iets kunnen doen dan hoor ik het graag.

Bedankt!

arnemauer, 2021-08-17:
Hoi Bas,

Ik heb de log nagekeken. Er staat niets geks in waardoor de plugin niet zou moeten werken.
Dit is het commando wat je pc naar de Ducobox stuurt:
'''nodeparaget 2 75
  Get PARA 75 of NODE 2
  --> 5334
  Done '''

Dat komt overeen met wat de plugin aan data verwacht. Zou je een screenshot willen posten van de geactiveerde plugins?

Trigun, 2021-09-05:
Hi Arne,

hierbij het overzicht. zoals je ziet wordt er bij het laatste device geen juiste waardes ingevuld.
[image]

Trigun, 2021-09-21:
Hi @arnemauer, heb jij voldoende info mbt bovenstaande? 

Dank alvast!

Hopperay17, 2022-02-14:
Hey,
Ik heb hetzelfde "probleem" P155 komt niet volledig op de juiste kolommen binnen en de temp staat op de Rela.Humi plek, temp blijft leeg.

unit staat bij mij op node 3

is hier al wat meer over bekend?
[image]

Log:
15136482: [P155] DUCO Ext. Hum. Sensor: Command confirmed by ducobox
15136484: [P155] DUCO Ext. Hum. Sensor: ROW:2 bytes:23
15136485: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 20204765742050415241203733206F66204E4F44452033 END
15136494: [P155] DUCO Ext. Hum. Sensor: Send command successfull!
15136518: [P155] DUCO Ext. Hum. Sensor: ROW:3 bytes:9
15136519: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 20202D2D3E20313739 END
15136520: [P155] DUCO Ext. Hum. Sensor: TEMP: 179 = 17.9°C
15136521: [P155] DUCO Ext. Hum. Sensor: ROW:4 bytes:6
15136521: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 2020446F6E65 END
15136523: [P155] DUCO Ext. Hum. Sensor: ROW:5 bytes:2
15136523: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 2020 END
15136524: [P155] DUCO Ext. Hum. Sensor: End of message received.
15136524: [P155] DUCO Ext. Hum. Sensor: Stop reading from serial
15136556: [P155] DUCO Ext. Hum. Sensor: Start read external sensor. NodeAddress: 3. Type: RH
15136964: [P155] DUCO Ext. Hum. Sensor: ROW:1 bytes:15
15136965: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 6E6F64657061726167657420333735 END
15136965: [P155] DUCO Ext. Hum. Sensor: Received invalid response
15136966: [P155] DUCO Ext. Hum. Sensor: Stop reading from serial

Thanks

arnemauer, 2022-02-14:
> 6E6F64657061726167657420333735

Welke firmware gebruik je? Dit staat onder het tabblad "Main" en dan achter "Git Build:".

Hopperay17, 2022-02-15:
Dit is
Git Build:	AFM 281cde1: Nov 23 2021

arnemauer, 2022-03-30:
[ESP_Easy_mega_20220330_custom_ESP8266_4M1M.zip](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/8383443/ESP_Easy_mega_20220330_custom_ESP8266_4M1M.zip)

@Hopperay17  Zou je deze firmware eens willen proberen? Als het goed is moet hij de waardes nu goed uitlezen.

Hopperay17, 2022-04-05:
Hey bedankt,
Ik heb er mee lopen stoeien mar krijg er helaas nog geen licht uit.

851536: EVENT: Clock#Time=Tue,11:51
6862743: [P155] DUCO Ext. Hum. Sensor: start read, eventid:5
6862744: [P155] DUCO Ext. Hum. Sensor: Read external sensor
6862744: [P155] DUCO Ext. Hum. Sensor: Start read external sensor. NodeAddress: 3. Type: temp
6862794: EVENT: SERGW_EXTB_HUMI#All=18.7,0
6863067: [P155] DUCO Ext. Hum. Sensor: ROW:1 bytes:15
6863067: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: nodepraget 3 73 END
6863068: [P155] DUCO Ext. Hum. Sensor: Received invalid response
6863068: [P155] DUCO Ext. Hum. Sensor: Stop reading from serial
6863688: [P155] DUCO Ext. Hum. Sensor: Start read external sensor. NodeAddress: 3. Type: RH
6864028: [P155] DUCO Ext. Hum. Sensor: ROW:1 bytes:16
6864029: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: nodeparaget 3 75 END
6864030: [P155] DUCO Ext. Hum. Sensor: Received invalid response
6864030: [P155] DUCO Ext. Hum. Sensor: Stop reading from serial

hij blijft op 0

gr

Trigun, 2022-04-05:
@arnemauer, ik zal ook even een test doen. Bedankt dat je ermee bezig bent!

Trigun, 2022-04-05:
@arnemauer, ik heb tevens de voorgestelde software gebruikt maar het resultaat is onveranderd.
[image]
P155 blijft als feedback een invalid response geven.
[image]

arnemauer, 2022-04-05:
> @arnemauer, ik heb tevens de voorgestelde software gebruikt maar het resultaat is onveranderd. [image] P155 blijft als feedback een invalid response geven. [image]

Kun je de weblog als tekst hier posten? Anders moet ik alles overtypen :P

prnt3r, 2022-04-11:
Hi! I have the same issue. External humidity sensor RH values are not coming though. The log for P155 indicates "Received invalid response". The temperature readings from the same sensor are registered OK. 
Hardware: WeMos D1 R2 WiFi ESP8266
Firmware: ESP_Easy_mega_20220331_custom_ESP8266_4M1M (older builds were showing the same behaviour)

printe3r, 2022-04-12:
Hi! I have a similar issue. The RH values from the external humidity sensor are not coming through. P155 log is giving “received invalid response”. The temperature values from the same sensor are being read OK.

-	Hardware: WeMos D1 R2 WiFi ESP8266 
-	Firmware: ESP_Easy_mega_20220331_custom_ESP8266_4M1M (earlier versions had the same issue)

printe3r, 2022-04-12:
Hoi! Nieuwe firmware erop gezet. Hierbij de log file. [P155 log](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/8473524/P155.log)

Ik heb overigens het volgende uit de log kunnen halen:

nodeparaget 4 73  
Get PARA 73 of NODE 4      
^  Unknown Cmd 
nodeparaget 4 75  
Get PARA 75 of NODE 4
nodeparaget 4 73  
Get PARA 73 of NODE 4  
--> 231  Done
nodeparaget 4 75  
Get PARA 75 of NODE 4
nodeparaget 4 73  
Get PARA 73 of NODE 4
nodeparaget 4 75  
Get PARA 75 of NODE 4
nodeparaget 4 73  
Get PARA 73 of NODE 4 
--> 231  Done      
^  Unknown Cmd
nodeparaget 4 75  
Get PARA 75 of NODE 4
nodeparaget 4 73  
Get PARA 73 of NODE 4 
--> 231  Done      
^  Unknown Cmd 
nodeparaget 4 75

Hopperay17, 2022-04-14:
Bovenstaande allemaal aangepast,
Helaas moest ik de module via usb update OTA ging wonderbaarlijk genoeg niet, vandaar dat het wat langer duurt.
Zie in de bijlage een dump.
ps Single event with all values: aan of uit heeft ook geen effect, maar moet deze aan of uit?

[Duco esp8266 dump.txt](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/8487934/Duco.esp8266.dump.txt)
 
 Thanks Gr

arnemauer, 2022-04-14:
Bedankt voor de informatie allen! Ik heb een bug gevonden waardoor hij onterecht aangeeft dat er een ongeldig commando is ontvangen en vervolgens stopt met het uitlezen van humidity. Ik zal vrijdag bezig om deze bug op te lossen.

Trigun, 2022-04-14:
Bedankt Arne!

arnemauer, 2022-04-15:
Wil je de nieuwe firmware van vandaag willen testen? https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/b756d3e5b3b4eedb8bd7c6c9add96b9b651b7181
Het zou nu opgelost moeten zijn.

printe3r, 2022-04-15:
Hoi! Net nieuwe firmware erop gezet. Het probleem lijkt nog hetzelfde :-) Hierbij de log.
[P155 log 20220415.txt](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/8498390/P155.log.20220415.txt)

arnemauer, 2022-04-16:
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/b756d3e5b3b4eedb8bd7c6c9add96b9b651b7181/_P155_DucoSerialExternalHumiditySensor.ino#L102-L110

Oeps foutje.... Tsjah, als je van Waar Onwaar maakt en van Onwaar Waar maakt dan krijg je dus nooit Onwaar 😝

printe3r, 2022-04-16:
Yay! Ik heb een humidity reading. Heel erg bedankt!

closed 2022-04-16 by arnemauer

Trigun, 2022-04-16:
Thnx @***@***.***>,

It’s working great!

Thanks for your efforts!

Closed #74<https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/74>.

Hopperay17, 2022-04-20:
> https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/b756d3e5b3b4eedb8bd7c6c9add96b9b651b7181/_P155_DucoSerialExternalHumiditySensor.ino#L102-L110
> 
> Oeps foutje.... Tsjah, als je van Waar Onwaar maakt en van Onwaar Waar maakt dan krijg je dus nooit Onwaar 😝

Maar is dit aangepast in de versie van ca3c3a9 15-04 ?
Of moet dit handmatig, ik zie geen andere bij de files?

Helaas bij mij nog geen input :-(

600655: [P155] DUCO Ext. Hum. Sensor: start read, eventid:0
600656: [P155] DUCO Ext. Hum. Sensor: Start read external sensor. NodeAddress: 3. Type: temp
600712: EVENT: Duco_extb_Humi#All=19.3,0.0
600999: [P155] DUCO Ext. Hum. Sensor: ROW:1 bytes:16
601000: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 6E6F6465706172616765742033203733 END
601001: [P155] DUCO Ext. Hum. Sensor: Command confirmed by ducobox
601003: [P155] DUCO Ext. Hum. Sensor: ROW:2 bytes:23
601004: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 20204765742050415241203733206F66204E4F44452033 END
601015: [P155] DUCO Ext. Hum. Sensor: Send command successfull!
601121: [P155] DUCO Ext. Hum. Sensor: ROW:3 bytes:9
601122: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 20202D2D3E20313933 END
601123: [P155] DUCO Ext. Hum. Sensor: TEMP: 193 = 19.3Â°C
601124: [P155] DUCO Ext. Hum. Sensor: ROW:4 bytes:6
601124: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 2020446F6E65 END
601125: [P155] DUCO Ext. Hum. Sensor: ROW:5 bytes:2
601125: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: 2020 END
601125: [P155] DUCO Ext. Hum. Sensor: End of message received.
601126: [P155] DUCO Ext. Hum. Sensor: Stop reading from serial

arnemauer, 2022-04-20:
Heb je in de map gekeken? https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/tree/master/Binaries/Binary%20based%20on%20ESPEasy-mega-20211105