arnemauer/Ducobox-ESPEasy-Plugin #75: DUCO SER GW: Serial reading timeout

Fusionyx-SVoskamp, 2021-06-05:
Hoi Arne,

Ik probeer een Wemos D1 mini (Lolin) te koppelen aan de docu silent zover ik na kan gaan staat alles goed echter ik blijf onderstaande zien in de log en de esp leest geen waardes uit enig idee waar ik kan zoeken naar de fout.

Heb 
1: Zwart TX -> RX ESP
2: Rood RX -> TX ESP
4: Geel Grn -> Gnd ESP

ESP wordt gevoed door een mini usb met 5v 1a adapter.

1020663: [P151] DUCO SER GW: Start readNetworkList
1020830: [P151] DUCO SER GW: Send command successfull!
1022222: WD : Uptime 17 ConnectFailures 0 FreeMem 11272 WiFiStatus 3
1023667: [P151] DUCO SER GW: Serial reading timeout
1023667: [P151] DUCO SER GW: Stop reading from serial
1052223: WD : Uptime 18 ConnectFailures 0 FreeMem 11248 WiFiStatus 3
1080660: [P151] DUCO SER GW: Start readNetworkList
1080826: [P151] DUCO SER GW: Send command successfull!
1082222: WD : Uptime 18 ConnectFailures 0 FreeMem 11272 WiFiStatus 3
1083667: [P151] DUCO SER GW: Serial reading timeout
1083668: [P151] DUCO SER GW: Stop reading from serial
1112223: WD : Uptime 19 ConnectFailures 0 FreeMem 11272 WiFiStatus 3
1140660: [P151] DUCO SER GW: Start readNetworkList
1140829: [P151] DUCO SER GW: Send command successfull!
1142226: WD : Uptime 19 ConnectFailures 0 FreeMem 7824 WiFiStatus 3
1143667: [P151] DUCO SER GW: Serial reading timeout
1143667: [P151] DUCO SER GW: Stop reading from serial

Ik heb ook een kabel bemaakt voor de Network Tool en die werkt prima.

Giel538, 2021-07-05:
Hoi.

Heb jij het al werkend? 
Ik zit momenteel met hetzelde probleem.

arnemauer, 2021-07-05:
Lost dit jullie probleem op? https://github.com/arnemauer/ducobox-espeasy-plugin/issues/47#issuecomment-678793188

Fusionyx-SVoskamp, 2021-07-05:
Hoi Arne, klink als een oplossing ga ik als ik even een momentje hebt deze week uitproberen. Laat het hier weten.

Fusionyx-SVoskamp, 2021-07-05:
Wel ik kon het niet laten om even de boel te meten en de tx pin te vinden. Heb deze doorgeknipt zodat ik hem alsnog later een keer kan solderen met een weerstandje ertussen of een tijdelijk brug. Hierna aangesloten en krijg nu netjes alle waardes door. Thanks! voor de juiste richting in te duwen.

closed 2021-07-05 by arnemauer

Giel538, 2021-07-05:
Ik heb hetzelfde gedaan. De pin doorgesneden en het werkt. Ik had ondertussen al een NodeMCU aangesloten en daar werkte het in 1x. 

@arnemauer Ik las ergens dat je de overstap gaat / ging maken naar home assistant. Daar kan je door middel van MQTT berichten je sensoren automatisch te voorschijn laten komen. Ik kan je wel helpen met de json van deze berichten mocht het nodig zijn.

Jover68, 2021-07-08:
Dacht eerder te hebben gelezen dat het bij mij niet werkte door een bepaalde versie printplaat van de Duco Box. Het was een gepriegel. Pin 2 van de CH340 chip los gesoldeerd en het werkt hier nu ook.