arnemauer/Ducobox-ESPEasy-Plugin #76: Zijn er nog Gateway's beschikbaar

klaaspiet, 2021-06-14:
Zijn er nog gateways beschikbaar.
zijn met een nieuwbouwhuis bezig en zoals het nu lijkt komt daar een Duco systeem in dus zou graag willen weten of ze nog beschikbaar zijn.  Dan zou ik wel interesse hebben.

arnemauer, 2021-06-14:
Hoi KlaasPiet, zie de link op de readme pagina :) https://github.com/arnemauer/Ducobox-ESPEasy-Plugin

closed 2021-06-14 by arnemauer

klaaspiet, 2022-08-23:
Huis is klaar en is een Duco geworden aanvraag is verstuurd voor een complete gateway. 
Nu nog ff wachten tot februari lees ik.
Eerst maar eens de lampen, zonneschermen en verbruikers in kaart brengen.

arnemauer, 2022-08-23:
Ik verwacht met 1-2 weken weer een aantal op voorraad te hebben. Je ontvangt tzt een email ;)

klaaspiet, 2022-09-29:
Heb je al weer onderdelen gekregen. Ben erg nieuwsgierig 😀