arnemauer/Ducobox-ESPEasy-Plugin #77: Gateway niet bereikbaar

arnemauer, 2021-07-20:
> ik heb nu de reboot rule erin staan, maar de gateway is toch niet meer benaderbaar . Dus ondanks de dagelijkse reboot alsnog ineens niet te benaderen . Misschien toch teveel Op wo 9 jun. 2021 om 22:04 schreef Gibletron ***@***.***>:
> […](#)
> En nog even de terugkoppeling: na enkele dagen zonder reboot geen probleem bemerkt, dus voor mij blijft de rule uit staan Mocht ik er toch weer tegenaan lopen dan zet ik de rule weer aan, maar lijkt nu niet nodig — You are receiving this because you were mentioned. Reply to this email directly, view it on GitHub <[#24 (comment)](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/24#issuecomment-858055377)>, or unsubscribe <https://github.com/notifications/unsubscribe-auth/AGRYIUPY453HSX6ILMDQ2G3TR7CNDANCNFSM4MRGSUKQ> .

Helaas blijkt dat alleen de reboot rule `On Clock#Time=All,01:10 do
  delay 65000 // Needed else it will reboot many times at 01:10
  reboot // reboot only this module
endon ` niet voldoende is . Moet met regelmaat de Wemos "met de hand" stroomvrij maken en zo resetten. De Wemos is geeft op dat moment wel gewoon reply op een ping request maar MQTT is verbroken en ESP Easy site niet meer beschikaar. Zijn er toevallig al gebruikers die een nieuwere ESPHome build "ge-compiled" hebben ?

_Originally posted by @Theoreet in https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/24#issuecomment-878328852_

arnemauer, 2021-07-20:
@theoreet
- Welke plugins heb je geactiveerd en wat is de interval?

- Welke controllers heb je ingesteld?

- Welke rules heb je geactiveerd?

Theoreet, 2021-07-21:
> @Theoreet
> 
> * Welke plugins heb je geactiveerd en wat is de interval?
[image]

> * Welke controllers heb je ingesteld?
[image]

> * Welke rules heb je geactiveerd?
[image]

arnemauer, 2021-08-04:
Wat is de interval van iedere plugin?

Is de gateway aangesloten met een kwalitatieve korte usb-kabel en goede usb-adapter (minimaal 1A)?

Kun je een screenshot maken van de "main" tab? Het gaat met name om Load, Free RAM, Free Stack.

Theoreet, 2021-08-31:
heeft even moeten duren mede door vakantie : 
[image]

De interval heb ik niks aan veranderd, en staat op 60s

Theoreet, 2021-09-24:
Inmiddels de voeding vervangen door een andere / betere, maar na dag 1 is de boel weer onbereikbaar ;(

Theoreet, 2021-10-05:
Kan / wil iemand een nieuwe bin file compileren met / op de nieuwere ESPEasy binaries ?
Op ESPHome zal nog fijner zijn :D

arnemauer, 2022-10-25:
Waarschijnlijk lost de nieuwste firmware je probleem op: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/4e4d3791da5bb5204a622e2e6e2d3380ea2f31c2

closed 2022-10-25 by arnemauer