arnemauer/Ducobox-ESPEasy-Plugin #78: Ducobox 325 RF supported?

jonskoglund, 2021-08-09:
Does the Ducobox 325 RF work with your plugin? I stumbled upon this and I'm very excited to integrate it with homeassistant.
I don't know what kind of ports I should look for on my unit. What PCB should my Ducobox have to be compatible?

arnemauer, 2021-08-10:
Voor zover ik weet heeft de Ducobox 325 RF geen connector voor een seriële verbinding. De RF gateway zou wel moeten werken.
Voor de seriele verbinding is deze connector nodig:

[image]

MartijnDisse, 2021-11-01:
I have an Ducobox Energy Premium 325 and that one has the required serial port. But I get strange serial values.

arnemauer, 2021-11-01:
> I have an Ducobox Energy Premium 325 and that one has the required serial port. But I get strange serial values.

De Ducobox 325 RF is een oud type ducobox (+-10 jaar oud) ;) Heel anders dan de Ducobox Energy Premium 325 (die heeft veel meer functies).

closed 2021-11-01 by arnemauer