arnemauer/Ducobox-ESPEasy-Plugin #79: Home Assistant integration - ESPHome - Native API

800xl, 2021-09-05:
Hi,
It would be great if the plugin could be ported from EasyESP towards ESPHome. This way the Native API could be used for Home Assistant removing the requirement of having a MQTT broker in between . https://esphome.io/components/api.html
Would this be something that could be considered?

I'm more then happy to work on a Home Assistant plugin to provide a seamless integration.

Don't want to start a full fork on this and have these site by site while ESPHome could be used still with a MQTT broker next to the Home Assistant direct integration.

JSEHV, 2021-09-21:
Migrating from previous ESPEasy setups is very easy. You just need to have ESPHome create a binary for you and then upload that in the ESPEasy web interface.

Is it this easy and did somebody try it?

Details: https://esphome.io/guides/migrate_espeasy.html

Jozziej, 2021-09-23:
Yes an ESPHome port would be great and apparently 
[Fwestenberg](https://github.com/fwestenberg?tab=repositories) Already made a port to ESPHome see his comment in another section.

https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/54

Theoreet, 2022-03-29:
iemand kennis van ESPHome om het project van @Fwestenberg verder op te pakken ? @

sanderlv, 2022-10-17:
Hier ben ik ook in geinteresseerd! Arne heeft een top ding gemaakt maar ik baal er wel van dat het niet in lijn is met al mijn andere ESPs (esphome)

tonnie84, 2023-11-07:
Ik zie dat met de Itho intergratie er ook een overstap is gemaakt van espeasy naar esphome. Het zou gaaf zijn wanneer dat met dit project ook lukt

https://github.com/jodur/ESPHOME-Itho

arnemauer, 2023-11-07:
Ik denk dat het zeker mogelijk is en ik wil er ook wel naar kijken. Het enige waar ik tegenaan loop is hoe ik het node id ga opslaan. De ducobox kent een node id toe aan een node bij het aanmelden. Het zou handig zijn als het node id en het network id dan automatisch opgeslagen kunnen worden in de configuratie van esphome. Iemand een idee?

Theoreet, 2023-12-31:
zal dit misschien kunnen helpen ? [https://github.com/Flamingo-tech/Open-AIR](url)

Theoreet, 2024-12-18:
iemand nog vorderingen ofzo kunnen maken ?

etamminga, 2025-01-18:
> Ik denk dat het zeker mogelijk is en ik wil er ook wel naar kijken. Het enige waar ik tegenaan loop is hoe ik het node id ga opslaan. De ducobox kent een node id toe aan een node bij het aanmelden. [...]

Esphome support zou gaaf zijn. Ik heb m nu in espeasy maar zou graag op dezelfde esp wat meer doen wat niet standaard in de gecompileerde bin zit. (1-wire ds18b20 bv). 

Kun je deze ID’s niet kwijt in HA? Verstop het ID in een configuration entity. Weet niet of die terug komen naar esphome maar lijkt me een mooie oplossing als dat werkt

svh1985, 2025-01-18:
Wat je zou kunnen doen is dat je na pairing de node-id naar log schrijft. Die kun je vervolgens kopiëren/plakken in de yaml, opnieuw compileren en firmware uploaden.

Theoreet, 2025-02-13:
> Esphome support zou gaaf zijn. Ik heb m nu in espeasy maar zou graag op dezelfde esp wat meer doen wat niet standaard in de gecompileerde bin zit. (1-wire ds18b20 bv).
>
> Kun je deze ID’s niet kwijt [...]

Kun je zoiets niet hard code in de ESPhome config kwijt ? En anders gedacht hoe doet ESPeasy dit nu ?

maxhelskens, 2025-07-09:
Has anyone made some progress on this? I was looking into buying the Duco Connectivity board for my Ducobox Focus, but if I get to choose between a 20 euro esp and a 200 euro pcb...
I have zero experience with developing for ESPhome, but am happy to help where I can.