arnemauer/Ducobox-ESPEasy-Plugin #81: Sensorwaarden uitlezen uit Ducobox Energy premium dmv Ducobox serial gateway

MartijnDisse, 2021-10-03:
Hi,

In je 'bestelformulier' zag ik dat ik vragen hier dien te stellen, dus hier komt die ;).

Mijn doel ik om de externe CO2 sensor en de box luchtvochtigheidssensor waardes uit te lezen uit mijn Ducobox energy premium.
Volgende de readme kan dat al met de 'Ducobox serial gateway'. Bij je 'bestelformulier' zie ik alleen de 'RF gateway' en ook een 'USB kabel' (zonder Wemos zo lijkt het), maar geen 'serial gateway' In je omschrijving van het 'bestelformulier' staat:

```
Ik heb nog een aantal complete units over. Je krijgt dan de volledige unit: 
- inclusief de kabel (voor het uitlezen van de sensoren)
- CC1101 module (voor het aansturen)
- wifi antenne (2,4Ghz)
- geflashed met de juiste firmware met de aanstuur en uitleesmodules
```

Moet ik dan gewoon de 'Ventilation Gateway' van 34EUR hebben voor mijn toepassing?

MartijnDisse, 2021-11-01:
Inmiddels heb zelf een Wemos D1 met kabeltje verbonden met de Ducobox Energy Premium. Ik heb zelf de binaries gecompileerd met de laatste versie van ESPEasy. Bij het compileren kreeg ik een error bij iedere P15X op de regel `Device[deviceCount].VType = SENSOR_TYPE_DUAL`
  dat de SENSOR_TYPE_DUAL (of andere sensor type) niet gedefineerd was. Ik heb dat toen verandert in 
`Device[deviceCount].VType = Sensor_VType::SENSOR_TYPE_DUAL`
Vervolgens de boel gecompileerd. Ik krijg nu geen meetwaarden, maar in de log staan dit soort berichten:
`1330892: [P153] DUCO BOX SENSOR: Pakket ontvangen: 202054454D50202D205348543231203A2020313733205B2E31B0435D20283029 END
1330893: [P153] DUCO BOX SENSOR: Pakket ontvangen: 2020 END
1340931: [P153] DUCO BOX SENSOR: Pakket ontvangen: 73656E736F72696E666F END
1340933: [P153] DUCO BOX SENSOR: Pakket ontvangen: 20205B53454E534F525D20494E464F END
1340940: [P153] DUCO BOX SENSOR: Pakket ontvangen: 202052482020202D205348543231203A2036323532205B2E3031255D20283029 END
1340961: [P153] DUCO BOX SENSOR: Pakket ontvangen: 202054454D50202D205348543231203A2020313734205B2E31B0435D20283029 END
1340963: [P153] DUCO BOX SENSOR: Pakket ontvangen: 2020 END
1350860: [P153] DUCO BOX SENSOR: Pakket ontvangen: 73656E736F72696E666F END
1350862: [P153] DUCO BOX SENSOR: Pakket ontvangen: 20205B53454E534F525D20494E464F END
1350869: [P153] DUCO BOX SENSOR: Pakket ontvangen:`
Enig idee wat dat kan zijn?

arnemauer, 2021-11-01:
Het pakket komt in hex. Je kan een hex naar ascii gebruiken om daar leesbare tekst van te maken.
https://www.rapidtables.com/convert/number/hex-to-ascii.html

1330892: [P153] DUCO BOX SENSOR: Pakket ontvangen: 202054454D50202D205348543231203A2020313733205B2E31B0435D20283029 END

is bijvoorbeeld:

  TEMP - SHT21 :  173 [.1°C] (0)

arnemauer, 2022-04-06:
De plugin werkt niet met jou Ducobox. De plugin verwacht de informatie in het volgende formaat:
```
            /* Example output:
                [SENSOR] INFO
                CO2 :  627 [ppm] (0)
                RH : 3246 [.01%] (0)
                TEMP :  235 [.1Â°C] (0)
            */
```

Jou ducobox geeft de volgende informatie:

```
  sensorinfo
  [SENSOR] INFO
  RH   - SHT21 : 6252 [.01%] (0)
  TEMP - SHT21 :  174 [.1°C] (0)
```

Welke type Ducobox Energy Premium heb je exact? Dan kan ik kijken of ik de plugin aan kan passen zodat hij deze waardes ook kan uitlezen.

MartijnDisse, 2022-04-13:
@arnemauer ; Dat zou wel heel fijn zijn! Ik heb een DucoBox Energy Premium 325-1ZS. Artikelnummer 0000-4359.

arnemauer, 2022-04-15:
Ik heb de code aangepast zodat de Ducbox Energy Premium ook werkt :)
6d1f6c1 

Wil je die eens testen of het werkt? Bij voorkeur met een stukje log 👍

closed 2022-10-25 by arnemauer