arnemauer/Ducobox-ESPEasy-Plugin #82: Koppelen is gelukt maar ducobox is niet aan te sturen.

ocmeros, 2021-11-05:
Ik heb zojuist de ducobox gekoppeld volgens de koppelingsprocedure. Op de devices pagina is een network id en device address toegevoegd. Echter lukt het niet om vervolgens de ducobox aan te sturen. 
In de log is het volgende te zien:
Command: VENTMODE
1557192: [P150] DUCO RF GW: Sent command for 'VENTMODE' & 'high' to DUCO unit
1557200: [P150] DUCO RF GW: 1557193 - 00,07,00,05,34,7D,10,44,13,FF,40,00,22,00,0E,6E,
1557200: [P150] DUCO RF GW: 1557199 - Send requestventilationmode done. mode: 6, override percentage: 0%, temp: 210 C, installermode: 0
1557200: [P150] DUCO RF GW: 1557199 - Start waiting for ack...
1557301: [P150] DUCO RF GW: TIMER => 1557301 1557199 102
1557301: [P150] DUCO RF GW: Start of RF signal received
1557302: [P150] DUCO RF GW: Ignoring RF noise
1557710: [P150] DUCO RF GW: 1557703 - CheckforAck: still waiting for ACK. Sending message again...
1557710: [P150] DUCO RF GW: 1557704 - 0F,07,00,05,34,7D,10,44,13,FF,40,00,22,00,0E,6E,
1557721: [P150] DUCO RF GW: TIMER => 1557720 1557710 10
1557721: [P150] DUCO RF GW: Start of RF signal received
1557721: [P150] DUCO RF GW: Ignoring RF noise
1558310: [P150] DUCO RF GW: 1558303 - CheckforAck: still waiting for ACK. Sending message again...
1558310: [P150] DUCO RF GW: 1558304 - 0F,07,00,05,34,7D,10,44,13,FF,40,00,22,00,0E,6E,
1558321: [P150] DUCO RF GW: TIMER => 1558320 1558316 4
1558321: [P150] DUCO RF GW: Start of RF signal received
1558322: [P150] DUCO RF GW: 1558321 - Received message: SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0005347d;
1558323: [P150] DUCO RF GW: 1558322 - Type:6; Bytes:9; Counter:3; RSSI:-50 (0x30);
1558323: [P150] DUCO RF GW: 1558322 - FF,
1558323: [P150] DUCO RF GW: 1558322 - Received messagetype: ACK
1558323: [P150] DUCO RF GW: 1558322 - Ack received!

Echter gaat de ducobox niet naar de high stand.

Ik heb de ESP geladen met een kant en klare binary en de esp flash tool (Flasher firmware 17-04-2020 5177743)

Hoe kan ik dit oplossen?

arnemauer, 2021-11-05:
Helpt het als je de RF power wat hoger zet?
"Radio power 10,3 dBm (0xC1)"

ocmeros, 2021-11-06:
Ik krijg dan telkens meteen een ACK. Echter reageerd de ducobox niet op de commando's. Deze blijft op status Automatic(0) staan. Kan het zijn dat de software van de ducobox moet worden aangepast o.i.d. Deze heeft hiervoor alleen met een gewone standenschakelaar gewerkt.

arnemauer, 2021-11-07:
Dan is de aanmelding waarschijnlijk niet geslaagd. De Ducobox heeft de laatste bevestiging van de gateway niet ontvangen waardoor de gateway niet is aangemeld. Kun je met een kabel en de Duco Network Tool kijken of de gateway tussen de lijst met nodes staat? Een andere optie is het aanmelden opnieuw proberen (network ID en node even leeghalen).

ocmeros, 2021-11-07:
Ik heb opnieuw een koppeling gedaan. De ducobox geeft nu een andere status door aan de gateway. Het aansturen van de box werkt echter nog niet, maar ik vermoed dat dat nu aan de box ligt (deze heeft de status PermanentManualMode2(12)). Als ik binnenkort weer even tijd heb ga ik eens proberen te verbinden met de Duco Network Tool, om te kijken of ik daarin iets kan aanpassen.

De  gateway ligt nu vlak naast de ducobox.

arnemauer, 2021-11-14:
De gateway moet **minimaal** 100cm van de Ducobox af staan om goed te werken. Als je hem in de behuizing van de Ducobox zet gaat het gegarandeerd niet werken. Dit staat ook in de handleiding van de Ducobox. Vergelijk het als twee mensen die naast elkaar staan en tegen elkaar staan te schreeuwen. Dan is het geluid zo hard dat je elkaar niet meer verstaat. Datzelfde geldt ook voor twee radio's die te dicht bij elkaar staan. Ze signaal wat ze van elkaar ontvangen is zo hard dat ze elkaar niet verstaan.

ocmeros, 2021-11-15:
Dat was inderdaad het probleem. Ik dacht 1 ESP te gebruiken voor zowel de serial gateway als de RF gateway. Ik heb het nu in 2 verschillende ESP's verwerkt en het werkt nu heel goed. Volgende stap is het koppelen met domoticz, maar dat moet lukken met de duidelijke handleiding in de WIKI 👍

closed 2021-11-15 by ocmeros