arnemauer/Ducobox-ESPEasy-Plugin #85: Use ducobox-ESPEasy-plugin for Duco IQ unit

jvdVlugt, 2022-03-11:
Is it possible to use this plugin for a DUCObox IQ-unit or is there an other way to get info from the IQ-unit in Domoticz?
I have a ESP8266 that I could use.

arnemauer, 2022-03-11:
https://www.duco.eu/nl/producten/mechanische-ventilatie/sturingscomponenten/iq-unit

Daarnaast communiceert de IQ-unit via het universele **ModBus-protocol (over RS485)** met een externe sturingsunit. Op deze manier kan de IQ-unit alle info weergeven van de componenten binnen het DUCO-netwerk of kan de externe unit het netwerk besturen. Alle ventilatieparameters zoals CO2, vocht, temperatuur, stand afvoerbox of roosteropeningen kunnen uitgelezen worden. Bovendien kunnen deze laatste twee parameters ook extern gewijzigd worden.

Je kan beter modbus gebruiken  ;)

jvdVlugt, 2022-03-11:
Is er een plugin voor ESPeasy voor modbus welke ik kan gebruiken?

arnemauer, 2022-03-11:
Https://www.google.nl

closed 2022-03-11 by arnemauer