arnemauer/Ducobox-ESPEasy-Plugin #86: Beschikbaarheid en meerdere boxen

arjan-io, 2022-03-12:
Hallo,

Ik ben geïnteresseerd in dit project maar een paar vraagjes;
1.  Zijn er RF-gateways beschikbaar
2. Ondersteund de RF-gateway meerdere boxen (wij hebben er 2)
3. Kan de RF-gateway gecombineerd worden met de serial-gateway om 1 MQTT connectie te hebben?
4. In het geval van 3, waarschijnlijk 1 seriële connectie per RF-gateway?

Bij voorbaat dank

arnemauer, 2022-03-30:
**1. Zijn er RF-gateways beschikbaar**
Ja, zie de link naar het formulier op de readme.

**2. Ondersteund de RF-gateway meerdere boxen (wij hebben er 2)**
Hardware technisch zou dat zeker een mogelijkheid zijn. Je kan zelf de code aanpassen om meerdere boxen te ondersteunen.

**3. Kan de RF-gateway gecombineerd worden met de serial-gateway om 1 MQTT connectie te hebben?**
Zeker. ESPEasy heeft één MQTT connectie waarover alle data van alle plugins wordt verstuurd.

**4. In het geval van 3, waarschijnlijk 1 seriële connectie per RF-gateway?**
Inderdaad 1 seriële connectie per gateway omdat de esp8266 maar één seriële TX/RX heeft.

arjan-io, 2022-03-31:
Bedankt voor de antwoorden! Ik weet niet of ik
1) de code zelf aangepast krijg
2) gezien de esp8266 maar 1 tx/rx heeft moet ik er wellicht toch aan 2.

Ik denk er nog even over en zal eventueel het formulier gebruiken

closed 2022-03-31 by arjan-io