arnemauer/Ducobox-ESPEasy-Plugin #90: Gateway traag en herstart steeds

xylone, 2022-04-01:
Bij het wijzigen van de devices werkt de gateway erg traag en hangt soms of krijgt een timeout.
Meestal worden de wijzigingen ook niet opgeslagen.
Als ik dan naar Main ga en kijk naar de uptime, staat deze weer op 1 min.
Ik heb verschillende voedingen en kabel geprobeerd.
Op dit moment hangt er een Raspberry Pi voeding aan (5.1V 2.5A).

xylone, 2022-04-01:
Wellicht heeft het er mee te maken, maar ik krijg dus ook deze meldingen (Error while reading/writing config.dat in 2170) bij het proberen te wijzigen van 1 device. 
[image]

Ik moet dan zeker een paar minuten wachten voordat ik er weer op kan komen.

xylone, 2022-04-01:
Hmm het lijkt een geheugen probleem te zijn van Device 8. 
Deze gaf problemen bij het wijzigen, maar na het verwijderen geeft deze problemen bij het toevoegen.
Nu een device toegevoegd op 7 en dat werkt gewoon.

De settings opslaan (Advanced->Settings->Save) zorgt ook voor een herstart.

arnemauer, 2022-04-01:
Gezien de geheugenfouten kan je het beste de firmware er opnieuw opzetten. 
Je kan hier de firmware en flasher downloaden: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/master/Binaries/Binary%20based%20on%20ESPEasy-mega-20211105/20220331%20Flasher%20Firmware%2003f6eea%2031-03-2022%20(ESPEasy%20mega-20211105).zip

Zorg dat je eerst je instellingen (met name van de RF plugin) opschrijft voordat je deze stappen uitvoert.
- Sluit de gateway aan op je pc;
- Open de flasher
- Selecteer de juiste poort en bij firmware kies je voor blank_4MB.bin;
- Flash de firmware;
- Selecteer vervolgens bij firmware "20220331 Firmware 03f6eea 31-03-2022 (ESPEasy mega-20211105).bin" en klik op flash;
- Alle plugin weer activeren en instellingen terugzetten;

Lost dit het probleem op?

treb33, 2022-04-04:
Zelfde issue. Ik ga straks proberen de firmware er opnieuw op te zetten.

treb33, 2022-04-04:
Helaas niet gelukt zie log:

######2022-04-04######
#######0.04.007#######
######FLASH INFO######
BIN file: 20220331 Firmware 03f6eea 31-03-2022 (ESPEasy mega-20211105).bin
COM port: (COM6) USB-SERIAL CH340 (Port_#0002.Hub_#0001) [offline]
Baud rate: 115200
######POST FLASH######
No post flash information entered...
######FLASH LOG######
[esptool.exe -vv -cd nodemcu -cb 115200 -cp COM6 -ca 0x00000 -cf "C:\Users\bert\Downloads\20220331 Flasher Firmware 03f6eea 31-03-2022 (ESPEasy mega-20211105)\bin\20220331 Firmware 03f6eea 31-03-2022 (ESPEasy mega-20211105).bin"]
[4-4-2022 15:55:32] esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
[4-4-2022 15:55:32] 	setting board to nodemcu
[4-4-2022 15:55:32] 	setting baudrate from 115200 to 115200
[4-4-2022 15:55:32] 	setting port from  to COM6
[4-4-2022 15:55:32] 	setting address from 0x00000000 to 0x00000000
[4-4-2022 15:55:32] 	espcomm_upload_file
[4-4-2022 15:55:32] 	espcomm_upload_mem
[4-4-2022 15:55:32] error: Failed to open COM6
[4-4-2022 15:55:32] error: espcomm_open failed
[4-4-2022 15:55:32] error: espcomm_upload_mem failed
[2022-04-04 15:55:32] STOPPED due to 3 errors! (try reset on the unit, then start a new flash attempt)

treb33, 2022-04-04:
Ik moest trouwens "Only active ports" uitvinken om de serial gateway te kunnen zien, dus wellicht gaat daar iets niet goed?

treb33, 2022-04-04:
Na het flashen weer aangesloten. Nu vraagt de gateway om een wifi wachtwoord. Dat was in eerste instantie niet.

xylone, 2022-04-04:
Excuus voor de wat late reactie. Was een druk weekend. ;-)
Zojuist geflashed (eerst blank en toen de firmware).
Kan verbinden met de AP, krijg netjes een ip adres (192.168.4.x) en wil dan verbinden met 192.168.4.1, maar daar lijkt hij niet mee te kunnen verbinden. Enig idee?

xylone, 2022-04-04:
Nogmaals geflased (blank en firmware).
Twee keer ging het flashen niet goed van de firmware. 
Even gedisconnect en weer geprobeerd en toen deed hij het wel.
En nu kan ik er ook op komen.
Ga nu alles weer opnieuw instellen.

arnemauer, 2022-04-04:
> Nogmaals geflased (blank en firmware). Twee keer ging het flashen niet goed van de firmware. Even gedisconnect en weer geprobeerd en toen deed hij het wel. En nu kan ik er ook op komen. Ga nu alles weer opnieuw instellen.

Zou je de logs van het flashen willen posten? Die staan in de map logs.

xylone, 2022-04-04:
######2022-04-04######
#######0.04.007#######
######FLASH INFO######
BIN file: 20220331 Firmware 03f6eea 31-03-2022 (ESPEasy mega-20211105).bin
COM port: (COM4) Silicon Labs CP210x USB to UART Bridge (Port_#0005.Hub_#0001)
Baud rate: 115200
######POST FLASH######
No post flash information entered...
######FLASH LOG######
[esptool.exe -vv -cd nodemcu -cb 115200 -cp COM4 -ca 0x00000 -cf "C:\Users\Justin\Downloads\20220331 Flasher Firmware 03f6eea 31-03-2022 (ESPEasy mega-20211105)\bin\20220331 Firmware 03f6eea 31-03-2022 (ESPEasy mega-20211105).bin"]
[4-4-2022 20:23:53] esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
[4-4-2022 20:23:53] 	setting board to nodemcu
[4-4-2022 20:23:53] 	setting baudrate from 115200 to 115200
[4-4-2022 20:23:53] 	setting port from  to COM4
[4-4-2022 20:23:53] 	setting address from 0x00000000 to 0x00000000
[4-4-2022 20:23:53] 	espcomm_upload_file
[4-4-2022 20:23:53] 	espcomm_upload_mem
[4-4-2022 20:23:53] 	setting serial port timeouts to 1000 ms
[4-4-2022 20:23:53] opening bootloader
[4-4-2022 20:23:53] resetting board
[4-4-2022 20:23:53] trying to connect
[4-4-2022 20:23:53] 	flush start
[4-4-2022 20:23:53] 	setting serial port timeouts to 1 ms
[4-4-2022 20:23:53] 	setting serial port timeouts to 1000 ms
[4-4-2022 20:23:53] 	flush complete
[4-4-2022 20:23:53] 	espcomm_send_command: sending command header
[4-4-2022 20:23:53] 	espcomm_send_command: sending command payload
[4-4-2022 20:23:53] 	read 0, requested 1
[4-4-2022 20:23:53] trying to connect
[4-4-2022 20:23:53] 	flush start
[4-4-2022 20:23:53] 	setting serial port timeouts to 1 ms
[4-4-2022 20:23:53] 	setting serial port timeouts to 1000 ms
[4-4-2022 20:23:53] 	flush complete
[4-4-2022 20:23:53] 	espcomm_send_command: sending command header
[4-4-2022 20:23:53] 	espcomm_send_command: sending command payload
[4-4-2022 20:23:53] 	espcomm_send_command: receiving 2 bytes of data
[4-4-2022 20:23:53] 	espcomm_send_command: receiving 2 bytes of data
[4-4-2022 20:23:53] 	espcomm_send_command: receiving 2 bytes of data
[4-4-2022 20:23:53] 	espcomm_send_command: receiving 2 bytes of data
[4-4-2022 20:23:53] 	espcomm_send_command: receiving 2 bytes of data
[4-4-2022 20:23:53] 	espcomm_send_command: receiving 2 bytes of data
[4-4-2022 20:23:53] 	espcomm_send_command: receiving 2 bytes of data
[4-4-2022 20:23:53] 	espcomm_send_command: receiving 2 bytes of data
[4-4-2022 20:23:53] Uploading 944352 bytes from C:\Users\Justin\Downloads\20220331 Flasher Firmware 03f6eea 31-03-2022 (ESPEasy mega-20211105)\bin\20220331 Firmware 03f6eea 31-03-2022 (ESPEasy mega-20211105).bin to flash at 0x00000000
[4-4-2022 20:23:53] 	erasing flash
[4-4-2022 20:23:53] 	size: 0e68e0 address: 000000
[4-4-2022 20:23:53] 	first_sector_index: 0
[4-4-2022 20:23:53] 	total_sector_count: 231
[4-4-2022 20:23:53] 	head_sector_count: 16
[4-4-2022 20:23:53] 	adjusted_sector_count: 215
[4-4-2022 20:23:53] 	erase_size: 0d7000
[4-4-2022 20:23:53] 	espcomm_send_command: sending command header
[4-4-2022 20:23:53] 	espcomm_send_command: sending command payload
[4-4-2022 20:23:53] 	setting serial port timeouts to 15000 ms
[4-4-2022 20:23:53] 	setting serial port timeouts to 1000 ms
[4-4-2022 20:23:53] 	espcomm_send_command: receiving 2 bytes of data
[4-4-2022 20:23:53] 	writing flash
[4-4-2022 20:23:54] ...	wrote 0, requested 1
[4-4-2022 20:23:54] error: failed sending 0xC0
[4-4-2022 20:23:54] 	read 0, requested 1
[4-4-2022 20:23:54] warning: espcomm_send_command: didn't receive command response
[4-4-2022 20:23:54] warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
[4-4-2022 20:23:54] warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
[4-4-2022 20:23:54] closing bootloader
[4-4-2022 20:23:54] 	flush start
[4-4-2022 20:23:54] 	setting serial port timeouts to 1 ms
[4-4-2022 20:23:54] 	setting serial port timeouts to 1000 ms
[4-4-2022 20:23:54] 	flush complete
[4-4-2022 20:23:54] error: espcomm_upload_mem failed
[2022-04-04 20:23:54] DONE! BUT WITH (2) ERRORS!
############################
[2022-04-04 20:23:54] The COM port was already in use... the post-flash did not complete.
[2022-04-04 20:23:54] JOB COMPLETED, over and out!

arnemauer, 2022-04-04:
Bedankt, had je de kabel van de ducobox losgehaald voordat je begonnen bent met flashen?

xylone, 2022-04-04:
De tweede keer wel ja. Nadat ik de blank had geflased.
Dat hielp dus.

Ben alleen bang dat het originele probleem niet verholpen is.
Zie dat de uptime weer steeds 1 minuut is, nadat ik een device heb toegevoegd of een beetje rond klik.

xylone, 2022-04-04:
Oh wacht... ik las je vraag verkeerd. Hij was helemaal niet aan de ducobox aangesloten. (Heb hem meegenomen naar de pc om te flashen)
Ik had hem gedisconnect van de pc tussen het flashen door.

arnemauer, 2022-04-04:
Zou je hem willen aansluiten op je pc. En dan met putty een seriële verbinding maken met de gateway.
Vervolgens via de browser een aantal acties uitvoeren (plugin toevoegen etc.). In putty zou je moeten zien dat hij reboot. 
Zou je die log hier neer willen zetten? Dan kunnen we kijken wat de reden is voor de reboots.

xylone, 2022-04-04:
Op welke snelheid draait hij? 115200 zie ik geen output. 9600 is scrambled.

xylone, 2022-04-04:
Krijgt overigens nu deze melding bij het openen van de webpagina:
Low memory. Cannot display webpage :-(

xylone, 2022-04-04:
Op 115200 werkt het toch wel (na een paar keer proberen).
Direct een dump:
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x40209a32 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000                                                                                                                                                                        000

>>>stack>>>

ctx: sys
sp: 3ffff910 end: 3fffffb0 offset: 0190
3ffffaa0:  3ffffad0 76726573 3ffffbd2 00000000
3ffffab0:  3ffffdf0 3ffffe18 00000000 4027a25c
3ffffac0:  00000000 3ffffe18 3fff0158 4020ab6b
3ffffad0:  6f635f0f 6e61706d 2d6e6f69 6b6e696c
3ffffae0:  63745f04 6f6c0570 006c6163 00000000
3ffffaf0:  00000000 00000000 00000000 00000000
3ffffb00:  00000000 00000000 00000000 00000000
3ffffb10:  00000000 00000000 00000000 00000000
3ffffb20:  00000000 00000000 00000000 00000000
3ffffb30:  00000000 00000000 00000000 00000000
3ffffb40:  00000000 00000000 00000000 00000000
3ffffb50:  00000000 00000000 00000000 00000000
3ffffb60:  00000000 00000000 00000000 00000000
3ffffb70:  00000000 00000000 00000000 00000000
3ffffb80:  00000000 00000000 00000000 00000000
3ffffb90:  00000000 00000000 00000000 00000000
3ffffba0:  00000000 00000000 00000000 00000000
3ffffbb0:  00000000 00000000 00000000 00000000
3ffffbc0:  00000000 00000000 00000000 00000000
3ffffbd0:  000c001c 00000001 00001194 4020000c
3ffffbe0:  3ffffdf0 00000001 3fff0158 40207b3d
3ffffbf0:  00000000 53414812 72422053 65676469
3ffffc00:  43344620 04383341 7061685f 63745f04
3ffffc10:  6f6c0570 006c6163 00000000 00000000
3ffffc20:  00000000 00000000 00000000 00000000
3ffffc30:  00000000 00000000 00000000 00000000
3ffffc40:  00000000 00000000 00000000 00000000
3ffffc50:  00000000 00000000 00000000 00000000
3ffffc60:  00000000 00000000 00000000 00000000
3ffffc70:  00000000 00000000 00000000 00000000
3ffffc80:  00000000 00000000 00000000 00000000
3ffffc90:  00000000 00000000 00000000 00000000
3ffffca0:  00000000 00000000 00000000 00000000
3ffffcb0:  00000000 00000000 00000000 00000000
3ffffcc0:  00000000 00000000 00000000 00000000
3ffffcd0:  00000000 00000000 00000000 00000000
3ffffce0:  00000000 00000000 00000000 00000000
3ffffcf0:  00000000 00100024 00000001 ffffffff
3ffffd00:  402a2c47 3fff64a4 3ffec2d0 0000002d
3ffffd10:  000000dd 3fff64a4 08480d00 00000003
3ffffd20:  00000000 00000000 0000001f 40100878
3ffffd30:  3ffec2ec 00000016 3fffc228 40105f31
3ffffd40:  4000050c 4010484e 3ffee898 00000000
3ffffd50:  4029b1d7 00000030 00000010 ffffffff
3ffffd60:  4029878e 3fffb634 3fffb634 00000020
3ffffd70:  00000192 00000192 00000039 b237a8c0
3ffffd80:  b237a8c0 00000050 00000000 00000000
3ffffd90:  00000000 00000001 00000000 00000030
3ffffda0:  00000016 000000d4 00000020 3ffeec6c
3ffffdb0:  00000016 3ffef2f8 00000020 401015fb
3ffffdc0:  4010188e 3fffc200 00000002 40101880
3ffffdd0:  00000005 00000000 00000020 40100878
3ffffde0:  4010000a 3fff64a4 00000005 401023d8
3ffffdf0:  3ffea6d5 00000002 3fff9c4c 4020a679
3ffffe00:  40100000 00000000 00000000 00000001
3ffffe10:  3fff0000 00000000 00000000 4020a9ba
3ffffe20:  00000000 0000000b 3ffffe70 3fff9ccc
3ffffe30:  00000000 0544d272 40103612 000014e9
3ffffe40:  3fffaaec 3fff0158 3fff0158 402088cd
[... 89 lines cut ...]
3553 : Info   : [P150] DUCO RF GW: 3550 - Received messagetype: Normal message
3554 : Info   : [P150] DUCO RF GW: 3551 - No free space in outboxQ, dropping mes                                                                                                                                                                        sage;
3555 : Info   : [P150] DUCO RF GW: 3551 - No free space in outboxQ, dropping mes                                                                                                                                                                        sage;
3568 : Info   : WIFI : Connected! AP: Atlantis (7E:8A:20:4C:B7:9A) Ch: 6 Duratio                                                                                                                                                                        n: 999 ms
arag3836 : Info   : [P150] DUCO RF GW: 3824 - Received messagetype: Normal messa                                                                                                                                                                        ge
3837 : Info   : [P150] DUCO RF GW: 3829 - SEND ACK done!
3837 : Info   : [P150] DUCO RF GeW: 3829 - Received request for parameter value                                                                                                                                                                         command
3838 : Info   : [P150] DUCO RF GW: 3829 - SEND parameter 164 done
3839 : Info   : [P150t] DUCO RF GW: 3836 - send response
3839 : Info   : [P150] DUCO RF GW: 3836 - Start waiting for ack...
 2 734152 : Info   : [P150] DUCO RF GW: 4139 - Received messagetype: Normal mess                                                                                                                                                                        age
4152 : Info   : [P150] DUCO RF GW: 4145 - SEND ACK done!
W: 4145 - Received request for parameter value command
4154 : Info   : [P150] DUCO RF GW: 4145 - SEND parameter 164 done
4154 : Info   :
                [P150] DUCO RF GW: 4151 - send response
4155 : Info   : [P150] DUCO RF GW: 4151 - Start waiting for ack...
[... cut ...]    command
4448 : Info   : [P150] DUCO RF GW: 4439 - SEND parameter 164 done
4449 : Info   n: [P150] DUCO RF GW: 4446 - send response
4449 : Info   : [P150] DUCO RF GW: 4446 - Start waiting for ack...
sorinfo
4924 : Info   : WIFI : DHCP IP: 192.168.55.178 (VENTILATION-GATEWAY) GW: 192.168                                                                                                                                                                        .55.1 SN: 255.255.255.0   duration: 1374 ms

xylone, 2022-04-04:
En nog 1:

sensorinfo

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x4020f8ac epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff5340 end: 3fff5ae0 offset: 0190
3fff54d0:  00000000 00000000 00000000 00000000
3fff54e0:  40100d4e 00000000 00000000 40100ddc
3fff54f0:  c0038015 00000000 00000000 00000000
3fff5500:  00000000 00000000 00000000 00000022
3fff5510:  3fffc200 40100d14 3fffc258 4000050c
3fff5520:  40004378 00000030 00000016 ffffffff
3fff5530:  60000200 00000008 00000000 80000000
3fff5540:  20000000 3fff795c 80000000 20344760
3fff5550:  80000000 3fff75b0 00000000 3fff7960
3fff5560:  000000a0 00344760 00344800 00000030
3fff5570:  40106af0 00000000 00000000 00000000
3fff5580:  401008c9 3fff75c4 00344000 00344100
3fff5590:  12170002 000000ff 00344000 40277145
3fff55a0:  00000100 3fff739c 00000000 4027dbd4
3fff55b0:  00000000 3fff55f0 00000100 00343200
3fff55c0:  3fff75c4 00000100 3fff759c 00000100
3fff55d0:  3fff5652 00044000 00000022 4027f700
3fff55e0:  00343e04 00000080 00000000 00000f23
3fff55f0:  00000001 00000f22 00000000 00000001
3fff5600:  3fff5652 00343e04 3fff67a4 4027c040
3fff5610:  00000000 400042db 3fff67a4 40281fa0
3fff5620:  40004b31 3fff7900 00000100 00344700
3fff5630:  40106af0 3fff5690 3fff67a4 00000001
3fff5640:  401008c9 3fff7900 00344700 00344800
3fff5650:  00000000 400042db 00344700 40277145
3fff5660:  40004b31 3fff77ec 00000100 4027dbd4
3fff5670:  40106af0 00000000 00000006 00000020
3fff5680:  3fff7900 00000100 00344800 00344900
3fff5690:  00000007 3fff0022 00344800 3fff78ec
3fff56a0:  3fff759c 00344700 3fff67a4 4027c05c
3fff56b0:  00000001 00000448 3fff74a4 40281e9a
3fff56c0:  00000005 3fff5700 00000000 00000100
3fff56d0:  00344704 00000003 3fff759c 3fff77d8
3fff56e0:  3fff759c 00344800 3fff67a4 3fff74a8
3fff56f0:  00000447 00044700 3fff67a4 4027f454
3fff5700:  00000100 3fff729c 00000000 3fff77ec
3fff5710:  00000005 00000007 0000043e 00002000
3fff5720:  3fff679c 3fff96bc 00000014 000000fb
3fff5730:  3fff67a4 00000000 3fff74a4 40280ed8
3fff5740:  00000000 00000000 3fff575a 3fff3c88
3fff5750:  00050001 000000ff 04480447 40257ccb
3fff5760:  3fff729c 00000005 00000019 00000000
3fff5770:  00000014 00000448 00000000 00000000
3fff5780:  00000447 00000500 00000014 3fff76d8
3fff5790:  00000005 00000000 00000000 00000000
3fff57a0:  3fff5d5c 3fff96ac 3fff9c0c 3fff76d8
3fff57b0:  00000014 00000014 00000500 4027e105
3fff57c0:  3fff74a4 00000485 00000485 40101343
3fff57d0:  00000001 00000500 3fff76c4 0002d046
3fff57e0:  3fff74a4 00000000 3fff67a4 4027e196
3fff57f0:  3fff74a4 4bc6a7f0 0002d014 0002d046
3fff5800:  00000514 00000001 3fff67a4 4027ebad
3fff5810:  00000000 3fff58ec 3fff5850 4027c30f
3fff5820:  00000000 3fff58ec 3fff96ac 402773b2
3fff5830:  00000513 3fff58ec 0002d046 4024a4c4
3fff5840:  00000500 3fff58ec 3fff58e0 4025818d
3fff5850:  40287888 00000000 000003e8 40278be0
[... 103 lines cut ...]
2418 : Info   : [P150] DUCO RF GW: 2410 - SEND ACK done!
2418 : Info   : [P150] DUCO RF GnW: 2410 - Received request for parameter value command
2419 : Info   : [P150] DUCO RF GW: 2411 - SEND parameter 164 done
2420 : Info   : [P150] DUCO RF GW: 2417 - send response
2421 : Info   : [P150] DUCO RF GW: 2417 - Start waiting for ack...
ode2566 : Info   : WiFi : Scan finished, found: 5
2569 : Info   : WiFi : Best AP candidate: Atlantis 7E:8A:20:4C:B7:9A Ch:6 (-44dBm) WPA2/PSK
2570 : Info   : WIFI : Connecting Atlantis 7E:8A:20:4C:B7:9A Ch:6 (-44dBm) WPA2/PSK attempt #0
p3573 : Info   : WIFI : Connected! AP: Atlantis (7E:8A:20:4C:B7:9A) Ch: 6 Duration: 994 ms
3631 : Info   : [P150] DUCO RF GW: 3606 - Received messagetype: Normal message
3631 : Info   : [P150] DUCO RF GW: 3612 - SEND ACK done!
3632 : Info   : [P150] DUCO RF GW: 3612 - Received request for parameter value comamand
3632 : Info   : [P150] DUCO RF GW: 3612 - SEND parameter 166 done
3633 : Info   : [P150] DUCO RF GW: 3618 - send response
3634 : Info   : [P150] DUCO RF GW: 3619 - Start waiting for ack...
3635 : Info   : [P150] DUCO RF GW: 3619 - Received messagetype: Normal message
3635 : Info   : [P150] DUCO RF GW: 3624 - SEND ACK done!
3636 : Info   : [P150] DUCO RF GW: 3624 - Received request for parameter value command
3637 : Info   : [P150] DUCO RF GW: 3624 - SEND parameter 166 done
3638 : Info   : [P150] DUCO RF GW: 3630 - send response
r3638 : Info   : [P150] DUCO RF GW: 3630 - Start waiting for ack...
3707 : Info   : [P150] DUCO RF GW: 3706 - Received messagetype: Normal message
3708 : Info   : [P150] DUCO RF GW: 3706 - No free space in outboxQ, dropping message;
3709 : Info   : [P150] DUCO RF aGW: 3707 - No free space in outboxQ, dropping message;
get 2 73

xylone, 2022-04-04:
En voor de zekerheid nog 1:

sensorinfo

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x4020f8ac epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff5190 end: 3fff5ae0 offset: 0190
3fff5320:  00000002 4000444e 00000010 ffffffff
3fff5330:  40100d4e 4000422e 60000200 40100ddc
3fff5340:  c0038015 40004aa0 000000fc 0034ca04
3fff5350:  00000100 3fff53d0 3fffc718 00000022
3fff5360:  3fffc200 40100d14 3fffc258 4000050c
3fff5370:  4000437d 00000030 00000016 ffffffff
3fff5380:  60000200 00000000 3fff5460 80000000
3fff5390:  20000000 00000000 80000000 20348000
3fff53a0:  80000000 3fff75b0 00002c28 3fff7a14
3fff53b0:  00000100 00348000 00348100 00000030
3fff53c0:  00000000 00000000 0000001f 40100878
3fff53d0:  ffffff7e 4027711c 3fffc228 40105f31
3fff53e0:  4000050c 4000422e 60000200 3fff5484
3fff53f0:  4027f0c8 00000030 0000001b ffffffff
3fff5400:  00000000 400042db 00000100 00000000
3fff5410:  40004b31 3fff5460 00000004 0032ae2c
3fff5420:  40106af0 00000014 00000000 3fff67a4
3fff5430:  401008c9 3fff5460 0032ae00 00000002
3fff5440:  0034c904 4027711c 0032ae00 40277145
3fff5450:  00000004 0068e101 40106aa8 4027dbee
3fff5460:  00000000 400042db 00000000 00000000
3fff5470:  40004b31 3fff7a14 00000100 00348000
3fff5480:  40106af0 4027711c 3fff67a4 00000001
3fff5490:  401008c9 3fff7a14 00348000 00348100
3fff54a0:  30000000 2e6c6e2e 00348000 40277145
3fff54b0:  0000002e 3fff54f0 00000001 4027dbd4
3fff54c0:  fffffffc 86a00000 46ff0001 ffff0003
3fff54d0:  3fff7a14 00000100 00000000 00008001
3fff54e0:  3fffbfd4 0002ae00 3fff67a4 3fff7a00
3fff54f0:  3fff759c 00348000 3fff67a4 4027c05c
3fff5500:  034b00e1 0351034e 03550354 40281e9a
3fff5510:  00000100 3fff739c 00000000 00000100
3fff5520:  00000460 0000001e 00000000 00000000
3fff5530:  00000000 00000000 00000000 00000100
3fff5540:  3fff67a4 00048000 00000024 4027f700
3fff5550:  00000000 00000080 00000000 00000038
3fff5560:  00000000 00000037 4027f09c 00000000
3fff5570:  3fff739c 00000000 00000020 00000100
3fff5580:  0000001f 00000000 00000000 00000000
3fff5590:  3fff5da4 00000010 00000020 3fff3c88
3fff55a0:  3fff5600 3fff5630 3fff67a4 40281152
3fff55b0:  3fffbfd4 00000000 3fff55c4 3fff55c0
3fff55c0:  0000001e 00000023 00000020 40101632
3fff55d0:  3fff5652 0004c000 00000026 4027f700
3fff55e0:  0034ca04 00000080 00000000 3fff3c88
3fff55f0:  3fff3cbc 3fff5630 3fff67a4 4027eaa9
3fff5600:  3fff5652 0034ca04 3fff67a4 4027c040
3fff5610:  3fffbfd4 0034ca04 3fff67a4 00000000
3fff5620:  402d4808 3fff3cbc 3fff679c 4027c5aa
3fff5630:  3fffa774 0000000c 3fff56a0 40278d5b
3fff5640:  0000000c 0004ca00 3fff67a4 4027ffa1
3fff5650:  3f7e0000 0000000c 3fff56a0 40278da4
3fff5660:  3fffbfd4 3fff56fc 3fff56a0 40278f90
3fff5670:  00000000 00000080 3fff5d5c 40286d19
3fff5680:  3fff56fc 0000000c 3fff56fc 402772dc
3fff5690:  00000000 402d4808 00000000 40257d39
3fff56a0:  3fffbfd4 000c000f 00ff56fc 40279290
[... 140 lines cut ...]
3623 : Info   : [P150] DUCO RF GW: 3615 - No free space in outboxQ, dropping message;
3624 : Info   : [P150] DUCO RF GW: 3615 - No frree space in outboxQ, dropping message;
3637 : Info   : WIFI : Connected! AP: Atlantis (7E:8A:20:4C:B7:9A) Ch: 6 Duration: 991 ms
aget 2 74
4227 : Info   : [P150] DUCO RF GW: 4214 - Received messagetype: Normal message
4227 : Info   : [P150] DUCO RF GW: 4220 - SEND ACK done!
4228 : Info   : [P150] DUCO RF GW: 4220 - Received request for parameter value command
4229 : Info   : [P150] DUCO RF GW: 4220 - SEND parameter 137 done
4229 : Info   : [P150] DUCO RF GW: 4226 - send response
4230 : Info   : [P150] DUCO RF GW: 4226 - Start waiting for ack...
4421 : Info   : [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: temp
nod4579 : Info   : [P150] DUCO RF GW: 4567 - Received messagetype: Normal message
4580 : Info   : [P150] DUCO RF GW: 4572 - SEND ACK done!
4580 : Info   : [P150] DUCO RF GeW: 4572 - Received request for parameter value command
4581 : Info   : [P150] DUCO RF GW: 4572 - SEND parameter 137 done
4582 : Info   : [P150] DUCO RF GW: 4579 - send response
4583 : Info   : [P150] DUCO RF GW: 4579 - Start waiting for ack...
parage4914 : Info   : [P150] DUCO RF GW: 4902 - Received messagetype: Normal message
4915 : Info   : [P150] DUCO RF GW: 4907 - SEND ACK done!
4916 : Info   : [P150] DUCO RF GtW: 4907 - Received request for parameter value command
4916 : Info   : [P150] DUCO RF GW: 4908 - SEND parameter 137 done
4917 : Info   :  [P150] DUCO RF GW: 4914 - send response
4918 : Info   : [P150] DUCO RF GW: 4914 - Start waiting for ack...
2 73
sensorinfo

xylone, 2022-04-04:
Nog een laatste: (Bij deze was ik alleen aan het rond klikken, de andere was ik devices aan het aanpassen of toevoegen)

sensorinfo

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x40209a32 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3ffff910 end: 3fffffb0 offset: 0190
3ffffaa0:  3ffffad0 76726573 3ffffbd2 00000000
3ffffab0:  3ffffdf0 3ffffe18 00000000 4027a25c
3ffffac0:  00000000 3ffffe18 3fff0158 4020ab6b
3ffffad0:  07626c02 736e645f 0464732d 7064755f
3ffffae0:  636f6c05 00006c61 00000000 00000000
3ffffaf0:  00000000 00000000 00000000 00000000
3ffffb00:  00000000 00000000 00000000 00000000
3ffffb10:  00000000 00000000 00000000 00000000
3ffffb20:  00000000 00000000 00000000 00000000
3ffffb30:  00000000 00000000 00000000 00000000
3ffffb40:  00000000 00000000 00000000 00000000
3ffffb50:  00000000 00000000 00000000 00000000
3ffffb60:  00000000 00000000 00000000 00000000
3ffffb70:  00000000 00000000 00000000 00000000
3ffffb80:  00000000 00000000 00000000 00000000
3ffffb90:  00000000 00000000 00000000 00000000
3ffffba0:  00000000 00000000 00000000 00000000
3ffffbb0:  00000000 00000000 00000000 00000000
3ffffbc0:  00000000 00000000 00000000 00000000
3ffffbd0:  000c0017 00000001 00001194 40200002
3ffffbe0:  3ffffdf0 00000001 3fff0158 40207b3d
3ffffbf0:  00000000 53414812 72422053 65676469
3ffffc00:  43344620 04383341 7061685f 63745f04
3ffffc10:  6f6c0570 006c6163 00000000 00000000
3ffffc20:  00000000 00000000 00000000 00000000
3ffffc30:  00000000 00000000 00000000 00000000
3ffffc40:  00000000 00000000 00000000 00000000
3ffffc50:  00000000 00000000 00000000 00000000
3ffffc60:  00000000 00000000 00000000 00000000
3ffffc70:  00000000 00000000 00000000 00000000
3ffffc80:  00000000 00000000 00000000 00000000
3ffffc90:  00000000 00000000 00000000 00000000
3ffffca0:  00000000 00000000 00000000 00000000
3ffffcb0:  00000000 00000000 00000000 00000000
3ffffcc0:  00000000 00000000 00000000 00000000
3ffffcd0:  00000000 00000000 00000000 00000000
3ffffce0:  00000000 00000000 00000000 00000000
3ffffcf0:  00000000 00100024 00000001 3ffede40
3ffffd00:  3fffb58a 40105cdb 3fffa6a8 3fff5f08
3ffffd10:  3fff674c 00000000 00000020 40100878
3ffffd20:  3ffeaf50 40297cb1 3fffa6a8 3fff5f08
3ffffd30:  40104787 00000032 00000002 00040000
3ffffd40:  00002200 3ffee110 00000010 00000000
3ffffd50:  3fff62cc 3fff6298 3fffb554 40297f3e
3ffffd60:  3ffeaf8c 3fff6298 3fffb554 4029835c
3ffffd70:  3fff61b0 2c9f0300 4000050c 3fffc278
3ffffd80:  00003780 3fffc200 00000022 3fff61b4
3ffffd90:  00000000 00000000 0000001f 40100878
3ffffda0:  00000016 000000d4 3fffc228 40105f31
3ffffdb0:  4000050c 3ffef2f8 00000020 401015fb
3ffffdc0:  4029fe37 00000030 00000010 ffffffff
3ffffdd0:  00000005 00000000 00000020 40100878
3ffffde0:  0000000a ec37a8c0 00000005 401023d8
3ffffdf0:  3ffea6d5 00000002 3fffa02c 4020a679
3ffffe00:  40100000 00000000 00000000 00000001
3ffffe10:  3fff0000 00000000 00000000 4020a9ba
3ffffe20:  00000000 0000000c 3ffffe70 3fffa154
[... 97 lines cut ...]
ra3766 : Info   : [P150] DUCO RF GW: 3766 - Received messagetype: Normal message
3767 : Info   : [P150] DUCO RF GW: 3766 - No free space in outboxQ,g dropping message;
3768 : Info   : [P150] DUCO RF GW: 3766 - No free space in outboxQ, dropping message;
et 2 74078 : Info   : [P150] DUCO RF GW: 4066 - Received messagetype: Normal message
4079 : Info   : [P150] DUCO RF GW: 4071 - SEND ACK done!
4080 : Info   : [P150] DUCO RF G4W: 4071 - Received request for parameter value command
4080 : Info   : [P150] DUCO RF GW: 4072 - SEND parameter 164 done
[P150] DUCO RF GW: 4078 - send response
4082 : Info   : [P150] DUCO RF GW: 4078 - Start waiting for ack...

4307 : Info   : [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: temp
n4401 : Info   : [P150] DUCO RF GW: 4395 - Received messagetype: Normal message
4402 : Info   : [P150] DUCO RF GW: 4400 - SEND ACK done!
4402 : Info   : [P150] DUCO RF oGW: 4400 - ???
4403 : Info   : [P150] DUCO RF GW: 4401 - ???
4403 : Info   : [P150] DUCO RF GW: 4401 - ???
4404 : Info   : [P150] DUCO dRF GW: 4401 - No data to send!
eparaget 2 74991 : Info   : [P150] DUCO RF GW: 4979 - Received messagetype: Normal message
4992 : Info   : [P150] DUCO RF GW: 4984 - SEND ACK done!
4992 : Info   : [P150] DUCO RF G3W: 4984 - Received request for parameter value command
4993 : Info   : [P150] DUCO RF GW: 4984 - SEND parameter 166 done
4994 : Info   : [P150] DUCO RF GW: 4991 - send response
4994 : Info   : [P150] DUCO RF GW: 4991 - Start waiting for ack...

sensorinfo

xylone, 2022-04-04:
Wilde er net nog 1 posten, omdat hij ook crashed bij het opslaan van de settings.
Vervolgens wilde hij de webpage niet meer laden en zag ik hem nogmaals rebooten in de putty log.
Het downloaden van de settings triggered het dus direct.
Wellicht kun je daar iets mee?
Denk dat het een geheugen probleem is op de ESP.
Zodra je meer geheugen probeert te gebruiken alloceert hij waarschijnlijk een stuk wat niet helemaal goed is, of buiten het bereik is?

treb33, 2022-04-04:
Ik heb het inmiddels werkend door via Putty het reset command te geven. Hierna geflashed en weer alles ingesteld volgens de wiki. Draait heel de avond al stabiel.

Jimboer, 2022-04-05:
Zelfde probleem hier met vorige week ontvangen gateway.
Na volgen wiki werkte in eerste instantie alles. Daarna gateway niet meer te benaderen, ook niet meer via 'rechtstreeks' wifi.
Na 24 uur uit te hebben gestaan, kon ik hem weer benaderen. Alle setting waren gewist. Opnieuw alles ingesteld en draait nu, maar wel met boot exceptions

[image]

xylone, 2022-04-05:
> Ik heb het inmiddels werkend door via Putty het reset command te geven. Hierna geflashed en weer alles ingesteld volgens de wiki. Draait heel de avond al stabiel.

Wat bedoel je precies? reset in de console van putty intikken?

treb33, 2022-04-05:
> > Ik heb het inmiddels werkend door via Putty het reset command te geven. Hierna geflashed en weer alles ingesteld volgens de wiki. Draait heel de avond al stabiel.
> 
> 
> 
> Wat bedoel je precies? reset in de console van putty intikken?

Klopt inderdaad.

arnemauer, 2022-04-05:
> > > Ik heb het inmiddels werkend door via Putty het reset command te geven. Hierna geflashed en weer alles ingesteld volgens de wiki. Draait heel de avond al stabiel.
> > 
> > 
> > Wat bedoel je precies? reset in de console van putty intikken?
> 
> Klopt inderdaad.

Had je eerst de blank_4mb.bin geflashed voordat je de firmware hebt geflashed?

arnemauer, 2022-04-05:
> Zelfde probleem hier met vorige week ontvangen gateway. Na volgen wiki werkte in eerste instantie alles. Daarna gateway niet meer te benaderen, ook niet meer via 'rechtstreeks' wifi. Na 24 uur uit [...]

Boot exceptions zijn in principe niet erg zolang ze niet ieder uur voorkomen :)

xylone, 2022-04-05:
Bij mij loopt hij toch wel op....
[image]
[image]

arnemauer, 2022-04-05:
@xylone Welke plugins heb je nu allemaal aanstaan? In jou espeasy logs zie ik een aantal dingen staan die ik ga uitzoeken.

xylone, 2022-04-05:
[image]

arnemauer, 2022-04-05:
@xylone Ik denk dat er een bug zit in de RF plugin. Zou je deze tijdelijk kunnen uitschakelen (niet verwijderen)? En dan de uptime checken.

xylone, 2022-04-05:
Dat is alleen de remote toch? Zojuist uitgezet.

arnemauer, 2022-04-05:
@xylone  klopt.

xylone, 2022-04-05:
Dit lijkt het inderdaad 'op te lossen':
[image]

treb33, 2022-04-05:
> @treb33 Had je eerst de blank_4mb.bin geflashed voordat je de firmware hebt geflashed? 
> 
> 
> 
> @Jimboer Boot exceptions zijn in principe niet erg zolang ze niet ieder uur voorkomen :) 

@arnemauer yes had ik gedaan.

Jimboer, 2022-04-05:
> @xylone Ik denk dat er een bug zit in de RF plugin. Zou je deze tijdelijk kunnen uitschakelen (niet verwijderen)? En dan de uptime checken.

Ik heb de RF plugin ook niet draaien en is nu stabiel.

treb33, 2022-04-05:
Wilde de RF gateway net aansturen (via HA) maar reageerde erg traag zie ik dat hij net is gereboot. 
Build:⋄
20116  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
AFM 03f6eea: Mar 31 2022
Plugin Count:⋄
55 [Normal]
Build Time:⋄
Mar 31 2022 18:49:16
Binary Filename:⋄
ESP_Easy_mega_20220331_custom_ESP8266_4M1M
Build Platform:⋄
Linux-5.13.19-1-pve-x86_64-with-glibc2.34
Git HEAD:⋄
pygit2_not_installed

De RF gateway nu ook tijdelijk uitgeschakeld.

xylone, 2022-04-06:
Draait nog steeds, dus dat is inderdaad het probleem. Hopelijk kun je het vinden en oplossen.
Mocht je nog meer info willen hebben, of iets willen testen, dan hoor ik het graag!
[image]

xylone, 2022-04-06:
Kan het zijn dat met deze firmware er geen data van de Serial GW meekomt?
Deze blijven allemaal 0:
[image]

arnemauer, 2022-04-06:
Wat staat er in de weblog? 
Het uitlezen van waardes kun je beter niet vaker dan 1x per 60 seconden doen omdat de plugins anders geen tijd hebben om verbinding te maken met de ducobox. Daardoor ontstaat een wachtrij waardoor sommige plugins niet aan de beurt komen.

xylone, 2022-04-06:
Heb de optie niet om debug aan te vinken: (De algemene debug staat wel aan)
[image]

Heb alle plugins op 60 sec gezet. (Wist niet meer waar ze op stonden na de firmware update ;-) )

arnemauer, 2022-04-06:
Die debug optie zit alleen in de taak waarbij output type ventilator status is.

xylone, 2022-04-06:
Ja gevonden! Pfff ook deze plugin was ik vergeten.... 
Eerst keer netjes de gehele wiki doorgelopen, maar na de 'reset' en nieuwe firmware dacht ik wel wat ik moest toevoegen... Niet dus...
Nu krijg ik inderdaad alles gewoon binnen. Dank!

Al enig idee wat het probleem met de RF-plugin is?

arnemauer, 2022-04-06:
Mooizo :) De "main plugin" zorgt ervoor dat de data wordt opgevraagd bij de ducobox. De andere taken laten puur en alleen de waarde zien die de "main plugin" heeft uitgelezen. Vandaar dat je niks zag. Wel een goed verbeterpunt.... hij checkt namelijk wel of de main plugin al is geactiveerd zodat je hem niet een tweede keer kan activeren maar hij checkt niet of hij uberhaupt geactiveerd is. In dat geval zouden de overige "value types" grijs moeten zijn.

arnemauer, 2022-04-06:
> Al enig idee wat het probleem met de RF-plugin is?

Ik ben het nog aan het uitzoeken. Ik verwacht het weekend een nieuwe firmware te kunnen publiceren.
Dit soort bugs zijn lastig te vinden omdat ze vrij "random" voorkomen.

arnemauer, 2022-04-10:
Hierbij de nieuwe firmware met diverse fixes:
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/3f5bdcebc17956ac53e2c2c3c65e1a93545074d7

Schrijf de instellingen (networkId, node address etc.) van de plugins op voordat je de update gaat uitvoeren.

Updaten is ook mogelijk via de webinterface: Tools -> Update.
Disable eerst de RF plugin voordat je de update uitvoert!

xylone, 2022-04-11:
Het lukt mij niet om de firmware via de webinterface te uploaden (Krijg een connection reset na een paar seconden), dus ik zal het vanavond even met de kabel doen.

xylone, 2022-04-11:
Firmware geflashed en draait nu. 
Moest alleen de network ID even opnieuw opgeven.
Rest van de settings waren bewaard gebleven.

arnemauer, 2022-04-11:
> Wel wordt de log redelijk overladen met logs van de RF GW. (Log vinkje staat uit) 

Klopt, bij niet alle debug berichten wordt nu gekeken naar het vinkje. Komt nog :)

> En ik heb het idee dat de ventilatie mode nu niet meer wordt uitgelezen. Waarde van de RF GW als de Serial GW blijven op 0 staan, de rest wordt wel geupdate: [image]

Staat de ducobox in een bepaalde stand? Aangezien countdown op 0 staat en ventilatiemodus op auto kan het ook zijn dat een van je sensoren een hogere afzuiging afdwingt i.v.m. bijvoorbeeld hoge vochtigheid of co2.

xylone, 2022-04-12:
Oke, top!
Ahhh ok, ja het werkt inderdaad wel als je het zelf veranderd via de weginterface of via een MQTT command en ook op de CO2 Sensor de stand aanpast.
Ik was in de veronderstelling dat ook de automatisch wisselingen op basis van CO2 of vochtigheid zichtbaar zouden zijn.
Om zo'n overzicht te maken, zodat je zou kunnen zien of hij aangaat op basis van de luchtkwaliteit:
[image]

Helaas lijkt hij toch nog af en toe te herstarten:
[image]

JJ-82, 2022-04-12:
Hier ook nog de nodige 'exceptions'. Weliswaar geen 16, maar 8... Ik heb momenteel geen automations draaien die wijzigingen op de ventilatiestand vragen. Laatste reboot is van 20 uur geleden. Omstreeks hetzelfde moment dat de laatste keer de ventilatiestand (via de reguliere wandbediening) gewijzigd is.

Jimboer, 2022-04-12:
Het is gelukt on RF gateway weer aan te zetten na firmwareversie van gisteren! Lijkt nu stabiel op eerste gezicht.

JJ-82, 2022-04-12:
> @JJ-82 heb je de filmwereld van gisteren erop gezet? Daar zitten nog een aantal bugfixes in.

Ik heb de versie van zondag erop gezet. Dat er gisteren alweer een verse was, heb ik blijkbaar gemist. Zet hem er morgenochtend gelijk op.😃

arnemauer, 2022-05-30:
Ik heb alle bugs gefixed en de gateway loopt nu weer stabiel :)
Ik hoor graag of de nieuwe firmware bij jullie ook goed draait. 
e0e49de

closed 2022-10-25 by arnemauer