arnemauer/Ducobox-ESPEasy-Plugin #91: CO2 waarde Domoticz als 'Luchtkwaliteit" werkt niet

Jimboer, 2022-04-05:
CO2 in Domoticz  werkt bij mij alleen als ik sensortype "Custom Sensor" gebruik.  Type "Luchtkwaliteit" blijft waarde altijd op 0 staan.
Zou te maken kunnen hebben dat er een integer moet worden doorgegeven, dus zonder decimalen.

arnemauer, 2022-04-05:
Wat staat er in de domoticz log en de espeasy log wanneer je type luchtkwaliteit gebruikt?

Jimboer, 2022-04-05:
18984895: [P154] DUCO Ext. CO2 Sensor: start read, eventid:2
18984896: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
18984896: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: CO2
18984930: HTTP : C001 Success! HTTP/1.1 200 OK^
18985256: [P154] DUCO Ext. CO2 Sensor: Received invalid response
18985256: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
18991961: WD : Uptime 317 ConnectFailures 0 FreeMem 6384 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
18993053: [P153] DUCO BOX SENSOR: start read, eventid:7
18993053: [P153] DUCO BOX SENSOR: Start readBoxSensors
18993090: HTTP : C001 Success! HTTP/1.1 200 OK^
18993296: [P153] DUCO BOX SENSOR: Received correct response on sensorinfo
18993301: [P153] DUCO BOX SENSOR: RH: 5265 = 52.65%
18993304: [P153] DUCO BOX SENSOR: TEMP: 203 = 20.3Â°C
18993304: [P153] DUCO BOX SENSOR: End of message received.
18993304: [P153] DUCO BOX SENSOR: Stop reading from serial
19020818: [P151] DUCO SER GW: New Countdown value: 0
19020819: [P151] DUCO SER GW: New Ventilation_Percentage value: 52
19020819: [P151] DUCO SER GW: New Current_Fan_Value value: 30
19020916: HTTP : C001 Success! HTTP/1.1 200 OK^
19020937: [P151] DUCO SER GW: End of message received.
19020937: [P151] DUCO SER GW: Stop reading from serial
19021750: [P151] DUCO SER GW: New value for Ventilation_Percentage
19021750: [P154] DUCO Ext. CO2 Sensor: start read, eventid:3
19021750: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
19021751: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: CO2
19021783: HTTP : C001 Success! HTTP/1.1 200 OK^
19021960: WD : Uptime 317 ConnectFailures 0 FreeMem 6144 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
19022094: [P154] DUCO Ext. CO2 Sensor: Received invalid response
19022094: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
19022746: [P154] DUCO Ext. CO2 Sensor: start read, eventid:4
19022746: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
19022747: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: temp
19023090: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
19024746: [P154] DUCO Ext. CO2 Sensor: Serial reading timeout
19024746: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
19024747: [P154] DUCO Ext. CO2 Sensor: start read, eventid:5
19024747: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
19024747: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: temp
19025090: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
19025918: [P154] DUCO Ext. CO2 Sensor: Received invalid response
19025919: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
19026746: [P156] DUCO FanSpeed: start read, eventid:6
19026746: [P156] DUCO FanSpeed: Start reading fanspeed
19026924: [P156] DUCO FanSpeed: Received correct response on fanspeed
19026928: [P156] DUCO FanSpeed: End of message received.
19026928: [P156] DUCO FanSpeed: Stop reading from serial
19044895: [P154] DUCO Ext. CO2 Sensor: start read, eventid:2
19044896: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
19044896: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: CO2
19044929: HTTP : C001 Success! HTTP/1.1 200 OK^
19045221: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
19045237: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
19045265: [P154] DUCO Ext. CO2 Sensor: CO2 PPM: 609 = 609 PPM
19045266: [P154] DUCO Ext. CO2 Sensor: End of message received.
19045266: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
19051960: WD : Uptime 318 ConnectFailures 0 FreeMem 6312 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
19053053: [P153] DUCO BOX SENSOR: start read, eventid:7
19053053: [P153] DUCO BOX SENSOR: Start readBoxSensors
19053094: HTTP : C001 Success! HTTP/1.1 200 OK^
19053302: [P153] DUCO BOX SENSOR: Received invalid response
19053302: [P153] DUCO BOX SENSOR: Stop reading from serial
19080738: [P154] DUCO Ext. CO2 Sensor: Serial port in use by taskid 0, set flag to read data later.
19080761: [P151] DUCO SER GW: Received invalid response on network
19080762: [P151] DUCO SER GW: Stop reading from serial
19080791: HTTP : C001 Success! HTTP/1.1 200 OK^
19080799: [P154] DUCO Ext. CO2 Sensor: start read, eventid:5
19080799: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
19080800: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: temp
19080802: [P154] DUCO Ext. CO2 Sensor: Received invalid response
19080802: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
[... 4 lines cut ...]
19080924: HTTP : C001 Success! HTTP/1.1 200 OK^
19081746: [P154] DUCO Ext. CO2 Sensor: start read, eventid:3
19081746: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
19081747: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: CO2
19081961: WD : Uptime 318 ConnectFailures 0 FreeMem 6992 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
19082084: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
19082086: [P154] DUCO Ext. CO2 Sensor: End of message received.
19082086: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
19082747: [P154] DUCO Ext. CO2 Sensor: start read, eventid:4
19082747: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
19082748: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: temp
19083102: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
[... cut ...]
19104896: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
19104896: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: CO2
19104931: HTTP : C001 Success! HTTP/1.1 200 OK^
19105223: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
19105237: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
19105565: [P154] DUCO Ext. CO2 Sensor: CO2 PPM: 610 = 610 PPM
19105566: [P154] DUCO Ext. CO2 Sensor: End of message received.
19105567: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial

Jimboer, 2022-04-05:
In Domoticz is niets te vinden in log over dummy apparaten.

arnemauer, 2022-04-05:
Wil je onder Tools -> Advanced -> Web Log Level op "debug dev" willen zetten? En dan de log willen posten?

Jimboer, 2022-04-05:
Hij haalt alleen geen co2 waarden meer op met debug dev

567760: WD   : Uptime 8 ConnectFailures 0 FreeMem 6800 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
597759: WD   : Uptime 9 ConnectFailures 0 FreeMem 6968 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
600597: [P154] DUCO Ext. CO2 Sensor: start read, eventid:5
600598: [P154] DUCO Ext. CO2 Sensor: Serial port in use by taskid 0, set flag to read data later.
600610: [P151] DUCO SER GW: Ducobox row found!
600610: [P151] DUCO SER GW: New Ventilationmode value: 0
600611: [P151] DUCO SER GW: New Countdown value: 0
600611: [P151] DUCO SER GW: New Ventilation_Percentage value: 10
600611: [P151] DUCO SER GW: New Current_Fan_Value value: 10
600623: [P156] DUCO FanSpeed: start read, eventid:6
600623: [P156] DUCO FanSpeed: Serial port in use, set flag to read data later.
600660: [P153] DUCO BOX SENSOR: start read, eventid:7
600660: [P153] DUCO BOX SENSOR: Serial port in use, set flag to read data later.
600708: HTTP : C001 Success! HTTP/1.1 200 OK^
600733: [P151] DUCO SER GW: End of message received.
600733: [P151] DUCO SER GW: Stop reading from serial
600847: HTTP : C001 Success! HTTP/1.1 200 OK^
600974: HTTP : C001 Success! HTTP/1.1 200 OK^
601726: [P154] DUCO Ext. CO2 Sensor: start read, eventid:2
601726: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
601727: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: CO2
602069: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
602726: [P154] DUCO Ext. CO2 Sensor: start read, eventid:3
602726: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
602727: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: CO2
603012: [P154] DUCO Ext. CO2 Sensor: Received invalid response
603012: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
603012: [P154] DUCO Ext. CO2 Sensor: End of message received.
603013: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
603726: [P154] DUCO Ext. CO2 Sensor: start read, eventid:4
603726: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
603727: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: temp
604072: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
604076: [P154] DUCO Ext. CO2 Sensor: End of message received.
604076: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
604726: [P154] DUCO Ext. CO2 Sensor: start read, eventid:5
604726: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
604727: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: temp
605066: [P154] DUCO Ext. CO2 Sensor: Received invalid response
605066: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
605726: [P156] DUCO FanSpeed: start read, eventid:6
605726: [P156] DUCO FanSpeed: Start reading fanspeed
605922: [P156] DUCO FanSpeed: Send command successfull!
606009: [P156] DUCO FanSpeed: Received invalid response
606009: [P156] DUCO FanSpeed: Stop reading from serial
606010: [P156] DUCO FanSpeed: End of message received.
606010: [P156] DUCO FanSpeed: Stop reading from serial
606726: [P153] DUCO BOX SENSOR: start read, eventid:7
606727: [P153] DUCO BOX SENSOR: Start readBoxSensors
606949: [P153] DUCO BOX SENSOR: Received invalid response
606950: [P153] DUCO BOX SENSOR: Stop reading from serial
627760: WD   : Uptime 9 ConnectFailures 0 FreeMem 6848 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init

arnemauer, 2022-04-06:
Waar staat de interval voor het uitlezen van de plugins nu op ingesteld?

closed 2022-04-11 by arnemauer