arnemauer/Ducobox-ESPEasy-Plugin #92: Join werkt niet

arjan-io, 2022-04-06:
Log:
`630313: [P150] DUCO RF GW: 630313 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00080fcf;Type:7; Bytes:13; Counter:2;
630314: [P150] DUCO RF GW: 630313 - 20,00,52,00,
630314: [P150] DUCO RF GW: 630313 - Ignore message, not our networkID.
631537: [P150] DUCO RF GW: 631536 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00084391;Type:7; Bytes:13; Counter:4;
631537: [P150] DUCO RF GW: 631536 - 20,00,52,00,
631538: [P150] DUCO RF GW: 631537 - Ignore message, not our networkID.
649426: WD : Uptime 8 ConnectFailures 3 FreeMem 6280 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
649812: HTTP: JOIN
649813: Command: JOIN
649814: [P150] DUCO RF GW: Sent command for 'join' to DUCO unit
649820: [P150] DUCO RF GW: 649814 - SendJoinPacket()
649821: [P150] DUCO RF GW: 649814 - SENT message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type:1; Bytes:13; Counter:1;
649821: [P150] DUCO RF GW: 649815 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
649821: [P150] DUCO RF GW: 649820 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
660364: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
660365: [P150] DUCO RF GW: State: 0.00
664158: [P150] DUCO RF GW: 664157 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008438d;Type:7; Bytes:13; Counter:2;
664158: [P150] DUCO RF GW: 664157 - 20,00,52,00,
664159: [P150] DUCO RF GW: 664158 - Ignore message, not our networkID.
679364: WD : Uptime 9 ConnectFailures 3 FreeMem 5824 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
683986: [P150] DUCO RF GW: EXIT PLUGIN_150
689809: [P150] DUCO RF GW: EXIT PLUGIN_150
709365: WD : Uptime 9 ConnectFailures 3 FreeMem 6840 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
739365: WD : Uptime 10 ConnectFailures 3 FreeMem 6800 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
749300: [P150] DUCO RF GW: 0-0-0-0
749418: [P150] DUCO RF GW: Extra Settings PLUGIN_150 loaded
749429: [P150] DUCO RF GW: Values set from config. DeviceID: 0, networkId: 0000, radio power: Default - radio power 0,6 dBm (0x8D)
749432: [P150] DUCO RF GW: CC1101 868Mhz transmitter initialized
749735: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
749735: [P150] DUCO RF GW: State: 0.00
769364: WD : Uptime 10 ConnectFailures 3 FreeMem 6192 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
787532: [P150] DUCO RF GW: 787531 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00084608;Type:7; Bytes:13; Counter:15
787532: [P150] DUCO RF GW: 787532 - 20,00,52,00,
787533: [P150] DUCO RF GW: 787532 - Ignore message, not our networkID.
813427: [P150] DUCO RF GW: 813426 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008473a;Type:7; Bytes:13; Counter:4;
813427: [P150] DUCO RF GW: 813426 - 20,00,52,00,
813427: [P150] DUCO RF GW: 813427 - Ignore message, not our networkID.
813431: WD : Uptime 11 ConnectFailures 4 FreeMem 6192 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
813434: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
813434: [P150] DUCO RF GW: State: 0.00
828684: MQTT : Connected to broker with client ID: Ventilation_gateway_1_1_1
828686: Subscribed to: Ventilation_gateway_1_1/#
829365: WD : Uptime 11 ConnectFailures 3 FreeMem 6200 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
837931: [P150] DUCO RF GW: 837930 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008462b;Type:0; Bytes:10; Counter:8;
837931: [P150] DUCO RF GW: 837930 - 00,
837931: [P150] DUCO RF GW: 837930 - Ignore message, not our networkID.
842555: [P150] DUCO RF GW: 842554 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008462b;Type:7; Bytes:13; Counter:9;
842555: [P150] DUCO RF GW: 842554 - 20,00,52,00,
842555: [P150] DUCO RF GW: 842555 - Ignore message, not our networkID.
845573: [P150] DUCO RF GW: 845572 - RECEIVED message: SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0007a953;Type:0; Bytes:10; Counter:1;
845573: [P150] DUCO RF GW: 845573 - 01,
845574: [P150] DUCO RF GW: 845573 - Ignore message, not our networkID.
852471: HTTP: JOIN
852471: Command: JOIN
852472: [P150] DUCO RF GW: Sent command for 'join' to DUCO unit
852479: [P150] DUCO RF GW: 852472 - SendJoinPacket()
852479: [P150] DUCO RF GW: 852473 - SENT message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type:1; Bytes:13; Counter:1;
852479: [P150] DUCO RF GW: 852473 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
852480: [P150] DUCO RF GW: 852479 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
859365: WD : Uptime 12 ConnectFailures 3 FreeMem 6168 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
869532: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
869533: [P150] DUCO RF GW: State: 0.00
889364: WD : Uptime 12 ConnectFailures 3 FreeMem 5992 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
913730: [P150] DUCO RF GW: 913729 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008296a;Type:0; Bytes:10; Counter:10
913730: [P150] DUCO RF GW: 913730 - 00,
913731: [P150] DUCO RF GW: 913730 - Ignore message, not our networkID.
918148: [P150] DUCO RF GW: 918148 - RECEIVED message: SRC:2; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0007a9ea;Type:7; Bytes:13; Counter:8;
918149: [P150] DUCO RF GW: 918148 - 20,00,52,00,
918149: [P150] DUCO RF GW: 918148 - Ignore message, not our networkID.
[... 112 lines cut ...]
1087830: [P150] DUCO RF GW: 1087829 - Ignore message, not our networkID.
1087952: [P151] DUCO SER GW: PCF DATA addr 0x38 value 0xc1
1091214: [P151] DUCO SER GW: PCF DATA addr 0x38 value 0x41
1091215: [P150] DUCO RF GW: 1091215 - RECEIVED message: SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:0007a953;Type:6; Bytes:9; Counter:9;
1091216: [P150] DUCO RF GW: 1091215 -
1091216: [P150] DUCO RF GW: 1091215 - Ignore message, not our networkID.
1091282: [P151] DUCO SER GW: PCF DATA addr 0x38 value 0xc1
1092120: [P151] DUCO SER GW: PCF DATA addr 0x38 value 0x41
1092122: [P150] DUCO RF GW: 1092121 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00084608;Type:0; Bytes:10; Counter:2
1092122: [P150] DUCO RF GW: 1092121 - 00,
1092122: [P150] DUCO RF GW: 1092121 - Ignore message, not our networkID.
1092182: [P151] DUCO SER GW: PCF DATA addr 0x38 value 0xc1
1099364: WD : Uptime 16 ConnectFailures 3 FreeMem 6048 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1100602: [P151] DUCO SER GW: PCF DATA addr 0x38 value 0x41
1100604: [P150] DUCO RF GW: 1100603 - RECEIVED message: SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0007a953;Type:7; Bytes:17; Counter:1
1100604: [P150] DUCO RF GW: 1100603 - 60,00,22,00,09,00,00,08,
1100604: [P150] DUCO RF GW: 1100603 - Ignore message, not our networkID.
1100683: [P151] DUCO SER GW: PCF DATA addr 0x38 value 0xc1
1101770: [P151] DUCO SER GW: PCF DATA addr 0x38 value 0x41
1101772: [P150] DUCO RF GW: 1101771 - RECEIVED message: SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0007a953;Type:6; Bytes:9; Counter:10
1101772: [P150] DUCO RF GW: 1101771 -
1101772: [P150] DUCO RF GW: 1101771 - Ignore message, not our networkID.
1101782: [P151] DUCO SER GW: PCF DATA addr 0x38 value 0xc1
1120183: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
1120184: [P150] DUCO RF GW: State: 0.00`

arjan-io, 2022-04-06:
Is er een manier om Network ID en Device Address handmatig uit de log te vissen?

rhvs, 2022-04-06:
Ik zou eerst even de Duco box resetten - dat heeft mij eerder geholpen.

arnemauer, 2022-04-06:
> Ik zou eerst even de Duco box resetten - dat heeft mij eerder geholpen.

Dit verwijderd wel al je instellingen!
Wat is de afstand tussen de gateway en je ducobox?

arjan-io, 2022-04-06:
> > Ik zou eerst even de Duco box resetten - dat heeft mij eerder geholpen.
> 
> Dit verwijderd wel al je instellingen! Wat is de afstand tussen de gateway en je ducobox?

Daar was ik ook bang voor, dat het hele rataplan op nieuw ingesteld moet worden. Er hangen 2 remotes aan. 1 op 230V en 1 met zo'n batterij. 

De afstand heb variëren tussen 1 meter tot 5 meter (andere kamer) van de box zelf. Ik had inderdaad gelezen dat dit eventueel een issue zou kunnen zijn....

Het Serial gedeelte doet het wel!

arnemauer, 2022-04-06:
Staat de box in installermodus?
Hij reageert namelijk niet op het join bericht van de gateway.

arjan-io, 2022-04-06:
Ik heb 5 seconden op de batterij-remote gedrukt tot lampje knipperde. Het lampje op de box knippert ook snel (groen). Dat is goed toch?

arjan-io, 2022-04-06:
Heb inmiddels een Factory Reset uitgevoerd, onze tweede box geprobeerd, de stekker van 1 box er uit gehaald om minder RF verkeer en storing te krijgen maar helaas lijkt niets te werken...

Nog suggesties?

arjan-io, 2022-04-07:
1 van de boxen reageert nu maar nu is er een invalid join package...

515184: HTTP: JOIN
515185: Command: JOIN
515185: [P150] DUCO RF GW: Sent command for 'join' to DUCO unit
515192: [P150] DUCO RF GW: 515185 - SendJoinPacket()
515192: [P150] DUCO RF GW: 515186 - SENT message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type:1; Bytes:13; Counter:1;
515193: [P150] DUCO RF GW: 515186 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
515193: [P150] DUCO RF GW: 515192 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
515421: [P150] DUCO RF GW: 515420 - RECEIVED message: SRC:2; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0007a9ea;Type:2; Bytes:13; Counter:5;
515422: [P150] DUCO RF GW: 515420 - 00,00,7C,3E,
515422: [P150] DUCO RF GW: 515421 - Received messagetype: JOIN2
515422: [P150] DUCO RF GW: 515421 - processJoin2Packet()
515422: [P150] DUCO RF GW: 515421 - joinPacketValidNetworkId()
515423: [P150] DUCO RF GW: 515421 - SendJoin3Packet: INVALID join2 packet received!
519132: [P150] DUCO RF GW: 519131 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008460b;Type:0; Bytes:10; Counter:15
519133: [P150] DUCO RF GW: 519132 - 00,
519133: [P150] DUCO RF GW: 519132 - Ignore message, not our networkID.
523914: [P150] DUCO RF GW: 523913 - RECEIVED message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008460b;Type:7; Bytes:13; Counter:1;

arjan-io, 2022-04-11:
Ik heb geprobeerd om de nieuwe firmware the flashen. Eerst OTA maar dan staat op de 'main' page nog steeds de oude firmware versie. 

Daarna heb ik de gateway aan een Raspberry Pi gehangen met de USB aansluiting. Ik krijg wel een ttyUSB1 te zien (dat is de juiste, er hangt al iets anders aan USB0). Ik probeer de tool esptool.py the gebruiken maar die zegt vervolgens 'No serial connection could be established'. Ik heb _geen_ RF-gateway geconfigureerd en zelfs eerst een factory reset gedaan en daarna alleen de wifi ingesteld.

Iemand ervaring met deze tool of nog een idee wat ik zou kunnen proberen?

arjan-io, 2022-04-11:
Less is more. Minder opties gaven wél resultaat. Vanavond of morgen proberen te linken met de box!

arnemauer, 2022-04-11:
@arjan-io  Heb je de versie van vandaag erop gezet? :)

arjan-io, 2022-04-12:
Nog steeds een Invalid JOIN2 packet - error

arnemauer, 2022-04-12:
Ik zie in mijn glazen bol dat er elk moment een log gepost gaat worden 😆

arjan-io, 2022-04-12:
Zelfde als eerder :|

811866: WD : Uptime 14 ConnectFailures 0 FreeMem 14016 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
818497: Command: JOIN
818498: [P150] RF GW: Sent command for 'join' to DUCO unit
818505: [P150] RF GW: 818498 - SendJoinPacket()
818505: [P150] RF GW: 818499 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
818506: [P150] RF GW: 818499 - Bytes:13; Counter:11; WaitForAck: 0;
818506: [P150] RF GW: 818499 - 00,01,00,00,7C,3E,08,02,0B,00,00,00,0C,
818506: [P150] RF GW: 818505 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
818681: [P150] RF GW: 818680 - RECEIVED message (inboxQ 0): SRC:2; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0007a9ea;Type: 2;
818682: [P150] RF GW: 818680 - Bytes:13; Counter:15; RSSI:-94 (0xD7); LQI: 0x40
818682: [P150] RF GW: 818680 - 00,00,7C,3E,
818682: [P150] RF GW: 818680 - Received messagetype: JOIN2
818682: [P150] RF GW: 818681 - processJoin2Packet()
818683: [P150] RF GW: 818681 - joinPacketValidNetworkId()
818683: [P150] RF GW: 818681 - SendJoin3Packet: INVALID join2 packet received!
831361: [P150] RF GW: 831360 - RECEIVED message (inboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008107b;Type: 7;
831362: [P150] RF GW: 831361 - Bytes:13; Counter:6; RSSI:-99 (0xCE); LQI: 0x7F
831362: [P150] RF GW: 831361 - 20,00,52,00,
831362: [P150] RF GW: 831361 - Ignore message, not our networkID.

arjan-io, 2022-04-12:
Ik moet trouwens de transmit power op max zetten om bovenstaande log te krijgen. Laat ik alle op default staan, kom ik niet verder dan

184801: HTTP: JOIN
184802: Command: JOIN
184803: [P150] RF GW: Sent command for 'join' to DUCO unit
184811: [P150] RF GW: 184804 - SendJoinPacket()
184811: [P150] RF GW: 184804 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
184812: [P150] RF GW: 184804 - Bytes:13; Counter:5; WaitForAck: 0;
184812: [P150] RF GW: 184804 - 00,01,00,00,7C,3E,08,02,05,00,00,00,0C,
184812: [P150] RF GW: 184810 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
186190: [P150] RF GW: 186189 - RECEIVED message (inboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00084742;Type: 0;
186191: [P150] RF GW: 186190 - Bytes:10; Counter:14; RSSI:-89 (0xE1); LQI: 0x2F
186191: [P150] RF GW: 186190 - 00,
186191: [P150] RF GW: 186190 - Ignore message, not our networkID.
190310: [P150] RF GW: 190309 - RECEIVED message (inboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00084742;Type: 7;
190310: [P150] RF GW: 190309 - Bytes:13; Counter:15; RSSI:-90 (0xDF); LQI: 0x32
190311: [P150] RF GW: 190309 - 20,00,52,00,
190311: [P150] RF GW: 190309 - Ignore message, not our networkID.
211887: WD : Uptime 4 ConnectFailures 0 FreeMem 9664 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
228163: [P150] RF GW: UPDATE by PLUGIN_READ
228163: [P150] RF GW: State: 0.00
239670: [P150] RF GW: 239669 - RE

arnemauer, 2022-04-12:
Als je de gateway op een meter afstand van je ducobox laat liggen. En je kijkt in de weblog. 
Zie je dan ook berichten van de Ducobox zelf voorbijkomen? (dus met SRC: 1). 

In jou log zie ik dit:
818505: [P150] RF GW: 818499 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
818506: [P150] RF GW: 818499 - Bytes:13; Counter:11; WaitForAck: 0;
818506: [P150] RF GW: 818499 - 00,01,00,00,7C,3E,08,02,0B,00,00,00,0C,
818506: [P150] RF GW: 818505 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
818681: [P150] RF GW: 818680 - RECEIVED message (inboxQ 0): **SRC:2; DEST:0; ORG.SRC:1; ORG.DEST:0;** Ntwrk:0007a9ea;Type: 2;
818682: [P150] RF GW: 818680 - Bytes:13; Counter:15; RSSI:-94 (0xD7); LQI: 0x40
818682: [P150] RF GW: 818680 - 00,00,7C,3E,
818682: [P150] RF GW: 818680 - Received messagetype: JOIN2
818682: [P150] RF GW: 818681 - processJoin2Packet()
818683: [P150] RF GW: 818681 - joinPacketValidNetworkId()
818683: [P150] RF GW: 818681 - SendJoin3Packet: INVALID join2 packet received!

Dat is een bericht van je Ducobox (ORG.SRC:1) die doorgestuurd wordt door een bedieningsschakelaar (SRC:2;). Dat heb ik een keer eerder gezien maar ik moet even de archieven in duiken om te zien hoe dat is opgelost.

Aanmelden via een repeater is in principe prima. Daarvoor moet ik een aantal kleine aanpassingen maken in de code.

arnemauer, 2022-04-12:
De makkelijkste optie is om de gateway op de ducobox aan te sluiten en de gateway aan te sluiten op een laptop.
Vervolgens de seriële plugin in "network tool" modus te zetten en vervolgens de Duco Network Tool te openen.
Zie ook: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/DUCO-Network-tool

Je kan vervolgens het service menu activeren en "driver log" aanzet. Daarna klik je op refresh.
Onder de knop communicatie kun je het networkid vinden van je ducobox (zodat je hem op de ducobox kan schrijven :) ).
In het menu kun je een DNT bestand opslaan. De duco network tool kan je daarna afsluiten. 

In de map mijn documenten\DucoDriver\Logs staat als het goed is een of meerdere tekstbestanden/logs. Als je de DNT én de logs hier wilt posten dan kan ik kijken welk type ducobox je hebt en conroleren of hij op een afwijkende frequentie werkt.

arjan-io, 2022-04-12:
Duidelijk verhaal. Dat wordt m'n oude windows laptop afstoffen dus er gaan wel wat dagen over heen alvorens ik hier op terug kan komen. Ik schijn ook nog te moeten werken

arjan-io, 2022-04-13:
In 'voorbereiding' wilde ik de Serial Gateway alvast instellen maar ik zie de optie _Vink 'Disable gateway serial to use gateway as usb-cable for network tool' aan_ niet.
[image]

arnemauer, 2022-04-13:
Heb je al op save geklikt?

arjan-io, 2022-04-13:
Gelukt!

arnemauer, 2022-04-13:
> Gelukt! Nu die Windows laptop nog ;)

Of virtualbox met Windows:)

arjan-io, 2022-04-15:
Ik heb een eerste connectie gelegd. Is het de bedoeling dat ik probeer de Gateway aan te melden als ik de Netwerk Tool heb draaien? Ik had nu de RF-kant voor de zekerheid uitgeschakeld...

log file:
15-4-2022 18:47:05 - >> FanParaSet 1 80
15-4-2022 18:47:05 - fanparaset 1 80
  
15-4-2022 18:47:06 - Done
  
> [DATAMNGR] SAVE - CRC 0x096B
  
15-4-2022 18:47:07 - >> dataSave
15-4-2022 18:51:07 - >> InstallerSet 1
15-4-2022 18:51:14 - >> InstallerSet 0
15-4-2022 18:51:36 - >> Network

DAT-file:
    ˇˇˇˇ          IDuco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null   ISystem, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089   Duco_PcTool.GUI.SaveObjectV4   
componentsfileInformation˘System.Collections.Generic.SortedDictionary`2[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[Duco_PcTool.InterfaceComponentsV2, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]]   %Duco_PcTool.Constants+FileInformation      	      QSystem.Drawing, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a˚ˇˇˇ%Duco_PcTool.Constants+FileInformation   projectName
familyNamestreetNumbercitycountryphoneemailreportPersonreportReferenceIdnotespicture1picture2picture3
problemImage1
problemImage2
dntVersionclientSignatureinstallerSignature
clientName
installerNameinstallerCompanyinterventionTypeproblemCauseproblemTypesystemstartupControlInfoSystem.Drawing.Image   System.Drawing.Image   System.Drawing.Image   System.Drawing.Image   System.Drawing.Image   System.Drawing.Image   System.Drawing.Image   "Duco_PcTool.Constants+ProblemCause   !Duco_PcTool.Constants+ProblemType   ‚System.Collections.Generic.Dictionary`2[[System.String, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[System.Object, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]]   

   3.10.4.0 (DNT/Windows)

¯ˇˇˇ"Duco_PcTool.Constants+ProblemCause   value__        ˜ˇˇˇ!Duco_PcTool.Constants+ProblemType   value__        

   ˘System.Collections.Generic.SortedDictionary`2[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[Duco_PcTool.InterfaceComponentsV2, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]]   _setÍSystem.Collections.Generic.TreeSet`1[[System.Collections.Generic.KeyValuePair`2[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[Duco_PcTool.InterfaceComponentsV2, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]], mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]]      	
   
   ÍSystem.Collections.Generic.TreeSet`1[[System.Collections.Generic.KeyValuePair`2[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[Duco_PcTool.InterfaceComponentsV2, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]], mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]]   CountComparerVersionItems  éSystem.Collections.Generic.SortedDictionary`2+KeyValuePairComparer[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[Duco_PcTool.InterfaceComponentsV2, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]]   ˜System.Collections.Generic.KeyValuePair`2[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[Duco_PcTool.InterfaceComponentsV2, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]][]      	      	      éSystem.Collections.Generic.SortedDictionary`2+KeyValuePairComparer[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[Duco_PcTool.InterfaceComponentsV2, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]]   keyComparerâSystem.Collections.Generic.GenericComparer`1[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]]   	
             ıSystem.Collections.Generic.KeyValuePair`2[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[Duco_PcTool.InterfaceComponentsV2, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]]ÚˇˇˇıSystem.Collections.Generic.KeyValuePair`2[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089],[Duco_PcTool.InterfaceComponentsV2, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]]   keyvalue -Duco_PcTool.Components.ComponentChildBmbBasic      	   ˇˇˇÚˇˇˇ   	   ÓˇˇˇÚˇˇˇ   	   
   âSystem.Collections.Generic.GenericComparer`1[[System.Int32, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]]       -Duco_PcTool.Components.ComponentChildBmbBasicZ   nodeNetworkRowrfSwitchInfotypesubtypeproductVersionsoftwareVersion
productSerial
networkStatuslocationautoUpdateFunctionsisInitialUpdatedlastUpdateAllInfolastUpdateBasicNetworkInfolastUpdateUserSettingslastUpdateServiceSettingslastUpdateSensorslastUpdateDeviceSpecificlastUpdateErrorslastUpdateRfStatisticsrfStatisticssensorValuesfanCalibrationfanPerformance	bmbIoInfofanParafanCtrlInfofrostProtectInfo
filterInfo
[... cut ...]
      d   «˝ˇˇY˛ˇˇ∆˝ˇˇX˛ˇˇ   ≈˝ˇˇW˛ˇˇ	   ƒ˝ˇˇX˛ˇˇ   =  manual 3d   
2      d   ¬˝ˇˇY˛ˇˇ¡˝ˇˇX˛ˇˇ   ¿˝ˇˇW˛ˇˇ
   ø˝ˇˇX˛ˇˇ   B  manual timeout   
      '  ¶         ÇSystem.Collections.Generic.KeyValuePair`2[[Duco_PcTool.Constants+Setting, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null],[Duco_PcTool.Constants+ConfigRow, Duco Network Tool, Version=3.10.4.0, Culture=neutral, PublicKeyToken=null]]Ω˝ˇˇY˛ˇˇº˝ˇˇX˛ˇˇ:   ª˝ˇˇW˛ˇˇˇˇˇˇ∫˝ˇˇX˛ˇˇ:   	Ë      
       ˇ   

arnemauer, 2022-04-15:
Zou je de .dnt willen inpakken in een zip en hier posten? :)

arjan-io, 2022-04-17:
Poging #2:
[DD_220416_Saturday.log](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/8501034/DD_220416_Saturday.log)
[DucoNetwork_RS1927202994_220416-205341.zip](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/8501035/DucoNetwork_RS1927202994_220416-205341.zip)

arjan-io, 2022-05-27:
Nieuwe gateway vandaag ontvangen en die werkt gelijk! Ik wacht met smart op de nieuwe firmware maar tot dusver bedankt voor alle hulp!

closed 2022-05-27 by arjan-io