arnemauer/Ducobox-ESPEasy-Plugin #95: CO2 value blijft 0

xylone, 2022-04-06:
Van 1 externe sensor krijg ik wel de data, van een tweede alleen de temperatuur.
Zie wel wat invalid responses, maar weet niet zeker of het van die sensor is.
Heb de sensors met trial and error toegevoegd.
Werkende sensor zit gewoon op 2 (Woonkamer) en daarna ben ik tot ID 10 doorgegaan, maar alleen op 3 krijg ik wel een reactie op de temperatuur, op de rest niets.

Zie de log voor details (heb niet alles gelogd)

Ik denk dat ik dus even met de Duco netwerk tool aan de slag moet, maar de optie om 'Disable gateway serial to use gateway as usb-cable for network tool' aan te vinken heb ik niet. 

[co2.txt](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/8428061/co2.txt)

arnemauer, 2022-04-06:
Heb je gekeken in de seriele plugin waar je "main plugin" hebt geselecteerd?
Heb je bij "Hardware type" versie "Ventilation gateway V8.3 and lower" geselecteerd?

xylone, 2022-04-06:
Staat op de default (DIY esp8266 hardware) (Allemaal overigens...)
[image]

xylone, 2022-04-06:
Ja dat lost het op. En ik heb hem kunnen verbinden.
Let wel op met het downloaden van de netwerk tool:
Er staat een button links die de oude versie download (2.x) en een link bij de handleidingen die de 3.x download.
Je moet dus de 3.x versie hebben.

Tot mijn verbazing is het dus geen CO2 sensor, maar een Humidity sensor.... (Dat wordt weer een email naar het bouwbedrijf, maar dat terzake)
Dus ik denk ik voeg de Externe Humidity Sensor even toe.
Maar helaas... Ook deze heeft moeite met het ontvangen van de informatie:

320097: [P155] DUCO Ext. Hum. Sensor: start read, eventid:8
320098: [P155] DUCO Ext. Hum. Sensor: Read external sensor
320098: [P155] DUCO Ext. Hum. Sensor: Start read external sensor. NodeAddress: 3. Type: temp
320438: [P155] DUCO Ext. Hum. Sensor: ROW:1 bytes:16
320439: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: nodeparaget 3 73 END
320439: [P155] DUCO Ext. Hum. Sensor: Command confirmed by ducobox
320439: [P155] DUCO Ext. Hum. Sensor: ROW:2 bytes:7
320440: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: ^ END
320440: [P155] DUCO Ext. Hum. Sensor: ROW:3 bytes:13
320441: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: Unknown Cmd END
320441: [P155] DUCO Ext. Hum. Sensor: ROW:4 bytes:0
320442: [P155] DUCO Ext. Hum. Sensor: End of message received.
320442: [P155] DUCO Ext. Hum. Sensor: Stop reading from serial
320515: [P155] DUCO Ext. Hum. Sensor: Start read external sensor. NodeAddress: 3. Type: RH
320918: [P155] DUCO Ext. Hum. Sensor: ROW:1 bytes:16
320919: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: nodeparaget 3 75 END
320920: [P155] DUCO Ext. Hum. Sensor: Received invalid response
320920: [P155] DUCO Ext. Hum. Sensor: Stop reading from serial
320921: [P155] DUCO Ext. Hum. Sensor: ROW:2 bytes:23
320921: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: Get PARA 75 of NODE 3 END

arnemauer, 2022-04-06:
Heb je de duco network tool volledig afgesloten voordat je de gateway/ducobox hebt afgekoppeld? 
De duco network tool zet de ducobox in een debug modus die continu debug informatie over de seriële verbinding stuurt.
Als de gateway dan een commando stuurt dan krijg je die debug informatie er tussendoor. 

Kun je een wat langere log hier posten?

xylone, 2022-04-07:
Even de title aangepast, zodat het matched met de discussie.

Ja ik had hem volgens mij wel afgesloten voor ontkoppelen. Voor de zekerheid gisteravond nogmaals geconnect en applicatie afgesloten voor ontkoppelen. (De Exit knop werkt in die tool overigens niet, dus met het kruisje afgesloten. In de hoop dat dat ook de 'debug' modus uit zet.)
Voor de zekerheid ook nog even de stroom van de Duco afgehaald.

In de netwerk tool zie ik values staan bij humidity, maar deze komt dus niet door:
[image]

Er zit volgens de netwerk tool ook nog een derde sensor (CO2) aan het systeem, maar die hangt denk ik bij de buren.... 😵‍💫 
[image]

Log:
[co2.txt](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/8440713/co2.txt)

arnemauer, 2022-04-07:
Het bereik van je humidity sensor en co2 is erg slecht. Je zou kunnen proberen om de zendsterkte van de humidity en co2 bedieningsschakelaar hoger te zetten. Daarvoor activeer je het servicemenu: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/wiki/DUCO-Network-tool#servicemenu Daarna kun je driverlog ook aanzetten (deze slaat de seriele data op tussen de network tool en de ducobox).  Vervolgens klik je op de schakelaar en aan de rechterkant heb je nu een dropdown menu met Service settings. Daar kun je (als de schakelaar het ondersteunt) de zendsterkte hoger zetten.

Die laatste co2 is waarschijnlijk de RF gateway. Die meldt zich aan als co2 sensor. En die hoef je ook niet uit te lezen qua co2 waarde :)

Daarnaast staat logging nog aan op je ducobox. Zie bijvoorbeeld:

```
108133525: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 5. Type: temp
108133878: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
108135513: [P154] DUCO Ext. CO2 Sensor: ROW:1 bytes:15
108135513: [P154] DUCO Ext. CO2 Sensor: Pakket ontvangen: **Search 4 (0xF0)** END
```

Dit kun je uitschakelen door de gateway in de network tool modus te zetten. En een terminal te openen en het volgende commando geven: coreloglevel 0x10

xylone, 2022-04-07:
**Bereik**
Ah, top! Ga ik vanavond even proberen. Had het service menu rechts al gezien, maar was daar nog even vanaf gebleven.

**Co2 sensor**
Ohhhh gelukkig! :-)

**Logging**
Zal ik ook even uitvoeren vanavond.

xylone, 2022-04-08:
**Bereik**
DONE: In het communicatie overzicht staat deze nu netjes op groen.

**Logging**
DONE:

46471772: WD   : Uptime 775 ConnectFailures 0 FreeMem 6328 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
46487329: [P155] DUCO Ext. Hum. Sensor: Init plugin done.
46487727: [P155] DUCO Ext. Hum. Sensor: start read, eventid:8
46487728: [P155] DUCO Ext. Hum. Sensor: Read external sensor
46487728: [P155] DUCO Ext. Hum. Sensor: Start read external sensor. NodeAddress: 3. Type: temp
46488072: [P155] DUCO Ext. Hum. Sensor: ROW:1 bytes:15
46488073: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: nodearaget 3 73 END
46488073: [P155] DUCO Ext. Hum. Sensor: Received invalid response
46488073: [P155] DUCO Ext. Hum. Sensor: Stop reading from serial
46488074: [P155] DUCO Ext. Hum. Sensor: ROW:2 bytes:23
46488075: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen:   Get PARA 73 of NODE 3 END
46488215: [P155] DUCO Ext. Hum. Sensor: Start read external sensor. NodeAddress: 3. Type: RH
46488549: [P155] DUCO Ext. Hum. Sensor: ROW:1 bytes:16
46488550: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: nodeparaget 3 75 END
46488550: [P155] DUCO Ext. Hum. Sensor: Received invalid response
46488551: [P155] DUCO Ext. Hum. Sensor: Stop reading from serial
46500729: [P156] DUCO FanSpeed: Fanspeed: 394 RPM
46500730: [P156] DUCO FanSpeed: ROW:3 bytes:6
46500730: [P156] DUCO FanSpeed: Pakket ontvangen:   Done END
46500730: [P156] DUCO FanSpeed: ROW:4 bytes:2
46500731: [P156] DUCO FanSpeed: Pakket ontvangen:    END
46500731: [P156] DUCO FanSpeed: End of message received.
46500731: [P156] DUCO FanSpeed: Stop reading from serial
46500755: [P154] DUCO Ext. CO2 Sensor: start read, eventid:10
46500755: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
46500756: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: temp
46500793: [P153] DUCO BOX SENSOR: start read, eventid:11
46500793: [P153] DUCO BOX SENSOR: Serial port in use, set flag to read data later.
46501091: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
46501107: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
46501770: WD   : Uptime 775 ConnectFailures 0 FreeMem 6880 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
46502070: [P154] DUCO Ext. CO2 Sensor: End of message received.
46502070: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
46502214: [P151] DUCO SER GW: Start readNetworkList
46502370: [P151] DUCO SER GW: Received invalid response on network
46502370: [P151] DUCO SER GW: Stop reading from serial
46503214: [P154] DUCO Ext. CO2 Sensor: start read, eventid:9
46503215: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
46503215: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: CO2
46503551: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
46503567: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
46504719: [P154] DUCO Ext. CO2 Sensor: End of message received.
46504719: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
46505214: [P153] DUCO BOX SENSOR: start read, eventid:11
46505215: [P153] DUCO BOX SENSOR: Start readBoxSensors
46505433: [P153] DUCO BOX SENSOR: Received correct response on sensorinfo
46505461: [P153] DUCO BOX SENSOR: Send command successfull!
46505465: [P153] DUCO BOX SENSOR: RH: 4642 = 46.42%
46505466: [P153] DUCO BOX SENSOR: TEMP: 200 = 20.0Â°C
46505466: [P153] DUCO BOX SENSOR: End of message received.
46505466: [P153] DUCO BOX SENSOR: Stop reading from serial
46531770: WD   : Uptime 776 ConnectFailures 0 FreeMem 5304 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
46548140: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen: nodeparaget 3 73 END
46548140: [P155] DUCO Ext. Hum. Sensor: Command confirmed by ducobox
46548141: [P155] DUCO Ext. Hum. Sensor: ROW:2 bytes:23
46548141: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen:   Get PARA 73 of NODE 3 END
46548155: [P155] DUCO Ext. Hum. Sensor: Send command successfull!
46548258: [P155] DUCO Ext. Hum. Sensor: ROW:3 bytes:9
46548259: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen:   --> 204 END
46548260: [P155] DUCO Ext. Hum. Sensor: TEMP: 204 = 20.4Â°C
46548261: [P155] DUCO Ext. Hum. Sensor: ROW:4 bytes:6
46548261: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen:   Done END
46548263: [P155] DUCO Ext. Hum. Sensor: ROW:5 bytes:2
46548263: [P155] DUCO Ext. Hum. Sensor: Pakket ontvangen:    END
[... 22 lines cut ...]
46561395: [P151] DUCO SER GW: Send command successfull!
46561770: WD   : Uptime 776 ConnectFailures 0 FreeMem 6688 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
46563987: [P151] DUCO SER GW: Received invalid response on network
46563987: [P151] DUCO SER GW: Stop reading from serial
46564215: [P154] DUCO Ext. CO2 Sensor: start read, eventid:10
46564215: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
46564216: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 2. Type: temp
46564561: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
46564565: [P154] DUCO Ext. CO2 Sensor: End of message received.
46564565: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
46565215: [P153] DUCO BOX SENSOR: start read, eventid:11
46565215: [P153] DUCO BOX SENSOR: Start readBoxSensors
46565421: [P153] DUCO BOX SENSOR: Received correct response on sensorinfo
46565426: [P153] DUCO BOX SENSOR: RH: 4675 = 46.75%
46565429: [P153] DUCO BOX SENSOR: TEMP: 200 = 20.0Â°C
46565429: [P153] DUCO BOX SENSOR: End of message received.
46565430: [P153] DUCO BOX SENSOR: Stop reading from serial

Helaas nog geen vochtigheid.
Nu zag ik in #96 dat je debug info van 1 node kunt uitlezen:
> coreloglevel 0xff
Done
> nodeinfo 3

Is dat nog nuttig om hier de log van te posten?

arnemauer, 2022-04-08:
Zou je een terminal op je pc willen openen en het commando "nodeparaget 3 73" een aantal keer willen geven?
Ik ben benieuwd wat hij dan als output geeft.

xylone, 2022-04-10:
214A 73 of NODE 3

arnemauer, 2022-04-11:
Is dit de enige output? Er zal minimaal iets van unknown command moeten staan.

Heb je copy paste gebruikt of heb je "nodeparaget 3 73" overgetypt? Als je hem plakt dan kan de ducobox het niet "bijhouden" en dan mis je een aantal letters in het commando. 

Zou je anders de log van de duco network tool kunnen posten? Die staan in "mijn documenten\DucoDriver\Logs".

xylone, 2022-04-11:
Ja ik vond het ook al erg weinig....
Ik heb het ingetikt.
Heb wel het idee dat putty \n niet helemaal goed afhandelt, dus wellicht wordt de regel steeds overschreven.
Waarschijnlijk kun je wel iets instellen in putty, maar ik kon het niet vinden. (Ben geen putty gebruiker (overigens ook geen Windows gebruiker ;-) ))
Zal het vanavond nogmaals proberen, maar dan met een terminal emulator op OSX of Linux.

arnemauer, 2022-04-11:
In putty kun je deze vinkjes aanzetten :)
[image]

xylone, 2022-04-11:
> nodeparaget 3 73
  Get PARA 73 of NODE 3
  --> 217
  Done

Dus alleen de temperatuur krijgt hij terug....

arnemauer, 2022-04-12:
En met nodeparaget 3 75?

xylone, 2022-04-13:
Yep, de vochtigheid:

> nodeparaget 3 75
  Get PARA 75 of NODE 3
  --> 5445
  Done

arnemauer, 2022-04-15:
Wil je de nieuwe firmware van vandaag willen testen? b756d3e
Het zou nu opgelost moeten zijn.

xylone, 2022-04-16:
Firmware geflashed en komt inderdaad nu mee. Top!
Zie ook dat je al met #93 bezig bent. ;-)

closed 2022-04-16 by xylone

arnemauer, 2022-04-16:
> Firmware geflashed en komt inderdaad nu mee. Top! Zie ook dat je al met #93 bezig bent. ;-)
Ik heb net een nieuwe firmware neergezet: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/c051e3833947d369fd999390c15f95666f39e048