arnemauer/Ducobox-ESPEasy-Plugin #97: Bug: 150 - RF Gateway: Ontvangen van "permanente standen" komen binnen als "tijdelijke stand"

arnemauer, 2022-04-09:
Afgelopen week de gateway ontvangen en vanochtend de meest verse firmware erop gezet. Spullenboel netjes aangesloten/ingesteld en met behulp van wat info hierboven de schakelaars in Home Assistant aan kunnen maken. Ik loop alleen vast op de stand van de schakelaars.

Hoog/Midden/Laag gaat goed, maar '15min' of 'permanent' wordt niet opgepakt. Dat wil zeggen, als ik voor een 'permanent hoog' kies, dan:
-  gaat de box keurig naar permanent hoog
-  de wandbediening geeft ook blauwe golfjes

Maar in Home Assistant zie ik na enkele seconden de schakelaar van 'permanent hoog' uitgaan en de schakelaar van '15 min hoog' aangaan.

Een snelle blik in MQTT Explorer leert dat 'Ventilationmode' VENTILATION_GATEWAY/Ventilation_gateway_RF/Ventilationmode), een single digit teruggeeft. Dit is altijd een 1/2/3. De voorlopende 1 die de status 'permanent' zou moeten duiden, komt niet voor. :-(

Wat zie ik over het hoofd?

_Originally posted by @JJ-82 in https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/24#issuecomment-1091684091_

arnemauer, 2022-04-10:
Opgelost :) In de volgende firmware werkt hij goed.

closed 2022-04-10 by arnemauer

JJ-82, 2022-04-10:
Lekker bezig! Thx!