arnemauer/Ducobox-ESPEasy-Plugin #100: Percentage override not working

printe3r, 2022-04-12:
Hi, I am able to send commands for ventilation mode over MQTT and HTTP. However, when I use an override percentage, it appears to be ignored. I am following the documented command format VENTMODE,MODE,PERCENTAGE,BUTTON_PRESSES
For example, command: VENTMODE,LOW,100. This sets the system to LOW and not to 100%.

- Hardware: WeMos D1 R2 WiFi ESP8266
- Firmware: ESP_Easy_mega_20220331_custom_ESP8266_4M1M (earlier versions had the same issue)

arnemauer, 2022-04-13:
The override percentage is overridden by your ventilationmode :) 
The override percentage only works in auto mode. In automode a co2/humidity sensor can control the ventilation based on the co2 ppm or humidity percentage in a room. If you manually choose an other ventilation mode, the ducobox will ignore the sensors override percentage.

printe3r, 2022-04-13:
Thank you for clarifying :) I am not getting the expected result with AUTO ventilation mode. But maybe that is because the max and min percentages in Auto mode are bound to limits set using the DUCO Network Tool?

arnemauer, 2022-06-30:
Nope, override percentage is always between 0 and 100. 
If i send this command: VENTMODE,AUTO,30 or VENTMODE,AUTO,66
Ventilation_Percentage: is 30 and 66. 

Could you check de weblog?

arnemauer, 2022-10-25:
Ik heb het nogmaals getest en bij mij werkt het goed. Ik sluit het issue omdat er geen reactie meer is gekomen.

closed 2022-10-25 by arnemauer