arnemauer/Ducobox-ESPEasy-Plugin #103: P150 - RF gateway: uitlezen CC1101 via SPI in interrupt

arnemauer, 2022-04-19:
Wanneer de CC1101 een pakket ontvangt zal hij een GDO pin hoog maken. De pin is verbonden met de ESP8266 en zorgt ervoor dat er  een interrupt service routine (ISR) wordt uigevoerd. Deze ISR wordt bij voorkeur zo kort mogelijk gehouden, zodat de esp8266 ook aan het hoofdprogramma toekomt en klaar is om nieuwe interrupts af te handelen. 

In de ISR wordt op dit moment data uit de CC1101 uitgelezen zodat de buffer in de CC1101 weer zo snel mogelijk leeg is om een nieuwe radiopakket te ontvangen. Dit uitlezen in een interrupt zorgt soms voor een exception waarna de ESP reboot.

Een mogelijke oplossing is om in de interrupt een variabele te markeren en in de hoofdcode te kijken of de variabele is gemarkeerd. Wanneer de variabele gemarkeerd is dan wordt de CC1101 uitgelezen en de data verwerkt. De controle of de variabele gemarkeerd is zal in 'PLUGIN_FIFTY_PER_SECOND' plaatsvinden. Dit komt erop neer dat de functie iedere 20ms wordt aangeroepen. Dit is wel best effort, door achtergrondtaken kan het zijn dat het langer duurt voordat de functie wordt aangeroepen.

arnemauer, 2022-04-24:
De gateway loopt erg stabiel wanneer de uitlezen van de CC1101 niet meer in de interrupt gebeurt.
Hierdoor is wel een bekende bug in de CC1101 naar boven gekomen. Zie daarvoor: https://www.ti.com/lit/er/swrz020e/swrz020e.pdf onder RXFIFO_OVERFLOW Issue.

De interrupt pin blijft in deze situatie continu hoog en de radio blijft in RX modus maar ontvangt geen pakketjes.

Gibletron, 2022-04-24:
@arnemauer mogelijk dat ik hier last van heb.
Ik heb de "fout" gemaakt mijn firmware een paar dagen geleden te updaten naar de laatst beschikbare versie
Sindsdien is het me niet meer gelukt om de gateway stabiel te krijgen.
Ik heb praktisch alle beschikbare (oudere) firmwares getest (uiteraard door eerst de 4MB lege bin te flashen)
Daarna een reboot, een flash van de specifieke firmware, vervolgens enkel Wifi ingesteld en de controller zodat hij tegen HA aan praat
Welke firmware ik ook test de gateway valt na ~15 minuten weg met als reset reason "exception"
Heb ondertussen 6 verschillende USB voedingen en kabels getest, zelfs de combinatie die altijd stabiel was (voor de firmware upgrade), gebruik nu een Samsing blokje van 2A met een originele samsung kabel, maar helaas zelfde probleem.

Mocht het makkelijker zijn om even via mail of ander platform te communiceren dan kan dat: post4gabe is mijn gmail adres

arnemauer, 2022-04-24:
> @arnemauer mogelijk dat ik hier last van heb. Ik heb de "fout" gemaakt mijn firmware een paar dagen geleden te updaten naar de laatst beschikbare versie Sindsdien is het me niet meer gelukt om de [...]

Zou je deze melding onder het tabblad discussion willen aanmaken? Dan post ik daar een bin die je kan testen :)

Gibletron, 2022-04-24:
> 
> Zou je deze melding onder het tabblad discussion willen aanmaken? Dan post ik daar een bin die je kan testen :)

https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/discussions/105#discussion-4031055

arnemauer, 2022-05-03:
Inmiddels een oplossing gevonden waardoor de RXFIFO_OVERFLOW regelmatig wordt gecheckt en indien nodig de radio opnieuw in RX gezet.

Nu is er wel een nieuwe bug opgetreden bij het opslaan van het network ID in CustomTaskSettings:

0x40285458 in flash_hal_write(unsigned int, unsigned int, unsigned char const*) at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/flash_hal.cpp:161
0x40283104 in spiffs_impl::SPIFFSImpl::spiffs_hal_write(unsigned int, unsigned int, unsigned char const*) at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs_api.h:339
0x40289968 in spiffs_phys_wr at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_cache.cpp:206
0x40287918 in spiffs_page_delete at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:919
0x402899c4 in spiffs_phys_wr at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_cache.cpp:225
0x40287a99 in spiffs_page_move at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:873
0x40287ea0 in spiffs_object_update_index_hdr at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:1046
0x40287918 in spiffs_page_delete at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:919
0x402899c4 in spiffs_phys_wr at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_cache.cpp:225
0x40288a99 in spiffs_object_modify at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:1666
0x40283120 in spiffs_impl::SPIFFSImpl::spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs_api.h:345
0x402856bc in spiffs_hydro_write at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_hydrogen.cpp:445
0x402899eb in spiffs_cache_page_allocate_by_fd at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_cache.cpp:256
0x4010128f in umm_free_core at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:351
0x40101558 in free at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:398
0x4029cda5 in operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc:48
0x40283047 in std::_Sp_counted_deleter<spiffs_impl::SPIFFSFileImpl*, std::__shared_ptr<spiffs_impl::SPIFFSFileImpl, (__gnu_cxx::_Lock_policy)0>::_Deleter<std::allocator<spiffs_impl::SPIFFSFileImpl> >, std::allocator<spiffs_impl::SPIFFSFileImpl>, (__gnu_cxx::_Lock_policy)0>::_M_destroy() at /root/.platformio/packages/toolchain-xtensa@2.40802.200502/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h:357
0x4028e54e in fs::File::close() at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.cpp:111
0x4025a758 in doSaveToFile(char const*, int, unsigned char const*, int, char const*) at src/src/Helpers_tmp/__tmpfile.cpp:2220
0x4028f0e0 in spiffs_cache_page_get_by_fd at ??:?
0x4027ef00 in random(long) at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WMath.cpp:45
0x4025a26c in fileExists(String const&) at src/src/Helpers_tmp/__tmpfile.cpp:1196 (discriminator 1)
0x4027efbc in String::~String() at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.cpp:125
0x4025c43c in SaveToFile(SettingsType::Enum, int, unsigned char const*, int, int) at src/src/Helpers_tmp/__tmpfile.cpp:2384
0x402d7ae6 in chip_v6_unset_chanfreq at ??:?
0x4027f990 in String::operator=(__FlashStringHelper const*) at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.cpp:294 (discriminator 1)
0x4025c6c5 in SaveToFile(SettingsType::Enum, int, unsigned char const*, int) at src/src/Helpers_tmp/__tmpfile.cpp:2366
0x4027efbc in String::~String() at /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.cpp:125
0x4022ec35 in Plugin_150(unsigned char, EventStruct*, String&) at src/_P150_DucoRFGateway.ino:701

arnemauer, 2022-05-22:
Ik ben nog dagelijks aan het testen en zoeken naar de bugs. Wanneer de ventilatiemodus veranderd dan stuurt de gateway de ventilatiestand direct door naar de controller met sendData(event). https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/6da449b9cdb0cc5097886c00981ba2d430552e35/_P150_DucoRFGateway.ino#L325

Dit commando zorgt ervoor dat de gateway niet meer reageert en de hardware watchdog de esp reset. Aangezien de hardware watchdog reset is er geen exception met debug informatie, in tegenstelling tot de software watchdog. 

Ik ben nu stuk voor de stuk de code eruit aan het halen om te kijken waar het precies mis gaat. Geen idee hoelang dit nog gaat duren maar ik ben er druk mee bezig! :)

Gibletron, 2022-05-22:
> Ik ben nog dagelijks aan het testen en zoeken naar de bugs. Wanneer de ventilatiemodus veranderd dan stuurt de gateway de ventilatiestand direct door naar de controller met sendData(event). [...]

Klinkt top! Moet zeggen dat het tot nu toe best goed gaat met statische IP instellingen
Hij blijft nu ongeveer 5 uur lang stabiel, en dat is acceptabel voor nu
Goed te horen dat je er nog aan werkt, en mocht ik iets kunnen testen dan hoor ik het graag :)

arnemauer, 2022-05-30:
Ik heb alle bugs gefixed en de gateway loopt nu weer stabiel :)
Ik hoor graag de nieuwe firmware bij jullie ook goed draait. 
e0e49de

Gibletron, 2022-05-30:
> 

Volgens mij is er nog één issue over..
Hij onthoud de Network ID niet na een reboot...

Gibletron, 2022-05-30:
> Ik heb alle bugs gefixed en de gateway loopt nu weer stabiel :) Ik hoor graag of de nieuwe firmware bij jullie ook goed draait. [e0e49de](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/e0e49de306ad8b9a94086ef6cb4dc80ceffe983e)

Volgens mij is er nog één issue over..
Hij onthoud de Network ID niet na een reboot...

arnemauer, 2022-05-30:
Klopt helemaal, ik had de binary er tijdelijk even afgehaald om het aan te passen. Het is inmiddels gefixt! :) 
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/tree/master/Binaries/Binary%20based%20on%20ESPEasy_mega-20220427

Gibletron, 2022-05-31:
> Klopt helemaal, ik had de binary er tijdelijk even afgehaald om het aan te passen. Het is inmiddels gefixt! :) https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/tree/master/Binaries/Binary%20based%20on%20ESPEasy_mega-20220427

Deze onthoud de NetworkID weer, top!
Wat ik wel zie is dat de fan RPM niet meer wordt uitgelezen

closed 2022-10-23 by arnemauer